beagleboard/Latest-Images #3: [4.19.x] BBXM: Regression in BBXM DVI LCD support?

RobertCNelson, 2020-03-03:
Between linux-image-4.19.90-armv7-x42 and
linux-image-4.19.94-armv7-x43, it seems, something has broken my
BBXM's ability to drive a DVI-D LCD panel that's attached to it,
either through Xorg or through the kernel's framebuffer.  There's
nothing terribly different between the two dmesg-s and I'm not sure
where else to look.

RobertCNelson, 2020-03-03:
Notes, both of these are working.. (Monitor: DELL 1707FP)

debian@beagleboard:~$ uname -r
4.19.103-armv7-x48

debian@beagleboard:~$ uname -r
4.19.94-armv7-x43

RobertCNelson, 2020-03-03:
```sudo /opt/scripts/tools/version.sh 
git:/opt/scripts/:[7993686e8e1643eaaf43b9ec412cac02ae8113bc]
model:[TI_OMAP3_BeagleBoard_xM]
dogtag:[BeagleBoard.org Debian Image 2020-03-01]
kernel:[4.19.94-armv7-x43]
nodejs:[v6.17.0]
/boot/uEnv.txt Settings:
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20200131.0-0rcnee0~stretch+20200131]
pkg:[bb-wl18xx-firmware]:[1.20200226.1-0rcnee0~stretch+20200226]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[librobotcontrol]:[1.0.4-git20190227.1-0rcnee0~stretch+20190327]
pkg:[firmware-ti-connectivity]:[20180825+dfsg-1rcnee1~stretch+20181217]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev gpio pwm eqep remoteproc admin spi iio docker tisdk weston-launch xenomai cloud9ide]
cmdline:[console=ttyO2,115200n8 root=/dev/mmcblk0p2 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0]
dmesg | grep remote
[  103.490478] smsc95xx 2-2.1:1.0 eth0: hardware isn't capable of remote wakeup
dmesg | grep pru
dmesg | grep pinctrl-single
[    1.295989] pinctrl-single 48002030.pinmux: 284 pins, size 568
[    1.296630] pinctrl-single 48002a00.pinmux: 46 pins, size 92
[    1.297424] pinctrl-single 480025a0.pinmux: 46 pins, size 92
dmesg | grep gpio-of-helper
lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 005: ID 413c:2003 Dell Computer Corp. Keyboard
Bus 002 Device 004: ID 0424:2504 Standard Microsystems Corp. 
Bus 002 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 002 Device 002: ID 0424:9514 Standard Microsystems Corp. SMC9514 Hub
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END
```

RobertCNelson, 2020-03-05:
For internal reference, it's this lcd: https://elinux.org/BeagleBone_xM_LCD7

i've got one somewhere...

nwf, 2021-06-12:
Out of some misplaced nostalgia, perhaps, I dusted this thing off and have it working using `4.19.191-armv7-x67` (the current `LTS419` kernel) and `5.4.123-armv7-x53` (`LTS54`)... **but** with the FDT *from `4.19.90-armv7-x42`*!  For the life of me I cannot see any meaningful difference between the FDTs supplied with `4.19.90` and `4.19.191`, and yet, the one from `4.19.90` boots and displays X with all these kernels, while booting these with their shipped FDTs merely boots and leaves the display with vertical streaks of random colors.

If you're feeling bored, I'd appreciate any insights or additional knobs to try varying.  The trick with the `4.19.90` FDT tragically does not allow `5.8.18-armv7-x26` (`STABLE`) or `5.9.16-armv7-x17` (`TESTING`) to boot; these hang shortly after jumping out of `u-boot`, AFAICT.  These kernels do boot using their supplied FDTs, but the display is not driven (and `dmesg` fails to contain `DSS: OMAP DSS rev 2.0`).

All relevant pieces of configuration should be up at https://www.ietfng.org/nwf/sysadmin/embcomp/bbxm.html including the u-boot boot script (which includes I2C commands to initialize the LCD), a FDT overlay for the touch controller, and the `xorg.conf` for actually using the thing.