beagleboard/Latest-Images #5: No USB network address on linux

GwenStouthuysen, 2020-03-03:
**Describe the bug**
A fresh install does boot and tries to install a network interface but it fails somehow.

ip addr command results:
6: enp0s20f0u1i2: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    link/ether 50:65:83:e8:0c:81 brd ff:ff:ff:ff:ff:ff
    inet6 fe80::99d5:7eaf:a92f:4ecc/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever
7: enp0s20f0u1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UNKNOWN group default qlen 1000
    link/ether 50:65:83:e8:0c:7f brd ff:ff:ff:ff:ff:ff
    inet6 fe80::a466:ae1d:c902:64d2/64 scope link noprefixroute 
       valid_lft forever preferred_lft forever
 

**Describe which image you are running 
https://rcn-ee.com/rootfs/bb.org/testing/2020-03-02/buster-console/bone-eMMC-flasher-debian-10.3-console-armhf-2020-03-02-1gb.img.xz

**Describe your hardware**
BeagelBone Black Wireless
Dell E5570

**Describe any Capes Installed**
None

**Please run this built-in script**
```sudo /opt/scripts/tools/version.sh```

**To Reproduce**
Steps to reproduce the behavior:
Fresh download installed and the network functionality 

**Expected behavior**
A clear and concise description of what you expected to happen.
Network interface to come up

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

RobertCNelson, 2020-03-03:
@GwenStouthuysen do you have a usb-serial adapter?  Can you run

```sudo /opt/scripts/tools/version.sh```

and report the output..

Regards,

RobertCNelson, 2020-03-03:
Just noticed this was the flasher, did you let it "flash" the eMMC, then reboot without the microSD card?  (while flashing, usb0/usb1 is not enabled..)

Regards,

RobertCNelson, 2020-03-03:
Flashed a BBBW, it does take a "LONG TIME" on first bootup after flashing, but i do get both usb devices showing up in dmesg..

```
[21966.253731] usb 1-9.2: new high-speed USB device number 18 using xhci_hcd
[21966.274485] usb 1-9.2: New USB device found, idVendor=1d6b, idProduct=0104, bcdDevice= 4.04
[21966.274487] usb 1-9.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[21966.274488] usb 1-9.2: Product: BeagleBoneBlack
[21966.274489] usb 1-9.2: Manufacturer: BeagleBoard.org
[21966.274490] usb 1-9.2: SerialNumber: 1651BBWG0005
[21966.301201] cdc_ether 1-9.2:1.2 eth2: register 'cdc_ether' at usb-0000:00:14.0-9.2, CDC Ethernet Device, b0:d5:cc:e5:e8:97
[21966.302249] usbcore: registered new interface driver cdc_ether
[21966.306710] cdc_acm 1-9.2:1.4: ttyACM0: USB ACM device
[21966.308302] usbcore: registered new interface driver cdc_acm
[21966.308304] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
[21966.310704] rndis_host 1-9.2:1.0 eth3: register 'rndis_host' at usb-0000:00:14.0-9.2, RNDIS device, b0:d5:cc:e5:e8:95
[21966.310725] usbcore: registered new interface driver rndis_host
[21966.313248] usbcore: registered new interface driver rndis_wlan
```
```
debian@beaglebone:~$ sudo /opt/scripts/tools/version.sh 
eeprom:[A335BNLTBWA51651BBWG0005]
model:[TI_AM335x_BeagleBone_Black_Wireless]
dogtag:[BeagleBoard.org Debian Console Image 2020-03-02]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2019.04-00002-g3d8c979660]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-boneblack-uboot-univ.dts]
UBOOT: Loaded Overlay:[AM335X-PRU-RPROC-4-19-TI-00A0]
UBOOT: Loaded Overlay:[BB-ADC-00A0]
UBOOT: Loaded Overlay:[BB-BBBW-WL1835-00A0]
UBOOT: Loaded Overlay:[BB-BONE-eMMC1-01-00A0]
UBOOT: Loaded Overlay:[BB-HDMI-TDA998x-00A0]
kernel:[4.19.94-ti-r35]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-19-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20200131.0-0rcnee0~buster+20200131]
pkg:[bb-wl18xx-firmware]:[1.20200226.1-0rcnee0~buster+20200226]
pkg:[kmod]:[26-1]
WARNING:pkg:[librobotcontrol]:[NOT_INSTALLED]
pkg:[firmware-ti-connectivity]:[20190114-2]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal netdev gpio pwm eqep remoteproc admin spi iio i2c docker tisdk weston-launch xenomai bluetooth cloud9ide]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 rng_core.default_quality=100 quiet]
dmesg | grep remote
[    1.598321] remoteproc remoteproc0: wkup_m3 is available
[    1.600506] remoteproc remoteproc1: 4a334000.pru is available
[    1.601036] remoteproc remoteproc2: 4a338000.pru is available
[    1.898203] remoteproc remoteproc0: powering up wkup_m3
[    1.898319] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[    1.901930] remoteproc remoteproc0: remote processor wkup_m3 is now up
dmesg | grep pru
[    1.600506] remoteproc remoteproc1: 4a334000.pru is available
[    1.600637] pru-rproc 4a334000.pru: PRU rproc node pru@4a334000 probed successfully
[    1.601036] remoteproc remoteproc2: 4a338000.pru is available
[    1.601137] pru-rproc 4a338000.pru: PRU rproc node pru@4a338000 probed successfully
dmesg | grep pinctrl-single
[    1.145521] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
[    1.158771] gpio-of-helper ocp:cape-universal: ready
END
```

```
debian@beaglebone:~$ systemd-analyze blame
    1min 10.424s generic-board-startup.service
         47.412s dev-mmcblk1p1.device
          5.826s bb-wl18xx-wlan0.service
          3.615s loadcpufreq.service
          3.565s systemd-udev-trigger.service
          2.476s networking.service
[... 42 lines cut ...]
Feb 14 10:13:40 beaglebone systemd[1]: bb-wl18xx-wlan0.service: Scheduled restart job, restart counter is at 1.
Feb 14 10:13:40 beaglebone systemd[1]: Stopped BB WL18xx wlan0 Service.
Feb 14 10:13:40 beaglebone systemd[1]: Starting BB WL18xx wlan0 Service...
Feb 14 10:13:40 beaglebone bb-wl18xx-wlan0[5178]: wlan_0_mac=[00:00:00:00:00:00], this forces internal wl18xx mac address...
Feb 14 10:13:40 beaglebone bb-wl18xx-wlan0[5178]: /lib/firmware/ti-connectivity/wl1271-nvs.bin=[00:00:00:00:00:00]
Feb 14 10:13:40 beaglebone bb-wl18xx-wlan0[5178]: wl18xx:wlan0:Power Management:off
Feb 14 10:13:41 beaglebone bb-wl18xx-wlan0[5178]: wl18xx:tether [iw phy phy0 interface add SoftAp0 type managed]
Feb 14 10:13:41 beaglebone bb-wl18xx-wlan0[5178]: wl18xx:tether [phy0 created]
Feb 14 10:13:41 beaglebone bb-wl18xx-wlan0[5178]: wl18xx:tether [ip link set dev SoftAp0 down]
Feb 14 10:13:41 beaglebone bb-wl18xx-wlan0[5178]: wl18xx:tether [ip link set dev SoftAp0 address B0:D5:CC:E5:E8:95]
Feb 14 10:13:41 beaglebone bb-wl18xx-wlan0[5178]: wl18xx:tether [ip link set dev SoftAp0 up]
Feb 14 10:13:41 beaglebone bb-wl18xx-wlan0[5178]: wl18xx:tether [ip addr add 192.168.8.1/24 broadcast 192.168.8.255 dev SoftAp0]
Feb 14 10:13:41 beaglebone bb-wl18xx-wlan0[5178]: wl18xx:tether [iptables -w -t nat -A POSTROUTING -o wlan0 -j MASQUERADE]
Feb 14 10:13:41 beaglebone bb-wl18xx-wlan0[5178]: wl18xx:tether [iptables -w -A FORWARD -i wlan0 -o SoftAp0 -m state --state RELATED,ESTABLISHED -j ACCEPT]
[... cut ...]
Feb 14 10:13:45 beaglebone bb-wl18xx-wlan0[5178]: See "systemctl status dnsmasq.service" and "journalctl -xe" for details.
Feb 14 10:13:45 beaglebone bb-wl18xx-wlan0[5178]: Configuration file: /tmp/hostapd-wl18xx.conf
Feb 14 10:13:45 beaglebone bb-wl18xx-wlan0[5178]: Using interface SoftAp0 with hwaddr b0:d5:cc:e5:e8:95 and ssid "BeagleBone-E895"
Feb 14 10:13:46 beaglebone bb-wl18xx-wlan0[5178]: SoftAp0: interface state UNINITIALIZED->ENABLED
Feb 14 10:13:46 beaglebone bb-wl18xx-wlan0[5178]: SoftAp0: AP-ENABLED
Feb 14 10:13:46 beaglebone systemd[1]: Started BB WL18xx wlan0 Service.
```

Regards,

RobertCNelson, 2020-03-03:
That doesn't look correct...
```
debian@beaglebone:~$ cat /etc/dnsmasq.d/SoftAp0
interface=SoftAp0
interface=usb0
port=53
dhcp-authoritative
domain-needed
bogus-priv
expand-hosts
cache-size=2048
dhcp-range=SoftAp0,192.168.8.50,192.168.8.150,10m
dhcp-range=usb0,.1,.1,2m
listen-address=127.0.0.1
listen-address=192.168.8.1
listen-address=.2
dhcp-option-force=interface:SoftAp0,option:dns-server,192.168.8.1
dhcp-option-force=interface:SoftAp0,option:mtu,1500
dhcp-option=usb0,3
dhcp-option=usb0,6
dhcp-leasefile=/var/run/dnsmasq.leases
```

RobertCNelson, 2020-03-03:
@GwenStouthuysen got it, the console images don't have git, so one file didn't get installed, a quick fix is to:

```
sudo cp -v /opt/scripts/boot/default/bb-boot /etc/default/
```
Then it'll look like:

```
debian@beaglebone:~$ cat /etc/dnsmasq.d/SoftAp0 
interface=SoftAp0
interface=usb0
interface=usb1
port=53
dhcp-authoritative
domain-needed
bogus-priv
expand-hosts
cache-size=2048
dhcp-range=SoftAp0,192.168.8.50,192.168.8.150,10m
dhcp-range=usb0,192.168.7.1,192.168.7.1,2m
dhcp-range=usb1,192.168.6.1,192.168.6.1,2m
listen-address=127.0.0.1
listen-address=192.168.8.1
listen-address=192.168.7.2
listen-address=192.168.6.2
dhcp-option-force=interface:SoftAp0,option:dns-server,192.168.8.1
dhcp-option-force=interface:SoftAp0,option:mtu,1500
dhcp-option=usb0,3
dhcp-option=usb0,6
dhcp-option=usb1,3
dhcp-option=usb1,6
dhcp-leasefile=/var/run/dnsmasq.leases
```

and we got the ip:

```
eth2: flags=-28605<UP,BROADCAST,RUNNING,MULTICAST,DYNAMIC>  mtu 1500
        inet 192.168.7.1  netmask 255.255.255.252  broadcast 192.168.7.3
        ether b0:d5:cc:e5:e8:95  txqueuelen 1000  (Ethernet)
        RX packets 17  bytes 2024 (1.9 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 50  bytes 12874 (12.5 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

eth3: flags=-28605<UP,BROADCAST,RUNNING,MULTICAST,DYNAMIC>  mtu 1500
        inet 192.168.6.1  netmask 255.255.255.252  broadcast 192.168.6.3
        inet6 fe80::b2d5:ccff:fee5:e897  prefixlen 64  scopeid 0x20<link>
        ether b0:d5:cc:e5:e8:97  txqueuelen 1000  (Ethernet)
        RX packets 19  bytes 2660 (2.5 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 85  bytes 18529 (18.0 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

Regards,

closed 2020-03-03 by RobertCNelson