beagleboard/Latest-Images #7: Beagle Bone Blue /dev/ttyO4 missing

Andrewiski, 2020-03-05:
Even if I load BB-UART4-00A0.dtbo  an ls /dev/ty*  does not return a ttyO4.

Aurdiopilot is looking for DSM on /dev/tty04 

Line 156  
https://github.com/ArduPilot/ardupilot/blob/05a0fe615b50fc2fa8dba11e541ba4f5740a1181/libraries/AP_HAL_Linux/HAL_Linux_Class.cpp

I saw a really old port from Jason and you https://github.com/ArduPilot/ardupilot/issues/6641
but in the post the dsm port is refered to as /dev/ttyS2 ????

Schematic for BeableBone blue show DSM as uart4 RX  so I assume its /dev/ttyO4 

Any ways what am I missing that is preventing /dev/ttyO4  from being an device?

 sudo /opt/scripts/tools/version.sh
git:/opt/scripts/:[8d8aa8dd017a0e9564a405f44e5590e47f7006b9]
eeprom:[A335BNLTBLA21708EL000063]
model:[TI_AM335x_BeagleBone_Blue]
dogtag:[BeagleBoard.org Debian Image 2019-12-01]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2019.04-00002-g3d8c979660]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-boneblue.dts]
UBOOT: Loaded Overlay:[AM335X-PRU-UIO-00A0]
UBOOT: Loaded Overlay:[BB-ADC-00A0]
kernel:[4.19.90-bone-rt-r42]
device-tree-override:[dtb=am335x-boneblue.dtb]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
uboot_overlay_options:[dtb_overlay=/lib/firmware/BB-I2C1-00A0.dtbo]
uboot_overlay_options:[dtb_overlay=/lib/firmware/BB-UART4-00A0.dtbo]
uboot_overlay_options:[dtb_overlay=/lib/firmware/BB-ADC-00A0.dtbo]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20200131.0-0rcnee0~stretch+20200131]
pkg:[bb-wl18xx-firmware]:[1.20190227.1-0rcnee0~stretch+20190227]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[librobotcontrol]:[1.0.4-git20190227.1-0rcnee0~stretch+20190327]
pkg:[firmware-ti-connectivity]:[20180825+dfsg-1rcnee1~stretch+20181217]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal netdev gpio pwm eqep remoteproc admin spi i2c tisdk weston-launch xenomai bluetooth cloud9ide]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 rng_core.default_quality=100 quiet]
dmesg | grep remote
[    0.987374] remoteproc remoteproc0: wkup_m3 is available
[    1.095633] remoteproc remoteproc0: powering up wkup_m3
[    1.095653] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[    1.097816] remoteproc remoteproc0: remote processor wkup_m3 is now up
dmesg | grep pru
dmesg | grep pinctrl-single
[    0.740494] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
END

RobertCNelson, 2020-03-05:
@Andrewiski , did 4.19.94-ti-rt-r36 solve the issue?  Please close this thread if it did, thanks!

Andrewiski, 2020-03-06:
Yes 4.19.94-ti-rt-r36 gives me /dev/tty04  and it does not seem to crash or slow down.

I do get the following error on start up that I did not get with 4.9

[    0.001551] timer_probe: no matching timers found
[    0.354897] l4_wkup_cm:clk:0010:0: failed to disable
[    1.119613] wkup_m3_ipc 44e11324.wkup_m3_ipc: could not get rproc handle
[    1.173572] pinctrl-single 44e10800.pinmux: pin PIN102 already requested by 48300180.eqep; cannot claim for ocp:P9_30_pinmux
[    1.184873] pinctrl-single 44e10800.pinmux: pin-102 (ocp:P9_30_pinmux) status -22
[    1.192402] pinctrl-single 44e10800.pinmux: could not request pin 102 (PIN102) from group pinmux_P9_30_spi_pin  on device pinctrl-single
[    1.204723] bone-pinmux-helper ocp:P9_30_pinmux: Error applying setting, reverse things back
[    1.527390] omap_voltage_late_init: Voltage driver support not added
[    1.534723] PM: Cannot get wkup_m3_ipc handle

closed 2020-03-06 by Andrewiski

epena29, 2020-07-18:
This Fix has become broke the beaglebone blue, not allowing the PRU1 to start because of a pin conflict. 

dmesg | grep pin output :>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>

 0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129412
[    0.082486] pinctrl core: initialized pinctrl subsystem
[    0.912507] pinctrl-single 44e10800.pinmux: 142 pins, size 568
[    1.114317] pinctrl-single 44e10800.pinmux: pin PIN102 already requested by 48300180.eqep; cannot claim for ocp:P9_30_pinmux
[    1.125716] pinctrl-single 44e10800.pinmux: pin-102 (ocp:P9_30_pinmux) status -22
[    1.133303] pinctrl-single 44e10800.pinmux: could not request pin 102 (PIN102) from group pinmux_P9_30_spi_pin  on device pinctrl-single
[    1.145754] bone-pinmux-helper ocp:P9_30_pinmux: Error applying setting, reverse things back
[    1.154368] bone-pinmux-helper: probe of ocp:P9_30_pinmux failed with error -22
[    1.384697] cpu cpu0: Dropping the link to regulator.4
>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>

output from version.sh:>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>

git:/opt/scripts/:[f495239e456d8ec99554a35ae04b33ddce5c26cc]
eeprom:[A335BNLTBLA21712EL004921]
model:[TI_AM335x_BeagleBone_Blue]
dogtag:[BeagleBoard.org Debian Buster IoT Image 2020-04-06]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2019.04-00002-g07d5700e21]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2019.04-00002-g07d5700e21]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-boneblue.dts]
UBOOT: Loaded Overlay:[AM335X-PRU-UIO-00A0]
UBOOT: Loaded Overlay:[BB-ADC-00A0]
kernel:[4.19.94-ti-r45]
nodejs:[v10.21.0]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20200709.0-0~buster+20200709]
pkg:[bb-wl18xx-firmware]:[1.20200702.0-0~buster+20200702]
pkg:[kmod]:[26-1]
pkg:[roboticscape]:[0.4.4-git20180608.0-0rcnee0~buster+20180609]:[GOT_REPLACED_BY_NEXT]
pkg:[librobotcontrol]:[1.0.5-git20200715.0-0~buster+20200716]
pkg:[firmware-ti-connectivity]:[20190717-2rcnee1~buster+20200305]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal bluetooth netdev i2c gpio pwm eqep remoteproc admin spi iio docker tisdk weston-launch xenomai cloud9ide]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet]
dmesg | grep remote
[   43.362822] remoteproc remoteproc0: wkup_m3 is available
[   43.960187] remoteproc remoteproc0: powering up wkup_m3
[   43.960222] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[   43.960516] remoteproc remoteproc0: remote processor wkup_m3 is now up
dmesg | grep pru
dmesg | grep pinctrl-single
[    0.912507] pinctrl-single 44e10800.pinmux: 142 pins, size 568
[    1.114317] pinctrl-single 44e10800.pinmux: pin PIN102 already requested by 48300180.eqep; cannot claim for ocp:P9_30_pinmux
[    1.125716] pinctrl-single 44e10800.pinmux: pin-102 (ocp:P9_30_pinmux) status -22
[    1.133303] pinctrl-single 44e10800.pinmux: could not request pin 102 (PIN102) from group pinmux_P9_30_spi_pin  on device pinctrl-single
dmesg | grep gpio-of-helper
lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END

>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>