beagleboard/Latest-Images #9: dnsmasq configuration not working for USB

MartinVoelz, 2020-03-09:
Hello,
I am using the console image from 01.03.2020  for 4.14.108-ti-r127.

I found that the beaglebone is not providing automatically the IP for the Windows RNDIS.

In the file   /usr/bin/bb_dnsmasq_config.sh   you have a wrong path to config file
should be: /opt/scripts/boot/default/bb_boot

Thus, the "SUBNET_USB0"  info is missing and the echo into SoftAP0 will be  .1 .1  instead of the full IP.

Greatings Martin Voelz

RobertCNelson, 2020-03-09:
@MartinVoelz should have been fixed via: https://github.com/RobertCNelson/omap-image-builder/commit/d1a58cc9b52d2755bc3e5966b9e5988080986053

Duplicate: https://github.com/beagleboard/Latest-Images/issues/5

Fix: ```sudo cp -v /opt/scripts/boot/default/bb-boot /etc/default/```

Regards,

MartinVoelz, 2020-03-09:
ok, the image I used is two days before your change.
It will work for me, but is it wise to simply copy it to a second location?
I would be afraid that you will get complaints like:
"I had changed the bb_boot file but it has no effect" 
when they were changing in the /opt/scripts/boot/default/bb-boot

RobertCNelson, 2020-03-09:
@MartinVoelz , it's been "/etc/default/bb-boot" since Dec 15, 2017..

The actual is bug is because i didn't add a backup config, when i ripped out usb0/usb1 configuration to it's own file, and didn't add teh back files into the else case here:

https://github.com/rcn-ee/repos/blob/master/bb-usb-gadgets/suite/stretch/debian/bb_dnsmasq_config.sh#L36-L38

Regards,

RobertCNelson, 2020-03-09:
@MartinVoelz, i pushed a failsafe: https://github.com/rcn-ee/customizations/commit/776c60d484dd8bcb3380fd53e7d4b873141cfbf8

closed 2020-03-09 by RobertCNelson