beagleboard/Latest-Images #13: No bluetooth in console image

JoseGoncalves, 2020-03-16:
**Describe the bug**
Bluetoth service fails to start.
`systemctl status` reports a `degraded` state.
`systemctl list-units --failed` reports:
```
  UNIT                        LOAD   ACTIVE SUB    DESCRIPTION                
● bb-wl18xx-bluetooth.service loaded failed failed BB WL18xx Bluetooth Service
```

**Describe which image you are running**
bone-debian-10.3-console-armhf-2020-03-16-1gb.img.xz

**Describe your hardware**
BeagleBone Green Wireless

**Describe any Capes Installed**
None

**Output of `/opt/scripts/tools/version.sh`**
```
eeprom:[A335BNLTGW1ABBGW18042503]
model:[TI_AM335x_BeagleBone_Green_Wireless]
dogtag:[BeagleBoard.org Debian Console Image 2020-03-16]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2019.04-00002-g07d5700e21]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-bonegreen-wireless-uboot-univ.dts]
UBOOT: Loaded Overlay:[AM335X-PRU-RPROC-4-19-TI-00A0]
UBOOT: Loaded Overlay:[BB-ADC-00A0]
UBOOT: Loaded Overlay:[BB-BBGW-WL1835-00A0]
UBOOT: Loaded Overlay:[BB-BONE-eMMC1-01-00A0]
kernel:[4.19.94-ti-r37]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-19-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20200312.0-0rcnee0~buster+20200312]
pkg:[bb-wl18xx-firmware]:[1.20200226.1-0rcnee0~buster+20200226]
pkg:[kmod]:[26-1]
WARNING:pkg:[librobotcontrol]:[NOT_INSTALLED]
pkg:[firmware-ti-connectivity]:[20190717-2rcnee1~buster+20200305]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal netdev gpio pwm eqep remoteproc admin spi iio i2c docker tisdk weston-launch xenomai bluetooth cloud9ide]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 rng_core.default_quality=100 quiet]
dmesg | grep remote
[    1.548058] remoteproc remoteproc0: wkup_m3 is available
[    1.550064] remoteproc remoteproc1: 4a334000.pru is available
[    1.550562] remoteproc remoteproc2: 4a338000.pru is available
[    1.717622] remoteproc remoteproc0: powering up wkup_m3
[    1.717741] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[    1.721414] remoteproc remoteproc0: remote processor wkup_m3 is now up
dmesg | grep pru
[    1.550064] remoteproc remoteproc1: 4a334000.pru is available
[    1.550176] pru-rproc 4a334000.pru: PRU rproc node pru@4a334000 probed successfully
[    1.550562] remoteproc remoteproc2: 4a338000.pru is available
[    1.550718] pru-rproc 4a338000.pru: PRU rproc node pru@4a338000 probed successfully
dmesg | grep pinctrl-single
[    1.072213] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
[    1.084583] gpio-of-helper ocp:cape-universal: ready
lsusb
Bus 001 Device 002: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END
```

**To Reproduce**
Boot the board

**Expected behavior**
Bluetooth services start and bluetooth LED is ON.

**Additional context**
I've fixed this by installing `bluez` package.
I think that `bluez` package should be installed by default in this image.

RobertCNelson, 2020-03-16:
```
debian@beaglebone:~$ df -h | grep mmc
/dev/mmcblk0p1  866M  458M  348M  57% /
debian@beaglebone:~$ sudo apt install bluez
Reading package lists... Done
Building dependency tree       
Reading state information... Done
The following additional packages will be installed:
  libasound2 libasound2-data libdw1
Suggested packages:
  pulseaudio-module-bluetooth libasound2-plugins alsa-utils
The following NEW packages will be installed:
  bluez libasound2 libasound2-data libdw1
0 upgraded, 4 newly installed, 0 to remove and 0 not upgraded.
Need to get 1450 kB of archives.
After this operation, 5244 kB of additional disk space will be used.
Do you want to continue? [Y/n]
```

RobertCNelson, 2020-03-16:
@JoseGoncalves , yeah i'm fine with this, it only add 5MB, thanks for the issue!

closed 2020-03-16 by RobertCNelson