beagleboard/Latest-Images #14: Ping does not work

GwenStouthuysen, 2020-03-17:
Using a Beaglebone black wireless.
Downloaded the eeprom programming version
connected to Wifi using connmanctl
try the internet connection using ping 
```
debian@beaglebone:~$ ping google.com
PING google.com(ams16s32-in-x0e.1e100.net (2a00:1450:400e:80c::200e)) 56 data bytes
^C
--- google.com ping statistics ---
99 packets transmitted, 0 received, 100% packet loss, time 430ms
```
The network is up and functions as the name is resolved.
apt-get update is fully functional

RobertCNelson, 2020-03-17:
@GwenStouthuysen which image?

RobertCNelson, 2020-03-17:
Works with 2020-03-16 (stretch-iot)
```
Connected wifi_8030dc306bc9_436f6e74656e745f446576_managed_psk
connmanctl> quit
debian@beaglebone:~$ ping google.com
PING google.com (172.217.4.46) 56(84) bytes of data.
64 bytes from lga15s46-in-f46.1e100.net (172.217.4.46): icmp_seq=1 ttl=55 time=26.0 ms
64 bytes from lga15s46-in-f46.1e100.net (172.217.4.46): icmp_seq=2 ttl=55 time=38.8 ms
^C
--- google.com ping statistics ---
2 packets transmitted, 2 received, 0% packet loss, time 1001ms
rtt min/avg/max/mdev = 26.061/32.435/38.809/6.374 ms
```

```
debian@beaglebone:~$ sudo /opt/scripts/tools/version.sh 
[sudo] password for debian: 
git:/opt/scripts/:[5fa30ffcf068996ec08e034d2af5c5a133f10709]
eeprom:[A335BNLTGW1ABBGW16053764]
model:[TI_AM335x_BeagleBone_Green_Wireless]
dogtag:[BeagleBoard.org Debian Image 2020-03-16]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2019.04-00002-g07d5700e21]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-bonegreen-wireless-uboot-univ.dts]
UBOOT: Loaded Overlay:[AM335X-PRU-RPROC-4-14-TI-00A0]
UBOOT: Loaded Overlay:[BB-ADC-00A0]
UBOOT: Loaded Overlay:[BB-BBGW-WL1835-00A0]
UBOOT: Loaded Overlay:[BB-BONE-eMMC1-01-00A0]
kernel:[4.14.108-ti-r127]
nodejs:[v6.17.0]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20200312.0-0rcnee0~stretch+20200312]
pkg:[bb-wl18xx-firmware]:[1.20200226.1-0rcnee0~stretch+20200226]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[librobotcontrol]:[1.0.4-git20190227.1-0rcnee0~stretch+20190327]
pkg:[firmware-ti-connectivity]:[20190717-2rcnee1~stretch+20200305]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev gpio pwm eqep remoteproc admin spi iio docker tisdk weston-launch xenomai cloud9ide]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 rng_core.default_quality=100 quiet]
dmesg | grep remote
[    1.226362] remoteproc remoteproc0: wkup_m3 is available
[    1.310946] remoteproc remoteproc0: powering up wkup_m3
[    1.311060] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[    1.315403] remoteproc remoteproc0: remote processor wkup_m3 is now up
[   12.496097] remoteproc remoteproc1: 4a334000.pru is available
[   12.503770] remoteproc remoteproc2: 4a338000.pru is available
dmesg | grep pru
[   12.464252] pruss 4a300000.pruss: creating PRU cores and other child platform devices
[   12.496097] remoteproc remoteproc1: 4a334000.pru is available
[   12.496243] pru-rproc 4a334000.pru: PRU rproc node /ocp/pruss_soc_bus@4a326004/pruss@0/pru@34000 probed successfully
[   12.503770] remoteproc remoteproc2: 4a338000.pru is available
[   12.503895] pru-rproc 4a338000.pru: PRU rproc node /ocp/pruss_soc_bus@4a326004/pruss@0/pru@38000 probed successfully
dmesg | grep pinctrl-single
[    0.854591] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    0.866029] gpio-of-helper ocp:cape-universal: ready
lsusb
Bus 001 Device 002: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END
```

GwenStouthuysen, 2020-03-17:
Debian 10 iot eeprom writer

Gwen

RobertCNelson, 2020-03-17:
and works with 2020-03-16 (buster-iot) bone-debian-10.3-iot-armhf-2020-03-16-4gb.img.xz
```
Connected wifi_8030dc306bc9_436f6e74656e745f446576_managed_psk
connmanctl> quit
debian@beaglebone:~$ ping google.com
PING google.com (172.217.4.46) 56(84) bytes of data.
64 bytes from lga15s46-in-f46.1e100.net (172.217.4.46): icmp_seq=1 ttl=55 time=30.8 ms
64 bytes from lga15s46-in-f46.1e100.net (172.217.4.46): icmp_seq=2 ttl=55 time=41.6 ms
64 bytes from lga15s46-in-f46.1e100.net (172.217.4.46): icmp_seq=3 ttl=55 time=34.2 ms
^C
--- google.com ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 5ms
rtt min/avg/max/mdev = 30.842/35.529/41.575/4.488 ms
```

```
debian@beaglebone:~$ sudo /opt/scripts/tools/version.sh 
git:/opt/scripts/:[5fa30ffcf068996ec08e034d2af5c5a133f10709]
eeprom:[A335BNLTGW1ABBGW16053764]
model:[TI_AM335x_BeagleBone_Green_Wireless]
dogtag:[BeagleBoard.org Debian IoT Image 2020-03-16]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2019.04-00002-g07d5700e21]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-bonegreen-wireless-uboot-univ.dts]
UBOOT: Loaded Overlay:[AM335X-PRU-RPROC-4-19-TI-00A0]
UBOOT: Loaded Overlay:[BB-ADC-00A0]
UBOOT: Loaded Overlay:[BB-BBGW-WL1835-00A0]
UBOOT: Loaded Overlay:[BB-BONE-eMMC1-01-00A0]
kernel:[4.19.94-ti-r37]
nodejs:[v10.15.2]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-19-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20200312.0-0rcnee0~buster+20200312]
pkg:[bb-wl18xx-firmware]:[1.20200226.1-0rcnee0~buster+20200226]
pkg:[kmod]:[26-1]
pkg:[librobotcontrol]:[1.0.4-git20190227.1-0rcnee0~buster+20190327]
pkg:[firmware-ti-connectivity]:[20190717-2rcnee1~buster+20200305]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal bluetooth netdev i2c gpio pwm eqep remoteproc admin spi iio docker tisdk weston-launch xenomai cloud9ide]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 rng_core.default_quality=100 quiet]
dmesg | grep remote
[    1.546987] remoteproc remoteproc0: wkup_m3 is available
[    1.549474] remoteproc remoteproc1: 4a334000.pru is available
[    1.550041] remoteproc remoteproc2: 4a338000.pru is available
[    1.717640] remoteproc remoteproc0: powering up wkup_m3
[    1.717759] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[    1.721393] remoteproc remoteproc0: remote processor wkup_m3 is now up
dmesg | grep pru
[    1.549474] remoteproc remoteproc1: 4a334000.pru is available
[    1.549594] pru-rproc 4a334000.pru: PRU rproc node pru@4a334000 probed successfully
[    1.550041] remoteproc remoteproc2: 4a338000.pru is available
[    1.550144] pru-rproc 4a338000.pru: PRU rproc node pru@4a338000 probed successfully
dmesg | grep pinctrl-single
[    1.071202] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
[    1.083563] gpio-of-helper ocp:cape-universal: ready
lsusb
Bus 001 Device 002: ID 05e3:0610 Genesys Logic, Inc. 4-port hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END
```

RobertCNelson, 2020-03-17:
@GwenStouthuysen please confirm the "exact" version you flashed..

I've confirmed both bone-debian-9.12-iot-armhf-2020-03-16-4gb.img.xz and bone-debian-10.3-iot-armhf-2020-03-16-4gb.img.xz "ping" work on a BeagleBone Green Wireless (using the wifi connection) to ping google.com with the default user (debian)

Sorry, without more information I will have to mark this as "fixed" as it works for me, on current test images.

Regards,

GwenStouthuysen, 2020-03-17:
bone-eMMC-flasher-debian-10.3-console-armhf-2020-03-16-1gb.img.xz

On beaglebone black wireless

RobertCNelson, 2020-03-17:
okay, i'll test it tomorrow..

RobertCNelson, 2020-03-18:
OKay, flashed bone-eMMC-flasher-debian-10.3-console-armhf-2020-03-16-1gb.img.xz to a BeagleBone Green Wireless this morning.. (Using WiFi access and default user debian)..

ping works fine

```
Connected wifi_8030dc306bc9_436f6e74656e745f446576_managed_psk
connmanctl> quit
debian@beaglebone:~$ ping google.com
PING google.com (172.217.6.110) 56(84) bytes of data.
64 bytes from ord37s03-in-f14.1e100.net (172.217.6.110): icmp_seq=1 ttl=55 time=65.1 ms
64 bytes from ord37s03-in-f14.1e100.net (172.217.6.110): icmp_seq=2 ttl=55 time=38.6 ms
64 bytes from ord37s03-in-f14.1e100.net (172.217.6.110): icmp_seq=3 ttl=55 time=31.3 ms
^C
--- google.com ping statistics ---
3 packets transmitted, 3 received, 0% packet loss, time 5ms
rtt min/avg/max/mdev = 31.348/45.035/65.136/14.522 ms
```

```
debian@beaglebone:~$ sudo /opt/scripts/tools/version.sh 
eeprom:[A335BNLTGW1ABBGW16053764]
model:[TI_AM335x_BeagleBone_Green_Wireless]
dogtag:[BeagleBoard.org Debian Console Image 2020-03-16]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2019.04-00002-g07d5700e21]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-bonegreen-wireless-uboot-univ.dts]
UBOOT: Loaded Overlay:[AM335X-PRU-RPROC-4-19-TI-00A0]
UBOOT: Loaded Overlay:[BB-ADC-00A0]
UBOOT: Loaded Overlay:[BB-BBGW-WL1835-00A0]
UBOOT: Loaded Overlay:[BB-BONE-eMMC1-01-00A0]
kernel:[4.19.94-ti-r37]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-19-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20200312.0-0rcnee0~buster+20200312]
pkg:[bb-wl18xx-firmware]:[1.20200226.1-0rcnee0~buster+20200226]
pkg:[kmod]:[26-1]
WARNING:pkg:[librobotcontrol]:[NOT_INSTALLED]
pkg:[firmware-ti-connectivity]:[20190717-2rcnee1~buster+20200305]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal netdev gpio pwm eqep remoteproc admin spi iio i2c docker tisdk weston-launch xenomai bluetooth cloud9ide]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 rng_core.default_quality=100 quiet]
dmesg | grep remote
[    1.548008] remoteproc remoteproc0: wkup_m3 is available
[    1.550031] remoteproc remoteproc1: 4a334000.pru is available
[    1.550534] remoteproc remoteproc2: 4a338000.pru is available
[    1.717635] remoteproc remoteproc0: powering up wkup_m3
[    1.717752] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[    1.721421] remoteproc remoteproc0: remote processor wkup_m3 is now up
dmesg | grep pru
[    1.550031] remoteproc remoteproc1: 4a334000.pru is available
[    1.550145] pru-rproc 4a334000.pru: PRU rproc node pru@4a334000 probed successfully
[    1.550534] remoteproc remoteproc2: 4a338000.pru is available
[    1.550633] pru-rproc 4a338000.pru: PRU rproc node pru@4a338000 probed successfully
dmesg | grep pinctrl-single
[    1.072432] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
[    1.085044] gpio-of-helper ocp:cape-universal: ready
END
```

```
debian@beaglebone:~$ sudo route
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
default         192.168.200.1   0.0.0.0         UG    0      0        0 wlan0
dns.google      192.168.200.1   255.255.255.255 UGH   0      0        0 wlan0
Edelta.DADNS.am 192.168.200.1   255.255.255.255 UGH   0      0        0 wlan0
192.168.6.0     0.0.0.0         255.255.255.0   U     0      0        0 usb1
192.168.7.0     0.0.0.0         255.255.255.0   U     0      0        0 usb0
192.168.8.0     0.0.0.0         255.255.255.0   U     0      0        0 SoftAp0
192.168.200.0   0.0.0.0         255.255.255.0   U     0      0        0 wlan0
192.168.200.1   0.0.0.0         255.255.255.255 UH    0      0        0 wlan0
```

Closing un-reproducable, probally end users network..

Regards,

closed 2020-03-18 by RobertCNelson

GwenStouthuysen, 2020-03-18:
Just tested ping -4 google.com

And that works, you might be right: there might be an issue with IPv6

Coupled my board to a different AP, of the same network: problem solved

Regards,

Gwen