beagleboard/Latest-Images #15: Cloud9 -> Something..

RobertCNelson, 2020-03-17:
https://theia-ide.org/ ?

MarkAYoder, 2020-03-17:
Looks interesting....

RobertCNelson, 2020-03-17:
[image]

Default install...

Korusuke, 2020-06-17:
https://github.com/cdr/code-server may be a good option too.

I am not sure how its performance would stack up against cloud9 or theia.
This would also enable us to connect and use vscode over ssh using [this](https://github.com/cdr/sshcode)

The major advantage I see as a vscode user for using theia/vscode over cloud9 is all the available extensions.

RobertCNelson, 2021-02-10:
random hacks:

```
https://github.com/cdr/code-server.git

sudo apt update 
sudo apt install yarnpkg curl gnupg nodejs build-essential libsecret-1-dev libx11-dev libxkbfile-dev gettext-base git rsync unzip jq shellcheck golang gulp

GO111MODULE=on go get mvdan.cc/sh/v3/cmd/shfmt
sudo cp -v ./go/bin/shfmt /usr/local/bin/

sudo ln -s /usr/bin/yarnpkg /usr/bin/yarn

sudo napm install -g yarn

git clone -b v3.8.1 https://github.com/cdr/code-server.git --depth=1
cd ./code-server/

yarn
yarn build
yarn build:vscode
yarn release

yarn release:standalone
yarn test:standalone-release
yarn package
```

RobertCNelson, 2021-02-11:
Okay, i got a hacked version working...

@MarkAYoder and @jadonk 

In buster...

```
sudo apt update
sudo apt install bb-code-server
```

Then just run as:

```
/opt/code-server/app/bin/code-server --auth none --bind-addr 192.168.1.34:3000
```

For extensions:

formulahendry.code-runner
eamodio.gitlens

Regards,

MarkAYoder, 2021-02-12:
@RobertCNelson I got code-server running on the Bone.  It looks interesting.  Are there any tutorials on how to use it  (rather than install it) that you'd recommend?

RobertCNelson, 2021-02-13:
@MarkAYoder honestly, i'm slowly figuring it out.. But yeah, totally new..

Under extensions, grab "formulahendry.code-runner"... i need to pre-isntall this..  after installing it, ctrl-f5, and you should get an arrow in the upper right.. just like cloud9's runner..

I"m going to push a socket activation option, it'll probably break everything.. so you might have to disable it..

https://github.com/rcn-ee/repos/blob/master/bb-code-server/suite/buster/debian/bb-code-server.service

Regards,

MarkAYoder, 2021-02-13:
@RobertCNelson This looks like a good place to start for VS in general:  https://code.visualstudio.com/docs/getstarted/introvideos

Let me know when you have a new version for me to try.

RobertCNelson, 2021-03-08:
v3.9.1 added compression...  ;)

```
voodoo@x6-am57xx-beagle-x15-2gb:~$ du -sh /opt/bb-code-server/
298M	/opt/bb-code-server/
```