beagleboard/Latest-Images #17: Disabling AP mode looses USB netwok connectivity

JoseGoncalves, 2020-03-22:
**Describe the bug**
If I disable the Wi-Fi AP mode service and afterwards I reboot my board, the virtual network interfaces on the USB connection are not properly setup (no IP addresses are set) and I'm unable to connect to the board via SSH.

**Describe which image you are running**
bone-debian-10.3-console-armhf-2020-03-19-1gb.img.xz

**Describe your hardware**
BeagleBone Green Wireless

**Describe any Capes Installed**
None

**To Reproduce**
Steps to reproduce the behavior:
1. $ sudo systemctl disable bb-wl18xx-wlan0.service
2. $ sudo reboot
3. The board startups without USB network interfaces properly setup on the development PC

**Expected behavior**
The board should boot up with the USB network interfaces properly setup.

JoseGoncalves, 2020-03-22:
I've mounted the microSD card in a PC after the failed boot, to check the syslog contents. Here it goes (only the part regarding the boot after the AP service is disabled):
```
Feb 14 10:12:04 beaglebone systemd-fsck[831]: rootfs: clean, 18885/57344 files, 108612/229376 blocks
Feb 14 10:12:04 beaglebone systemd[1]: Starting Flush Journal to Persistent Storage...
Feb 14 10:12:04 beaglebone systemd[1]: Started Create Static Device Nodes in /dev.
Feb 14 10:12:04 beaglebone systemd[1]: Starting udev Kernel Device Manager...
Feb 14 10:12:04 beaglebone systemd[1]: Reached target Local File Systems (Pre).
Feb 14 10:12:04 beaglebone systemd[1]: Reached target Local File Systems.
Feb 14 10:12:04 beaglebone systemd[1]: Condition check resulted in Commit a transient machine-id on disk being skipped.
Feb 14 10:12:04 beaglebone systemd-udevd[850]: Network interface NamePolicy= disabled on kernel command line, ignoring.
Feb 14 10:12:04 beaglebone systemd[1]: Started Flush Journal to Persistent Storage.
Feb 14 10:12:04 beaglebone systemd[1]: Starting Create Volatile Files and Directories...
Feb 14 10:12:04 beaglebone systemd[1]: Started udev Kernel Device Manager.
Feb 14 10:12:04 beaglebone systemd[1]: Started Create Volatile Files and Directories.
Feb 14 10:12:04 beaglebone systemd[1]: Starting Update UTMP about System Boot/Shutdown...
Feb 14 10:12:04 beaglebone systemd[1]: Started Entropy daemon using the HAVEGE algorithm.
Feb 14 10:12:04 beaglebone systemd[1]: Starting Network Time Synchronization...
Feb 14 10:12:04 beaglebone systemd[1]: Started Update UTMP about System Boot/Shutdown.
Feb 14 10:12:04 beaglebone systemd[1]: Started udev Coldplug all Devices.
Feb 14 10:12:04 beaglebone systemd[1]: Starting Helper to synchronize boot up for ifupdown...
Feb 14 10:12:04 beaglebone systemd[1]: Started Helper to synchronize boot up for ifupdown.
Feb 14 10:12:04 beaglebone systemd[1]: Starting Raise network interfaces...
Feb 14 10:12:04 beaglebone systemd[1]: Started Network Time Synchronization.
Feb 14 10:12:04 beaglebone systemd[1]: Reached target System Initialization.
Feb 14 10:12:04 beaglebone systemd[1]: Listening on D-Bus System Message Bus Socket.
Feb 14 10:12:04 beaglebone systemd[1]: Reached target Sockets.
Feb 14 10:12:04 beaglebone systemd[1]: Reached target Basic System.
Feb 14 10:12:04 beaglebone systemd[1]: Starting LSB: Load kernel modules needed to enable cpufreq scaling...
Feb 14 10:12:04 beaglebone systemd[1]: Started D-Bus System Message Bus.
Feb 14 10:12:04 beaglebone systemd[1]: Starting Connection service...
Feb 14 10:12:04 beaglebone systemd[1]: Starting System Logging Service...
Feb 14 10:12:04 beaglebone systemd[1]: Started Daily Cleanup of Temporary Directories.
Feb 14 10:12:04 beaglebone systemd[1]: Starting Login Service...
Feb 14 10:12:04 beaglebone systemd[1]: Started Regular background program processing daemon.
Feb 14 10:12:04 beaglebone systemd[1]: Condition check resulted in getty on tty2-tty6 if dbus and logind are not available being skipped.
Feb 14 10:12:04 beaglebone systemd[1]: Starting WPA supplicant...
Feb 14 10:12:04 beaglebone systemd[1]: Reached target System Time Synchronized.
Feb 14 10:12:04 beaglebone kernel: [    0.000000] Booting Linux on physical CPU 0x0
Feb 14 10:12:04 beaglebone kernel: [    0.000000] Linux version 4.19.94-ti-r39 (voodoo@w8-imx6q-wandboard-2gb) (gcc version 8.3.0 (Debian 8.3.0-6)) #1buster SMP PREEMPT Wed Mar 18 21:31:04 UTC 2020
Feb 14 10:12:04 beaglebone kernel: [    0.000000] CPU: ARMv7 Processor [413fc082] revision 2 (ARMv7), cr=10c5387d
Feb 14 10:12:04 beaglebone kernel: [    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
Feb 14 10:12:04 beaglebone kernel: [    0.000000] OF: fdt: Machine model: TI AM335x BeagleBone Green Wireless
Feb 14 10:12:04 beaglebone kernel: [    0.000000] Memory policy: Data cache writeback
Feb 14 10:12:04 beaglebone kernel: [    0.000000] cma: Reserved 48 MiB at 0x9c800000
Feb 14 10:12:04 beaglebone kernel: [    0.000000] On node 0 totalpages: 130560
Feb 14 10:12:04 beaglebone kernel: [    0.000000]   Normal zone: 1148 pages used for memmap
Feb 14 10:12:04 beaglebone kernel: [    0.000000]   Normal zone: 0 pages reserved
Feb 14 10:12:04 beaglebone kernel: [    0.000000]   Normal zone: 130560 pages, LIFO batch:31
Feb 14 10:12:04 beaglebone kernel: [    0.000000] CPU: All CPU(s) started in SVC mode.
Feb 14 10:12:04 beaglebone kernel: [    0.000000] AM335X ES2.1 (sgx neon)
Feb 14 10:12:04 beaglebone kernel: [    0.000000] random: get_random_bytes called from start_kernel+0xac/0x4b4 with crng_init=0
Feb 14 10:12:04 beaglebone kernel: [    0.000000] percpu: Embedded 17 pages/cpu s38988 r8192 d22452 u69632
Feb 14 10:12:04 beaglebone kernel: [    0.000000] pcpu-alloc: s38988 r8192 d22452 u69632 alloc=17*4096
Feb 14 10:12:04 beaglebone kernel: [    0.000000] pcpu-alloc: [0] 0 
Feb 14 10:12:04 beaglebone kernel: [    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129412
Feb 14 10:12:04 beaglebone kernel: [    0.000000] Kernel command line: console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet
Feb 14 10:12:04 beaglebone kernel: [    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
Feb 14 10:12:04 beaglebone kernel: [    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[... cut ...]
Feb 14 10:14:59 beaglebone systemd[1]: Stopped Daily rotation of log files.
Feb 14 10:14:59 beaglebone systemd[1]: Stopped target System Time Synchronized.
Feb 14 10:14:59 beaglebone systemd[1]: Stopped target Graphical Interface.
Feb 14 10:14:59 beaglebone systemd[1]: Stopped target Multi-User System.
Feb 14 10:14:59 beaglebone systemd[1]: Stopping Login Service...
Feb 14 10:14:59 beaglebone kernel: [  191.464419] Bluetooth: hci0: sending frame failed (-49)
Feb 14 10:14:59 beaglebone systemd[1]: Stopping BB WL18xx Bluetooth Service...
```

RobertCNelson, 2020-03-22:
@JoseGoncalves yeah... that would happen if you disable bb-wl18xx-wlan0.service...

usb0/usb1 and SoftAp0 have a shared requirement, dnsmasq, which get's set here:

https://github.com/beagleboard/customizations/blob/master/usr/bin/bb_dnsmasq_config.sh

On a normal bootup, generic-board-startup.service runs:

https://github.com/beagleboard/customizations/blob/master/lib/systemd/system/generic-board-startup.service

Which calls:

https://github.com/RobertCNelson/boot-scripts/blob/master/boot/generic-startup.sh

Then calls:

https://github.com/RobertCNelson/boot-scripts/blob/master/boot/am335x_evm.sh

The board gets detected:

https://github.com/RobertCNelson/boot-scripts/blob/master/boot/am335x_evm.sh#L124-L199

For non "wifi" dnsmasq_usb0_usb1 get's set which then calls:

/usr/bin/bb_dnsmasq_config.sh directly

https://github.com/RobertCNelson/boot-scripts/blob/master/boot/am335x_evm.sh#L729-L731

For the boards with WiFi, wl18xx module can take forever to load, so am335x_evm.sh does not call /usr/bin/bb_dnsmasq_config.sh and instead that is left for bb-wl18xx-wlan0.service to handle

Regards,

So yeah... How do YOU want to fix this?

RobertCNelson, 2020-03-22:
@JoseGoncalves the current theory, uses would disable "TETHER_ENABLED=yes" in /etc/default/bb-wl18xx

RobertCNelson, 2020-03-22:
(wrong button for comment)...
https://github.com/rcn-ee/repos/blob/master/bb-wl18xx-firmware/suite/stretch/debian/postinst#L19-L55

RobertCNelson, 2020-03-22:
Yeah, this is already implemented..

```
sudo sed 's:TETHER_ENABLED=yes:TETHER_ENABLED=no:g' /etc/default/bb-wl18xx
```
Regards,

RobertCNelson, 2020-03-22:
NO that's not entirerly true, when you disable tether..

https://github.com/rcn-ee/repos/blob/master/bb-wl18xx-firmware/suite/stretch/debian/bb-wl18xx-wlan0#L193-L195

/usr/bin/bb-wl18xx-tether doesn't get called which fire up dnsmasq setting..

```
	if [ "x${TETHER_ENABLED}" = "xyes" ] ; then
		/usr/bin/bb-wl18xx-tether || true
	fi
```

```
	if [ "x${TETHER_ENABLED}" = "xyes" ] ; then
		/usr/bin/bb-wl18xx-tether || true
	else
		/usr/bin/bb_dnsmasq_config.sh || true
	fi
```

then from here:

https://github.com/beagleboard/customizations/blob/master/usr/bin/bb_dnsmasq_config.sh#L32-L34

read the TETHER_ENABLED variable..

Regards,

JoseGoncalves, 2020-03-22:
Hit @RobertCNelson. I will try to see if I can find a way to get USB interfaces up without depending on the `bb-wl18xx-wlan0.service`. 

My ultimate goal is to setup a wireless mesh network with BeagleBones, and, for that, I need to previously shut down the existing wlan0 and SoftAp0 interfaces.

I think I have already tried set the config `TETHER_ENABLED=no`with the same results, i.e., the USB network interfaces are not started properly (but I need to recheck that).

RobertCNelson, 2020-03-22:
@JoseGoncalves , we cross posts, i think the fix in 

```
	if [ "x${TETHER_ENABLED}" = "xyes" ] ; then
		/usr/bin/bb-wl18xx-tether || true
	else
		/usr/bin/bb_dnsmasq_config.sh || true
	fi
```
in /usr/bin/bb-wl18xx-wlan0

and then patch /usr/bin/bb_dnsmasq_config.sh to read /etc/default/bb-wl18xx's TETHER_ENABLED=no would allow you do do that..

The main point of "/usr/bin/bb-wl18xx-wlan0" is, TI's firmware/driver sucks, set the MAC address..  Then call teh rest of the stack..

Regards,

RobertCNelson, 2020-03-22:
https://github.com/beagleboard/customizations/blob/master/usr/bin/bb_dnsmasq_config.sh#L52-L55

```
unset got_softap0
if [ -d /sys/class/net/${ap_interface} ] ; then
	got_softapt0=enable
fi
```
```
unset got_softap0
if [ "x${TETHER_ENABLED}" = "xyes" ] ; then
if [ -d /sys/class/net/${ap_interface} ] ; then
	got_softapt0=enable
fi
fi
```

yeah, those 2 changes...

RobertCNelson, 2020-03-22:
@JoseGoncalves okay that works: SoftAp0 is gone..

```
debian@beaglebone:~$ sudo ifconfig -a
[sudo] password for debian: 
can0: flags=128<NOARP>  mtu 16
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 10  (UNSPEC)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 43  

can1: flags=128<NOARP>  mtu 16
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 10  (UNSPEC)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        device interrupt 44  

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 480  bytes 34400 (33.5 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 480  bytes 34400 (33.5 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

usb0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.7.2  netmask 255.255.255.252  broadcast 192.168.7.3
        inet6 fe80::f65e:abff:fe6e:1972  prefixlen 64  scopeid 0x20<link>
        ether f4:5e:ab:6e:19:72  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 41  bytes 9346 (9.1 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

usb1: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.6.2  netmask 255.255.255.252  broadcast 192.168.6.3
        inet6 fe80::f65e:abff:fe6e:1976  prefixlen 64  scopeid 0x20<link>
        ether f4:5e:ab:6e:19:76  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 43  bytes 7646 (7.4 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=-28605<UP,BROADCAST,RUNNING,MULTICAST,DYNAMIC>  mtu 1500
        inet 192.168.1.210  netmask 255.255.252.0  broadcast 192.168.3.255
        inet6 fe80::6ec3:74ff:fee6:1225  prefixlen 64  scopeid 0x20<link>
        ether 6c:c3:74:e6:12:25  txqueuelen 1000  (Ethernet)
        RX packets 70  bytes 7557 (7.3 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 73  bytes 12687 (12.3 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

and i cna login:

```
voodoo@hestia:/opt/git_repo/debian$ ssh debian@192.168.7.2
The authenticity of host '192.168.7.2 (192.168.7.2)' can't be established.
ECDSA key fingerprint is SHA256:hrg7KQ8N7V+8l2MF6D/VnIVhX5Eg3WB/qZjm5qx5fdo.
Are you sure you want to continue connecting (yes/no/[fingerprint])? yes
Warning: Permanently added '192.168.7.2' (ECDSA) to the list of known hosts.
Debian GNU/Linux 9

BeagleBoard.org Debian Image 2020-03-01

Support/FAQ: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian

default username:password is [debian:temppwd]

debian@192.168.7.2's password: 

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Sun Mar 22 14:59:11 2020
debian@beaglebone:~$ 
```

RobertCNelson, 2020-03-22:
```
debian@beaglebone:~$ cat /etc/dnsmasq.d/SoftAp0 
interface=usb0
interface=usb1
port=53
dhcp-authoritative
domain-needed
bogus-priv
expand-hosts
cache-size=2048
dhcp-range=usb0,192.168.7.1,192.168.7.1,2m
dhcp-range=usb1,192.168.6.1,192.168.6.1,2m
listen-address=127.0.0.1
listen-address=192.168.7.2
listen-address=192.168.6.2
dhcp-option=usb0,3
dhcp-option=usb0,6
dhcp-option=usb1,3
dhcp-option=usb1,6
dhcp-leasefile=/var/run/dnsmasq.leases
``

RobertCNelson, 2020-03-22:
https://github.com/RobertCNelson/bb-wl18xx-firmware/commit/58b0a1ffef1e6c424de228d8ae1cbee06f485c69

RobertCNelson, 2020-03-22:
and https://github.com/rcn-ee/customizations/commit/d496f17d16bc65fc106f57d886f87e78986c1cce

closed 2020-03-22 by RobertCNelson

RobertCNelson, 2020-03-22:
and all pushed out to the apt repo.. @JoseGoncalves hopefully that helps setup your base for mesh on top..

PS, if mesh needs some special options, it might be a good idea to use /etc/default/bb-wl18xx and have /usr/bin/bb-wl18xx-wlan0 set it up?

JoseGoncalves, 2020-03-22:
I can confirm that, with the changes that you have performed, I can disable the AP interface with the `TETHER_ENABLED` variable, while retaining the USB network connectivity.
Thank you for the quick fix!

Your idea to support wi-fi mesh seems fine to me, i.e., to add some `MESH_ENABLED` variable in `/etc/default/bb-wl18xx` and to adjust `/usr/bin/bb-wl18xx-wlan0` script to support the required changes. I will look into it, after I have the mesh properly set...