beagleboard/Latest-Images #20: Issues with TIDL examples in cloud9-examples

jadonk, 2020-03-27:
**Describe the bug**
This is a collection of issues to be fixed in cloud9-examples or other places in the image to get the TIDL examples running.

# Seems the only real issue right now is the missing libopencv-dev package.

I do need to somehow make the default runner be the Beagle Makefile runner.

**Describe which image you are running (example: bone-debian-10.2-iot-armhf-2020-02-03-4gb.img.xz) **
am57xx-debian-9.12-iot-tidl-armhf-2020-03-24-4gb.img.xz

**Describe your hardware**
BeagleBone AI, Logitech C922, USB Type-C power brick, USB Type-C hub with pass-through power

**Describe any Capes Installed**
BeagleBone Fan Cape

**Please run this built-in script**
```sudo /opt/scripts/tools/version.sh```
debian@beaglebone:/var/lib/cloud9$ sudo /opt/scripts/tools/version.sh
git:/opt/scripts/:[43e2e0e554b10d779fdbc730bb0d5d197e467d02]
model:[BeagleBoard.org_BeagleBone_AI]
dogtag:[BeagleBoard.org Debian IoT TIDL Image 2020-03-24]
UBOOT: Booted Device-Tree:[am5729-beagleboneai.dts]
kernel:[4.14.108-ti-r130]
nodejs:[v6.17.0]
/boot/uEnv.txt Settings:
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20200312.0-0rcnee0~stretch+20200312]
pkg:[bb-wl18xx-firmware]:[1.20200322.0-0rcnee0~stretch+20200322]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[librobotcontrol]:[1.0.4-git20190227.1-0rcnee0~stretch+20190327]
pkg:[firmware-ti-connectivity]:[20190717-2rcnee1~stretch+20200305]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev gpio pwm eqep remoteproc admin spi iio docker tisdk weston-launch xenomai cloud9ide]
cmdline:[console=ttyS0,115200n8 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 rng_core.default_quality=100 quiet]
dmesg | grep remote
[    4.416240] remoteproc remoteproc0: 4b234000.pru is available
[    4.418539] remoteproc remoteproc1: 4b238000.pru is available
[    4.426604] remoteproc remoteproc2: 4b2b4000.pru is available
[    4.427317] remoteproc remoteproc3: 4b2b8000.pru is available
[   34.947215] remoteproc remoteproc4: 58820000.ipu is available
[   34.975987] remoteproc remoteproc5: 55020000.ipu is available
[   35.036312] remoteproc remoteproc6: 40800000.dsp is available
[   35.080973] remoteproc remoteproc5: powering up 55020000.ipu
[   35.081021] remoteproc remoteproc5: Booting fw image dra7-ipu2-fw.xem4, size 3751356
[   35.091456] remoteproc remoteproc7: 41000000.dsp is available
[   35.223770] remoteproc remoteproc4: powering up 58820000.ipu
[   35.223824] remoteproc remoteproc4: Booting fw image dra7-ipu1-fw.xem4, size 7051536
[   35.428696] remoteproc remoteproc5: registered virtio0 (type 7)
[   35.428722] remoteproc remoteproc5: remote processor 55020000.ipu is now up
[   35.494829] remoteproc remoteproc4: registered virtio1 (type 7)
[   35.494856] remoteproc remoteproc4: remote processor 58820000.ipu is now up
[   35.656527] Modules linked in: omap_remoteproc virtio_rpmsg_bus rpmsg_core cmemk(O) uio_pdrv_genirq uio spidev pruss_soc_bus pru_rproc pruss pruss_intc
[   35.657634] Modules linked in: omap_remoteproc virtio_rpmsg_bus rpmsg_core cmemk(O) uio_pdrv_genirq uio spidev pruss_soc_bus pru_rproc pruss pruss_intc
[   35.658605] Modules linked in: omap_remoteproc virtio_rpmsg_bus rpmsg_core cmemk(O) uio_pdrv_genirq uio spidev pruss_soc_bus pru_rproc pruss pruss_intc
[   35.874871] remoteproc remoteproc7: powering up 41000000.dsp
[   35.874922] remoteproc remoteproc7: Booting fw image dra7-dsp2-fw.xe66, size 21014532
[   35.888629] remoteproc remoteproc6: powering up 40800000.dsp
[   35.888671] remoteproc remoteproc6: Booting fw image dra7-dsp1-fw.xe66, size 21014532
[   36.021653] remoteproc remoteproc6: registered virtio2 (type 7)
[   36.021673] remoteproc remoteproc6: remote processor 40800000.dsp is now up
[   36.036804] remoteproc remoteproc7: registered virtio3 (type 7)
[   36.036827] remoteproc remoteproc7: remote processor 41000000.dsp is now up
dmesg | grep pru
[    4.408190] pruss 4b200000.pruss: creating PRU cores and other child platform devices
[    4.416240] remoteproc remoteproc0: 4b234000.pru is available
[    4.416301] pru-rproc 4b234000.pru: PRU rproc node /ocp/pruss_soc_bus@4b226004/pruss@0/pru@34000 probed successfully
[    4.418539] remoteproc remoteproc1: 4b238000.pru is available
[    4.418591] pru-rproc 4b238000.pru: PRU rproc node /ocp/pruss_soc_bus@4b226004/pruss@0/pru@38000 probed successfully
[    4.420084] pruss 4b280000.pruss: creating PRU cores and other child platform devices
[    4.426604] remoteproc remoteproc2: 4b2b4000.pru is available
[... 56 lines cut ...]
 i: device........... : default
 i: Desired Resolution: 640 x 480
 i: filter........... : ./dummy.tidl.so
 i: filter args ..... :
 o: www-folder-path......: /usr/share/mjpg-streamer/www/
 o: HTTP TCP port........: 8080
 o: HTTP Listen Address..: (null)
 o: username:password....: disabled
 o: commands.............: enabled
bind: Address already in use
bind: Address already in use
 o: server_thread(): bind(8080) failed
/var/lib/cloud9/common/Makefile:169: recipe for target 'start' failed
make: *** [start] Error 1
debian@beaglebone:/var/lib/cloud9/BeagleBone/AI/tidl$ sudo lsof -i :8080
COMMAND  PID     USER   FD   TYPE DEVICE SIZE/OFF NODE NAME
nginx   1832     root    8u  IPv4  19568      0t0  TCP *:http-alt (LISTEN)
nginx   1832     root    9u  IPv6  19569      0t0  TCP *:http-alt (LISTEN)
nginx   1835 www-data    8u  IPv4  19568      0t0  TCP *:http-alt (LISTEN)
nginx   1835 www-data    9u  IPv6  19569      0t0  TCP *:http-alt (LISTEN)
nginx   1836 www-data    8u  IPv4  19568      0t0  TCP *:http-alt (LISTEN)
nginx   1836 www-data    9u  IPv6  19569      0t0  TCP *:http-alt (LISTEN)
```

work-around: https://github.com/beagleboard/cloud9-examples/commit/55ad6a8d62ded969ff76bba934cc669373a3d97a

jadonk, 2020-03-27:
How do I experience the issue with /dev/video0 so that I can test for it?

RobertCNelson, 2020-03-27:
@jadonk upgrade the kernel one more: 4.14.108-ti-r131 add's the new driver..  (the firmware for the vpe has been installed for over 2 years..)

jadonk, 2020-03-27:
nginx had static file serving turned off.

Just found
```
-       copying firmware file /tmp/cloud9-examples/bitflip.pru1_1.out to /lib/firmware/am57xx-pru1_1-fw
cp: cannot create regular file '/lib/firmware/am57xx-pru1_1-fw': Permission denied
```

jadonk, 2020-04-01:
Anything left to close here? Are the firmware files touched now?

RobertCNelson, 2020-04-01:
Not really, buster-iot-tidl is still fubar'd.. firing up stretch-iot-tidl to compare..

RobertCNelson, 2020-04-01:
Debian Stretch:
```
Python path: /usr/lib/python3.7/dist-packages:/usr/local/lib/python3.7/dist-packages
/var/lib/cloud9/common/Makefile:28: MODEL=TI_AM5728_BeagleBoard-X15_rev_C,TARGET=classification.tidl,COMMON=/var/lib/cloud9/common
/var/lib/cloud9/common/Makefile:147: GEN_DIR=/tmp/cloud9-examples,CHIP=am57xx,PROC=tidl,PRUN=,PRU_DIR=,EXE=.so
CXX     classification.tidl.cpp
LD      /tmp/cloud9-examples/classification.tidl.o
ti-mct-heap-check -c
-- ddr_heap1 ------------------------------
   Addr : 0xa2000000
   Size : 0x16000000
   Avail: 0x16000000
   Align: 0x80
-----------------------------------------

sudo mjpg_streamer -i "input_opencv.so -r 640x480 -d /dev/video1 \
--filter ./classification.tidl.so" -o "output_http.so -p 8090 -w /usr/share/mjpg-streamer/www"
[sudo] password for debian:
MJPG Streamer Version.: 2.0
 i: device........... : /dev/video1
 i: Desired Resolution: 640 x 480
 i: filter........... : ./classification.tidl.so
 i: filter args ..... :
Initializing filter
loading configuration
allocating execution object pipelines (EOP)
allocating I/O memory for each EOP
Allocating input and output buffers
Allocating input and output buffers
Allocating input and output buffers
Allocating input and output buffers
num_eops=4
About to start ProcessFrame loop!!
http://localhost:8080/?action=stream
 o: www-folder-path......: /usr/share/mjpg-streamer/www/
 o: HTTP TCP port........: 8090
 o: HTTP Listen Address..: (null)
 o: username:password....: disabled
 o: commands.............: enabled
(441)=beer_glass
(441)=beer_glass
(898)=water_bottle
(898)=water_bottle
(898)=water_bottle
(898)=water_bottle
(504)=coffee_mug
(898)=water_bottle
(898)=water_bottle
(898)=water_bottle
(898)=water_bottle
(898)=water_bottle
(504)=coffee_mug
(898)=water_bottle
(441)=beer_glass
```

RobertCNelson, 2020-04-01:
debian buster

```
Python: python3
Python path: /usr/lib/python3.7/dist-packages:/usr/local/lib/python3.7/dist-packages
/var/lib/cloud9/common/Makefile:28: MODEL=TI_AM5728_BeagleBoard-X15_rev_C,TARGET=classification.tidl,COMMON=/var/lib/cloud9/common
/var/lib/cloud9/common/Makefile:147: GEN_DIR=/tmp/cloud9-examples,CHIP=am57xx,PROC=tidl,PRUN=,PRU_DIR=,EXE=.so
ti-mct-heap-check -c
sudo mjpg_streamer -i "input_opencv.so -r 640x480 -d /dev/video1 \
--filter ./classification.tidl.so" -o "output_http.so -p 8090 -w /usr/share/mjpg-streamer/www"
[sudo] password for debian:
MJPG Streamer Version.: 2.0
 i: device........... : /dev/video1
 i: Desired Resolution: 640 x 480
 i: filter........... : ./classification.tidl.so
 i: filter args ..... :
ERROR: could not find input plugin
       Perhaps you want to adjust the search path with:
       # export LD_LIBRARY_PATH=/path/to/plugin/folder
       dlopen: ./classification.tidl.so: unexpected reloc type 0x03
make: *** [/var/lib/cloud9/common/Makefile:170: start] Error 1
```

jadonk, 2020-04-14:
Didn't we find a work-around for this?