beagleboard/Latest-Images #22: Debian 10 (Buster) IoT TIDL - Cloud9 TIDL Classification C++ errors

jomoengineer, 2020-04-01:
**Describe the bug**
A clear and concise description of what the bug is.

When attempting to run the Cloud9 TiDl classification example, the compile of the code failed for C++ errors. 

**Describe which image you are running (example: bone-debian-10.2-iot-armhf-2020-02-03-4gb.img.xz) **

`am57xx-debian-10.3-iot-tidl-armhf-2020-03-30-6gb.img.xz`

**Describe your hardware**
Beaglebone AI
**Describe any Capes Installed**
none

**Please run this built-in script**
```$ sudo /opt/scripts/tools/version.sh 
[sudo] password for debian: 
git:/opt/scripts/:[62ea733f04b46cbcc4988eb0748bddac2301ede0]
model:[BeagleBoard.org_BeagleBone_AI]
dogtag:[BeagleBoard.org Debian IoT TIDL Image 2020-03-30]
UBOOT: Booted Device-Tree:[am5729-beagleboneai.dts]
kernel:[4.14.108-ti-r131]
nodejs:[v10.15.2]
/boot/uEnv.txt Settings:
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20200330.0-0rcnee0~buster+20200330]
pkg:[bb-wl18xx-firmware]:[1.20200322.0-0rcnee0~buster+20200322]
pkg:[kmod]:[26-1]
pkg:[librobotcontrol]:[1.0.4-git20190227.1-0rcnee0~buster+20190327]
pkg:[firmware-ti-connectivity]:[20190717-2rcnee1~buster+20200305]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal bluetooth netdev i2c gpio pwm eqep remoteproc admin spi iio docker tisdk weston-launch xenomai cloud9ide]
cmdline:[console=ttyS0,115200n8 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 rng_core.default_quality=100 quiet]
dmesg | grep remote
[   13.271712] remoteproc remoteproc0: 58820000.ipu is available
[   13.311571] remoteproc remoteproc1: 55020000.ipu is available
[   13.335782] remoteproc remoteproc2: 40800000.dsp is available
[   13.337417] remoteproc remoteproc3: 41000000.dsp is available
[   13.749361] remoteproc remoteproc1: powering up 55020000.ipu
[   13.749410] remoteproc remoteproc1: Booting fw image dra7-ipu2-fw.xem4, size 3751356
[   13.783533] remoteproc remoteproc0: powering up 58820000.ipu
[   13.783588] remoteproc remoteproc0: Booting fw image dra7-ipu1-fw.xem4, size 7051536
[   14.123099] remoteproc remoteproc1: registered virtio0 (type 7)
[   14.123124] remoteproc remoteproc1: remote processor 55020000.ipu is now up
[   14.167399] remoteproc remoteproc0: registered virtio1 (type 7)
[   14.167423] remoteproc remoteproc0: remote processor 58820000.ipu is now up
[   14.327724] Modules linked in: omap_remoteproc virtio_rpmsg_bus rpmsg_core usb_f_ecm usb_f_mass_storage usb_f_rndis u_ether libcomposite uio_pdrv_genirq uio cmemk(O)
[   14.328755] Modules linked in: omap_remoteproc virtio_rpmsg_bus rpmsg_core usb_f_ecm usb_f_mass_storage usb_f_rndis u_ether libcomposite uio_pdrv_genirq uio cmemk(O)
[   15.564218] remoteproc remoteproc2: powering up 40800000.dsp
[   15.564271] remoteproc remoteproc2: Booting fw image dra7-dsp1-fw.xe66, size 21014532
[   15.639806] remoteproc remoteproc2: registered virtio2 (type 7)
[   15.639829] remoteproc remoteproc2: remote processor 40800000.dsp is now up
[   15.644150] remoteproc remoteproc3: powering up 41000000.dsp
[   15.644197] remoteproc remoteproc3: Booting fw image dra7-dsp2-fw.xe66, size 21014532
[   15.728606] remoteproc remoteproc3: registered virtio3 (type 7)
[   15.728632] remoteproc remoteproc3: remote processor 41000000.dsp is now up
[   31.460534] remoteproc remoteproc4: 4b234000.pru is available
[   31.471250] remoteproc remoteproc5: 4b238000.pru is available
[   31.478244] remoteproc remoteproc6: 4b2b4000.pru is available
[   31.486382] remoteproc remoteproc7: 4b2b8000.pru is available
dmesg | grep pru
[   25.372762] pruss_uio_shmem 4b200000.pruss_shmem: Allocating gdev
[   25.372791] pruss_uio_shmem 4b200000.pruss_shmem: Allocating info
[   25.372812] pruss_uio_shmem 4b200000.pruss_shmem: Requesting resource
[   25.372870] pruss_uio_shmem 4b200000.pruss_shmem: Mapping resource
[   25.373223] pruss_uio_shmem 4b200000.pruss_shmem: Registering with uio driver
[   25.430856] pruss_uio_shmem 4b200000.pruss_shmem: Saving platform data
[   25.431290] pruss_uio_shmem 4b280000.pruss_shmem: Allocating gdev
[   25.431314] pruss_uio_shmem 4b280000.pruss_shmem: Allocating info
[   25.431334] pruss_uio_shmem 4b280000.pruss_shmem: Requesting resource
[   25.431390] pruss_uio_shmem 4b280000.pruss_shmem: Mapping resource
[   25.431432] pruss_uio_shmem 4b280000.pruss_shmem: Registering with uio driver
[   25.442587] pruss_uio_shmem 4b280000.pruss_shmem: Saving platform data
[   30.112698] pruss 4b200000.pruss: creating PRU cores and other child platform devices
[   30.292405] pruss 4b280000.pruss: creating PRU cores and other child platform devices
[   31.460534] remoteproc remoteproc4: 4b234000.pru is available
[   31.460732] pru-rproc 4b234000.pru: PRU rproc node /ocp/pruss_soc_bus@4b226004/pruss@0/pru@34000 probed successfully
[   31.471250] remoteproc remoteproc5: 4b238000.pru is available
[   31.471434] pru-rproc 4b238000.pru: PRU rproc node /ocp/pruss_soc_bus@4b226004/pruss@0/pru@38000 probed successfully
[   31.478244] remoteproc remoteproc6: 4b2b4000.pru is available
[   31.478467] pru-rproc 4b2b4000.pru: PRU rproc node /ocp/pruss_soc_bus@4b2a6004/pruss@0/pru@34000 probed successfully
[   31.486382] remoteproc remoteproc7: 4b2b8000.pru is available
[   31.486630] pru-rproc 4b2b8000.pru: PRU rproc node /ocp/pruss_soc_bus@4b2a6004/pruss@0/pru@38000 probed successfully
dmesg | grep pinctrl-single
[    0.782355] pinctrl-single 4a003400.pinmux: 282 pins at pa fc003400 size 1128
dmesg | grep gpio-of-helper
lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END
```

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Cloud9 Web interface
2. Click on cloud9 -> BeagleBome -> tidl -> classification.tidl.cpp
3. Select Runner: C or C++ (Beagle Makefile)
4. See error
```
Started /var/lib/cloud9/: classification.tidl.cpp
File path: /var/lib/cloud9/BeagleBone/AI/tidl
File: /var/lib/cloud9/BeagleBone/AI/tidl/classification.tidl.cpp
Arguments: 
File name: classification.tidl.cpp
File extension: cpp
File base name: classification.tidl
Packages: ~.c9/packages
Project path: /var/lib/cloud9/
Project name: projectname
Hostname: localhost
Hostname path: https://undefined/BeagleBone/AI/tidl/classification.tidl.cpp
URL: http://localhost
Port: 8080
IP: 0.0.0.0
Command: BeagleBone/AI/tidl/classification.tidl.cpp
Python: python3
Python path: /usr/lib/python3.7/dist-packages:/usr/local/lib/python3.7/dist-packages
/var/lib/cloud9/common/Makefile:28: MODEL=BeagleBoard.org_BeagleBone_AI,TARGET=classification.tidl,COMMON=/var/lib/cloud9/common
/var/lib/cloud9/common/Makefile:147: GEN_DIR=/tmp/cloud9-examples,CHIP=am57xx,PROC=tidl,PRUN=,PRU_DIR=,EXE=.so
CXX     classification.tidl.cpp
classification.tidl.cpp: In function ‘int tf_postprocess(uchar*)’:
classification.tidl.cpp:352:97: error: the type ‘const tf_postprocess(uchar*)::<lambda(val_index&, val_index&)>’ of ‘constexpr’ variable ‘cmp’ is not literal
   auto constexpr cmp = [](val_index &left, val_index &right) { return left.first > right.first; };
                                                                                                 ^
classification.tidl.cpp:352:25: note: ‘tf_postprocess(uchar*)::<lambda(val_index&, val_index&)>’ is not literal because:
   auto constexpr cmp = [](val_index &left, val_index &right) { return left.first > right.first; };
                         ^
cc1plus: note:   ‘tf_postprocess(uchar*)::<lambda(val_index&, val_index&)>’ is a closure type, which is only literal in C++17 and later
make: *** [/var/lib/cloud9/common/Makefile:222: /tmp/cloud9-examples/classification.tidl.o] Error 1
```

**Expected behavior**
Complete the compile successfully and allow view from WebCam.

**Screenshots**
none

**Additional context**

jomoengineer, 2020-04-01:
Also, the blinkLED.js script fails for:

```
Debugger listening on ws://127.0.0.1:15454/de0ca06d-421a-49c9-a483-cfe8c4ba3ca9
For help, see: https://nodejs.org/en/docs/inspector
Debugger attached.
info: No pinmux for P9_15
error: Unable to write to /sys/class/gpio/gpio48/value
(node:1636) [INSPECTOR_ASYNC_STACK_TRACES_NOT_AVAILABLE] Warning: Warning: Async stack traces in debugger are not available on 32bit platforms. The feature is disabled.
error: Unable to write to /sys/class/gpio/gpio48/value
error: Unable to write to /sys/class/gpio/gpio48/value
error: Unable to write to /sys/class/gpio/gpio48/value

```

RobertCNelson, 2020-04-01:
@jadonk have you seen that build error before?

```
Python path: /usr/lib/python3.7/dist-packages:/usr/local/lib/python3.7/dist-packages
/var/lib/cloud9/common/Makefile:28: MODEL=BeagleBoard.org_BeagleBone_AI,TARGET=classification.tidl,COMMON=/var/lib/cloud9/common
/var/lib/cloud9/common/Makefile:147: GEN_DIR=/tmp/cloud9-examples,CHIP=am57xx,PROC=tidl,PRUN=,PRU_DIR=,EXE=.so
CXX     classification.tidl.cpp
classification.tidl.cpp: In function ‘int tf_postprocess(uchar*)’:
classification.tidl.cpp:352:97: error: the type ‘const tf_postprocess(uchar*)::<lambda(val_index&, val_index&)>’ of ‘constexpr’ variable ‘cmp’ is not literal
   auto constexpr cmp = [](val_index &left, val_index &right) { return left.first > right.first; };
                                                                                                 ^
classification.tidl.cpp:352:25: note: ‘tf_postprocess(uchar*)::<lambda(val_index&, val_index&)>’ is not literal because:
   auto constexpr cmp = [](val_index &left, val_index &right) { return left.first > right.first; };
                         ^
cc1plus: note:   ‘tf_postprocess(uchar*)::<lambda(val_index&, val_index&)>’ is a closure type, which is only literal in C++17 and later
make: *** [/var/lib/cloud9/common/Makefile:222: /tmp/cloud9-examples/classification.tidl.o] Error 1
```

RobertCNelson, 2020-04-01:
@jomoengineer 

> Also, the blinkLED.js script fails for:
>
> ```
> Debugger listening on ws://127.0.0.1:15454/de0ca06d-421a-49c9-a483-cfe8c4ba3ca9
> For help, see: https://nodejs.org/en/docs/inspector
> Debugger attached. [...]

Sadly, all the gpio demos need work on the bbai

RobertCNelson, 2020-04-01:
@jomoengineer this should fix it: https://github.com/beagleboard/cloud9-examples/pull/31

RobertCNelson, 2020-04-01:
Debian Buster

```
debian@beaglebone:/var/lib/cloud9/BeagleBone/AI/tidl$ cat /etc/dogtag
BeagleBoard.org Debian IoT TIDL Image 2020-03-30
```

```
tarted /var/lib/cloud9/: classification.tidl.cpp
File path: /var/lib/cloud9/BeagleBone/AI/tidl
File: /var/lib/cloud9/BeagleBone/AI/tidl/classification.tidl.cpp
Arguments:
File name: classification.tidl.cpp
File extension: cpp
File base name: classification.tidl
Packages: ~.c9/packages
Project path: /var/lib/cloud9/
Project name: projectname
Hostname: localhost
Hostname path: https://undefined/BeagleBone/AI/tidl/classification.tidl.cpp
URL: http://localhost
Port: 8080
IP: 0.0.0.0
Command: BeagleBone/AI/tidl/classification.tidl.cpp
Python: python3
Python path: /usr/lib/python3.7/dist-packages:/usr/local/lib/python3.7/dist-packages
/var/lib/cloud9/common/Makefile:28: MODEL=TI_AM5728_BeagleBoard-X15_rev_C,TARGET=classification.tidl,COMMON=/var/lib/cloud9/common
/var/lib/cloud9/common/Makefile:147: GEN_DIR=/tmp/cloud9-examples,CHIP=am57xx,PROC=tidl,PRUN=,PRU_DIR=,EXE=.so
CXX     classification.tidl.cpp
LD      /tmp/cloud9-examples/classification.tidl.o
ti-mct-heap-check -c
-- ddr_heap1 ------------------------------
   Addr : 0xa2000000
   Size : 0x16000000
   Avail: 0x16000000
   Align: 0x80
-----------------------------------------

sudo mjpg_streamer -i "input_opencv.so -r 640x480 -d /dev/video1 \
--filter ./classification.tidl.so" -o "output_http.so -p 8090 -w /usr/share/mjpg-streamer/www"
[sudo] password for debian:
MJPG Streamer Version.: 2.0
 i: device........... : /dev/video1
 i: Desired Resolution: 640 x 480
 i: filter........... : ./classification.tidl.so
 i: filter args ..... :
Initializing filter
loading configuration
allocating execution object pipelines (EOP)
allocating I/O memory for each EOP
Allocating input and output buffers
Allocating input and output buffers
Allocating input and output buffers
num_eops=4
About to start ProcessFrame loop!!
http://localhost:8080/?action=stream
 o: www-folder-path......: /usr/share/mjpg-streamer/www/
 o: HTTP TCP port........: 8090
 o: HTTP Listen Address..: (null)
 o: username:password....: disabled
 o: commands.............: enabled
(898)=water_bottle
```

RobertCNelson, 2020-04-01:
@jomoengineer , no merged, from cloud 9, just run:

```
cd /var/lib/cloud9/BeagleBone/AI/tidl
git pull
make clean
```

Which will merge all the fixes and clean a tmp file, then click run like before..

Regards,

jomoengineer, 2020-04-02:
Hey Robert,

Yeah, that seems to do the trick.
Thanks!

Jon

closed 2020-04-02 by RobertCNelson