beagleboard/Latest-Images #23: [PRU] v4.19.x-ti names

RobertCNelson, 2020-04-01:
Users are seeing an odd situation with the PRU names..

```
echo '/lib/firmware/PRU_gpioToggle.out' > /sys/class/remoteproc/remoteproc1/firmware
```
doesn't work.. 

```
remoteproc remoteproc1: Direct firmware load for /lib/firmware/PRU_gpioToggle.out failed with error -2
```

However using.. 
```
/lib/firmware/am335x-pru0-fw
/lib/firmware/am335x-pru1-fw
```

Then run:

```
sudo update-initramfs -uk `uname -r`
sudo reboot
```

mvduin, 2020-04-22:
Those paths are not stable and shouldn't be hardcoded.  Instead, use udev rules to create symlinks.

mvduin, 2020-04-22:
is perhaps remoteproc1 not a pru device?  These sysfs paths are absolutely not guaranteed to be stable and should not be hardcoded.  Udev rules should be used to create stable symlinks for any path in /dev and /sys that users need unless the kernel guarantees their stability (e.g. for i2c controllers and uarts the numbering is determined by DT).

I'm downloading the latest image right now, I'll see if I can cook up a suggestion for remoteproc.

mvduin, 2020-04-22:
Also, I just saw the rules in 86-remoteproc-noroot.rules that touch, chgrp, and chmod all pru firmware files each time any remoteproc device is discovered.... that's just really gross.

Just make those firmware files symlinks to some place that's writable by the remoteproc group.  And regardless of how it's done, it only needs to be done once when the image is built, not on every boot, let alone each time any remoteproc device is discovered by udev.

mvduin, 2020-04-23:
Suggested replacement for 86-remoteproc-noroot.rules:
```
# identify remoteproc cores on am335x
SUBSYSTEM=="remoteproc", KERNELS=="4a334000.pru",  ENV{REMOTEPROC_NAME}="pruss-core0"
SUBSYSTEM=="remoteproc", KERNELS=="4a338000.pru",  ENV{REMOTEPROC_NAME}="pruss-core1"

# identify remoteproc cores on am572x
SUBSYSTEM=="remoteproc", KERNELS=="4b234000.pru",  ENV{REMOTEPROC_NAME}="pruss1-core0"
SUBSYSTEM=="remoteproc", KERNELS=="4b238000.pru",  ENV{REMOTEPROC_NAME}="pruss1-core1"
SUBSYSTEM=="remoteproc", KERNELS=="4b2b4000.pru",  ENV{REMOTEPROC_NAME}="pruss2-core0"
SUBSYSTEM=="remoteproc", KERNELS=="4b2b8000.pru",  ENV{REMOTEPROC_NAME}="pruss2-core1"
SUBSYSTEM=="remoteproc", KERNELS=="58820000.ipu",  ENV{REMOTEPROC_NAME}="ipu1"
SUBSYSTEM=="remoteproc", KERNELS=="55020000.ipu",  ENV{REMOTEPROC_NAME}="ipu2"
SUBSYSTEM=="remoteproc", KERNELS=="40800000.dsp",  ENV{REMOTEPROC_NAME}="dsp1"
SUBSYSTEM=="remoteproc", KERNELS=="41000000.dsp",  ENV{REMOTEPROC_NAME}="dsp2"

# set permissions and create symlinks
SUBSYSTEM=="remoteproc", ENV{REMOTEPROC_NAME}!="", ACTION=="add", \
        RUN+="/bin/chgrp -R remoteproc '/sys%p'", \
        RUN+="/bin/chmod -R g=u '/sys%p'", \
        RUN+="/bin/mkdir -p /dev/remoteproc", \
        RUN+="/bin/ln -sT '/sys/class/remoteproc/%k' /dev/remoteproc/%E{REMOTEPROC_NAME}"
```
Tested on BBB running latest buster-iot image with kernel 4.19:
```
debian@beaglebone:~$ ls -ld /dev/remoteproc/*
lrwxrwxrwx 1 root root 33 Apr 23 10:31 /dev/remoteproc/pruss-core0 -> /sys/class/remoteproc/remoteproc1
lrwxrwxrwx 1 root root 33 Apr 23 10:31 /dev/remoteproc/pruss-core1 -> /sys/class/remoteproc/remoteproc2
```
And on BBAI running previous stretch-iot image with kernel 4.14:
```
debian@bbai:~$ ls -ld /dev/remoteproc/*
lrwxrwxrwx 1 root root 33 Apr 23 08:28 /dev/remoteproc/dsp1 -> /sys/class/remoteproc/remoteproc6
lrwxrwxrwx 1 root root 33 Apr 23 08:28 /dev/remoteproc/dsp2 -> /sys/class/remoteproc/remoteproc7
lrwxrwxrwx 1 root root 33 Apr 23 08:28 /dev/remoteproc/ipu1 -> /sys/class/remoteproc/remoteproc4
lrwxrwxrwx 1 root root 33 Apr 23 08:28 /dev/remoteproc/ipu2 -> /sys/class/remoteproc/remoteproc5
lrwxrwxrwx 1 root root 33 Apr 23 08:28 /dev/remoteproc/pruss1-core0 -> /sys/class/remoteproc/remoteproc0
lrwxrwxrwx 1 root root 33 Apr 23 08:28 /dev/remoteproc/pruss1-core1 -> /sys/class/remoteproc/remoteproc1
lrwxrwxrwx 1 root root 33 Apr 23 08:28 /dev/remoteproc/pruss2-core0 -> /sys/class/remoteproc/remoteproc2
lrwxrwxrwx 1 root root 33 Apr 23 08:28 /dev/remoteproc/pruss2-core1 -> /sys/class/remoteproc/remoteproc3
```

RobertCNelson, 2020-04-23:
Thanks! @mvduin for taking a look at this, it is now pushed out as "bb-customizations_1.20200423.0"

closed 2020-05-04 by jadonk