beagleboard/Latest-Images #24: Chrome on BeagleBoard.org Debian IoT Image 2020-03-26 error messages

kenmartin-unix, 2020-04-01:
Running Chromium 80.0.3987.149 triggers many error messages
on a plain BBB BeagleBoard.org Debian IoT Image 2020-03-26

Xlib:  extension "XInputExtension" missing on display ":1.0".
Xlib:  extension "RANDR" missing on display ":1.0".
Xlib:  extension "XInputExtension" missing on display ":1.0".
Xlib:  extension "XInputExtension" missing on display ":1.0".
Xlib:  extension "XInputExtension" missing on display ":1.0".
[21322:21322:0401/160515.441645:ERROR:gl_surface_glx.cc(453)] glxQueryVersion failed
[21322:21322:0401/160515.446727:ERROR:gl_initializer_x11.cc(148)] GLSurfaceGLX::InitializeOneOff failed.
[21322:21322:0401/160515.560110:ERROR:viz_main_impl.cc(161)] Exiting GPU process due to errors during initialization
[21325:21345:0401/160517.606984:ERROR:ssl_client_socket_impl.cc(941)] handshake failed; returned -1, SSL error code 1, net_error -200
[21357:1:0401/160520.501654:ERROR:child_process_sandbox_support_impl_linux.cc(79)] FontService unique font name matching request did not receive a response.
[21357:1:0401/160520.506248:ERROR:child_process_sandbox_support_impl_linux.cc(79)] FontService unique font name matching request did not receive a response.
[21357:1:0401/160520.544280:ERROR:child_process_sandbox_support_impl_linux.cc(79)] FontService unique font name matching request did not receive a response.
[21357:1:0401/160520.559400:ERROR:child_process_sandbox_support_impl_linux.cc(79)] FontService unique font name matching request did not receive a response.
Xlib:  extension "XInputExtension" missing on display ":1.0".
[21354:21354:0401/160521.158329:ERROR:sandbox_linux.cc(374)] InitializeSandbox() called with multiple threads in process gpu-process.
[21354:21354:0401/160521.334909:ERROR:gpu_channel_manager.cc(459)] ContextResult::kFatalFailure: Failed to create shared context for virtualization.
Xlib:  extension "XInputExtension" missing on display ":1.0".
Xlib:  extension "XInputExtension" missing on display ":1.0".
debian@beaglebone:~$ 

Firefox 68.6.0esr (32 bit) runs fine and much quicker then Chromium
once you turn off pre-fetching in settings

Both of the above were run under a VNC session.

Thank you

RobertCNelson, 2020-04-01:
@kenmartin-unix thanks for the feedback. I'm glad to hear Firefox is running faster then Chromium..

2 questions come to mind, what's the install size of Firefox vs Chromium, and is it usable? ;)

RobertCNelson, 2020-04-01:
As far as the error's... yeah your running thru vnc, and we have no xorg video drivers, so yeah.. that's normal..

kenmartin-unix, 2020-04-01:
===> 2 questions come to mind, what's the install size of Firefox vs
Chromium

There are files in many places. How do I determine that?

Yes it is sluggish , and usable when required.
It's a lot to ask of a little single core machine.
I will say it generates a lot of traffic to the SD
Card. I see this displaying mmc0 on USR0 led.

😎

kenmartin-unix, 2020-04-06:
Resolved

closed 2020-04-06 by kenmartin-unix