beagleboard/Latest-Images #28: RE: Config-pin or Setting Up GPIO Pins on the BBB for the RelayCape (Nodejs and Node and PinMux)

silver2row, 2020-04-07:
**Describe the bug**
Pin muxing

**Describe which image you are running
BeagleBoard.org Debian Buster IoT Image 2020-04-06

**Describe your hardware**
BBB

**Describe any Capes Installed**
RelayCape

**Please run this built-in script**
```sudo /opt/scripts/tools/version.sh```
git:/opt/scripts/:[b39ec679648a6be8f25f48bd1c9784c1fc5a0c46]
eeprom:[A335BNLT0A5C3516BBBK2568]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Buster IoT Image 2020-04-06]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2019.04-00002-g07d5700e21]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-boneblack-uboot-univ.dts]
UBOOT: Loaded Overlay:[AM335X-PRU-RPROC-4-19-TI-00A0]
UBOOT: Loaded Overlay:[BB-ADC-00A0]
UBOOT: Loaded Overlay:[BB-BONE-eMMC1-01-00A0]
UBOOT: Loaded Overlay:[BB-HDMI-TDA998x-00A0]
UBOOT: Loaded Overlay:[BBORG_RELAY-00A2]
kernel:[4.19.94-ti-r42]
nodejs:[v10.15.2]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-19-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20200403.0-0rcnee0~buster+20200403]
pkg:[bb-wl18xx-firmware]:[1.20200322.0-0rcnee0~buster+20200322]
pkg:[kmod]:[26-1]
pkg:[librobotcontrol]:[1.0.4-git20190227.1-0rcnee0~buster+20190327]
pkg:[firmware-ti-connectivity]:[20190717-2rcnee1~buster+20200305]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal bluetooth netdev i2c gpio pwm eqep remoteproc admin spi iio docker tisdk weston-launch xenomai cloud9id                                  e]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait uboot_detected_capes=BBORG_RELAY, coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.                                  default_quality=100 quiet]
dmesg | grep remote
[   54.555262] remoteproc remoteproc0: wkup_m3 is available
[   54.571908] remoteproc remoteproc0: powering up wkup_m3
[   54.571931] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[   54.572208] remoteproc remoteproc0: remote processor wkup_m3 is now up
[   59.456359] remoteproc remoteproc1: 4a334000.pru is available
[   59.458396] remoteproc remoteproc2: 4a338000.pru is available
dmesg | grep pru
[   59.456359] remoteproc remoteproc1: 4a334000.pru is available
[   59.456554] pru-rproc 4a334000.pru: PRU rproc node pru@4a334000 probed successfully
[   59.458396] remoteproc remoteproc2: 4a338000.pru is available
[   59.458529] pru-rproc 4a338000.pru: PRU rproc node pru@4a338000 probed successfully
dmesg | grep pinctrl-single
[    0.921924] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
[    0.935164] gpio-of-helper ocp:cape-universal: ready
lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END

**To Reproduce**
Steps to reproduce the behavior:
1. Make a node.js file w/ bonescript for reading a LED/Light.
2. Run the source w/ node myfile.js but nodejs works but only at times w/ this error: info: No pinmux for P9_42
3. Wait for the error to produce in time.

4. Here is the error:
    ERROR: open() for /sys/devices/platform/ocp/ocp:P9_42_pinmux/state failed, No such file or directory

**Expected behavior**
The relay2 on the RelayCape is supposed to toggle on/off in time.

**Screenshots**
I tried config-pin and I have not set up the GPIO in the filesystem yet.

**Additional context**
New image, new issue. Please advise.

RobertCNelson, 2020-04-07:
Turn Relay 2 on:

```
echo 1 > /sys/class/leds/relay-jp2/brightness
```

Turn Relay 2 off:

```
echo 0 > /sys/class/leds/relay-jp2/brightness
```

Regards,

silver2row, 2020-04-07:
Hello, 

I forgot the simple LED example:

   const b = require('bonescript');

    const light = "P9_42";
    b.pinMode(light, b.OUTPUT);

    var state = 0;

    function toggleLIGHT() {
        state = state ^ 1;
        b.digitalWrite(light, state);
    }

    setInterval(toggleLIGHT, 900);
    console.log("Blinking on and off on the BBB!"); 

I found this source or something similar on the examples at /var/lib/cloud9/.

Seth

P.S. Just for reference:

1. node myfile.js 
    
    info: No pinmux for P9_42
    Blinking on and off on the BBB!

2. nodejs myfile.js

    info: No pinmux for P9_42
    Blinking on and off on the BBB!

Both 1. and 2. now result in the same thing, working on Relay2, with error code and w/ Relay2 working while Relay1 is out.

RobertCNelson, 2020-04-07:
P9_42 is already exported from the overlay..

silver2row, 2020-04-07:
Hello,

Okay. No issue. So, just using the file descriptor in the a file would be good. Got it.

Seth

P.S. Sorry to waste your time on this effort.

closed 2020-04-07 by silver2row

RobertCNelson, 2020-04-07:
@silver2row if you want to help the next user, add an "example" doc here:

https://github.com/beagleboard/bb.org-overlays/tree/master/examples

All 4 relays on this board are controlled thru the led subsystem, as it's quick and easy for permissions/etc..

Regards,

silver2row, 2020-04-07:
Hello @RobertCNelson ,

Okay. I will add an example there soon.

Seth

RobertCNelson, 2020-04-07:
@silver2row ps, since it's an LED, you can also thus add system triggers to the "relay"... imagine hav'ing the relay 'turn on' any mmc access or cpu cycle.. yeah you can do some stupid cool things with it..

silver2row, 2020-04-08:
Hello @RobertCNelson ,

Thank you again for your support. 

Seth