beagleboard/Latest-Images #31: Add country config for hostapd

JoseGoncalves, 2020-04-18:
**Is your feature request related to a problem? Please describe.**
When using an auto-generated hostapd config there is no way to set country regulation for proper setting of transmit power and channel availability.

**Describe the solution you'd like**
To add a variable in `/etc/default/bb-wl18xx`, e.g., `USE_PERSONAL_COUNTRY`, to be able to set the country code (ISO/IEC 3166-1). Use that variable in the `/usr/bin/bb-wl18xx-tether` script for proper setting of `/tmp/hostapd-wl18xx.conf`.

RobertCNelson, 2020-04-20:
@JoseGoncalves this should be wired in now...

RobertCNelson, 2020-04-20:
https://github.com/rcn-ee/repos/commit/fb8390a4555b1df50acc373e8d393c5a395dd2a4

closed 2020-04-20 by RobertCNelson