beagleboard/Latest-Images #32: Update comment about USE_GENERATED_HOSTAPD

JoseGoncalves, 2020-04-20:
A documentation issue. In `/etc/default/bb-wl18xx` the `USE_GENERATED_HOSTAPD` config var is documented like this:
```
# USE_GENERATED_HOSTAPD: use generated version of /etc/hostapd.conf; set to no so user can modify /etc/hostapd.conf
USE_GENERATED_HOSTAPD=yes
```
Nevertheless, for a generated `hostapd` config, the script `/usr/bin/bb-wl18xx-tether` is creating the file in `/tmp`. The comment shoud be updated to indicate this.

RobertCNelson, 2020-04-20:
This will help a little bit, but it won't fix the issue:

https://github.com/RobertCNelson/bb-wl18xx-firmware/commit/d32826077d965af213e7a4cf9f2f23361c14a935

JoseGoncalves, 2020-04-20:
The comment, as it is, is misleading, because a user could think the hostapd config file is always (genertaed and non-generated) /etc/hostapd.conf.
 
My suggestion for something more clear is something like this:
```
# USE_GENERATED_HOSTAPD: use generated version of hostapd.conf; if set to yes a (non-modificable) config file will be generated at /tmp/hostapd-wl18xx.conf; if set to no user should create a config file named /etc/hostapd.conf.

closed 2020-04-20 by RobertCNelson

RobertCNelson, 2020-04-20:
https://github.com/rcn-ee/repos/commit/fb8390a4555b1df50acc373e8d393c5a395dd2a4