beagleboard/Latest-Images #33: PKCS#7 signature not signed with a trusted key

JoseGoncalves, 2020-04-20:
**Describe the bug**
After installing the `wireless-regdb` package from `buster-backports`:
```
debian@beaglebone:~$ dpkg -s wireless-regdb
Package: wireless-regdb
Status: install ok installed
Priority: optional
Section: net
Installed-Size: 43
Maintainer: Ben Hutchings <ben@decadent.org.uk>
Architecture: all
Multi-Arch: foreign
Version: 2019.06.03-1~bpo10+1
Suggests: crda
Description: wireless regulatory database for Linux
 This package provides a machine-readable database of legal
 regulations on radio emissions that affect use of wireless networking.
 The Linux kernel and drivers use this to maintain compliance with
 those regulations.
Homepage: https://wireless.wiki.kernel.org/en/developers/regulatory/wireless-regdb
```

I start seing an error outputed by dmesg regarding a PKCS#7 signature:
```
debian@beaglebone:~$ dmesg | grep -C1 cfg80211
[   45.509878] Bluetooth: BNEP socket layer initialized
[   46.081340] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[   46.094504] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[   46.107736] PKCS#7 signature not signed with a trusted key
[   46.119170] cfg80211: loaded regulatory.db is malformed or signature is missing/invalid
[   47.536739] wlcore: wl18xx HW: 183x or 180x, PG 2.2 (ROM 0x11)
```

This error seems that it does not affect the BeagleBone wireless functionality, nevertheless I thought it would be better to report this.

**Describe which image you are running**
bone-debian-10.3-console-armhf-2020-04-13-1gb.img.xz

**Describe your hardware**
BeagleBone Green Wireless

**Describe any Capes Installed**
None

**Expected behavior**
No errors displayed by dmesg

RobertCNelson, 2020-04-20:
@JoseGoncalves 

wireless-regdb : https://tracker.debian.org/pkg/wireless-regdb

We don't have Debian's signing key..

closed 2020-04-20 by RobertCNelson

JoseGoncalves, 2020-04-20:
Don't quite understand what needs to be signed. Is it the wireless driver?
Would'nt it be possible to add a self-gen key to the trusted keys in  `/lib/crda/pubkeys/` to silence this error?

RobertCNelson, 2020-04-20:
```
cfg80211: loaded regulatory.db is malformed or signature is missing/invalid
```
Our kernel isn't signed with debian's key...

Go try adding a self-gen key, i don't think it'll do anything..

JoseGoncalves, 2020-04-20:
The following wiki seems to address this issue:
https://github.com/linuxwacom/input-wacom/wiki/Kernel-Module-Signing

Nevertheless, for me it's enough that the wireless regulatory information shipped in package `wireless-regdb` is correctly read, and needed to use a recent one (provided by buster-backports). Just don't like to see messages colored red in the dmesg outputs...

JoseGoncalves, 2020-04-20:
Found a soultion for this issue:
```
debian@beaglebone:~$ sudo update-alternatives --set regulatory.db /lib/firmware/regulatory.db-upstream
```
after a reboot, the error disappeared:
```
debian@beaglebone:~$ dmesg | grep -C1 cfg80211
[   49.935875] Bluetooth: BNEP socket layer initialized
[   49.959420] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[   50.108989] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[   51.552765] wlcore: wl18xx HW: 183x or 180x, PG 2.2 (ROM 0x11)
```

RobertCNelson, 2020-04-20:
Awesome! Good Job @JoseGoncalves , where was that even documented?

JoseGoncalves, 2020-04-21:
Did not found any documentation on it.

Tried that after browsing the kernel sources were those errors are printed (`net/wireless/reg.c` and `certs/system_keyring.c`) and after verifying that the `wireless-regdb`package from `buster-backports` installs two versions of the `regulatory.db` file, one signed by Debian and another signed by upstream:

```
debian@beaglebone:~$ update-alternatives --list regulatory.db
/lib/firmware/regulatory.db-debian
/lib/firmware/regulatory.db-upstream
```

JoseGoncalves, 2020-04-21:
Found a `wireless-regdb` bug report (from the package maintainer) were the `regulatory.db` alternatives are addressed:

https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=950332