beagleboard/Latest-Images #34: Lost connection to the Bonescript page, will not interact anymore

craytb, 2020-04-22:
As per the topic above, I tried updating the software of the Beaglebone Black, and now I cannot connect to like the bonescript page or follow the setup as per the original device, it just goes straight to Cloud9.

Also, it is not really clear on how to 'pick' your update, like one that says 'newbies to Beaglebone may want to start here'

Beaglebone Black
No Capes
**Describe any Capes Installed**

**Please run this built-in script**
`
git:/opt/scripts/:[62ea733f04b46cbcc4988eb0748bddac2301ede0]
eeprom:[A335BNLTEIA04218BBBK0437]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian LXQt Image 2020-04-01]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2019.04-00002-g07d5700e21]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-boneblack-uboot-univ.dts]
UBOOT: Loaded Overlay:[AM335X-PRU-RPROC-4-14-TI-00A0]
UBOOT: Loaded Overlay:[BB-ADC-00A0]
UBOOT: Loaded Overlay:[BB-BONE-eMMC1-01-00A0]
UBOOT: Loaded Overlay:[BB-HDMI-TDA998x-00A0]
kernel:[4.14.108-ti-r131]
nodejs:[v6.17.0]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20200331.0-0rcnee0~stretch+20200331]
pkg:[bb-wl18xx-firmware]:[1.20200322.0-0rcnee0~stretch+20200322]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[librobotcontrol]:[1.0.4-git20190227.1-0rcnee0~stretch+20190327]
pkg:[firmware-ti-connectivity]:[20190717-2rcnee1~stretch+20200305]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev gpio pwm eqep remoteproc admin spi iio docker tisdk weston-launch xenomai cloud9ide]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet]
dmesg | grep remote
[    0.992673] remoteproc remoteproc0: wkup_m3 is available
[    1.205257] remoteproc remoteproc0: powering up wkup_m3
[    1.205374] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[    1.209934] remoteproc remoteproc0: remote processor wkup_m3 is now up
[   89.730447] remoteproc remoteproc1: 4a334000.pru is available
[   89.745358] remoteproc remoteproc2: 4a338000.pru is available
dmesg | grep pru
[   88.883170] pruss 4a300000.pruss: creating PRU cores and other child platform devices
[   89.730447] remoteproc remoteproc1: 4a334000.pru is available
[   89.730612] pru-rproc 4a334000.pru: PRU rproc node /ocp/pruss_soc_bus@4a326004/pruss@0/pru@34000 probed successfully
[   89.745358] remoteproc remoteproc2: 4a338000.pru is available
[   89.745504] pru-rproc 4a338000.pru: PRU rproc node /ocp/pruss_soc_bus@4a326004/pruss@0/pru@38000 probed successfully
dmesg | grep pinctrl-single
[    0.684736] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
dmesg | grep gpio-of-helper
[    0.697309] gpio-of-helper ocp:cape-universal: ready
lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END
debian@beaglebone:/sys/module$

**Expected behavior**
I expected the startup page to be there with the bonescripts and getting started stuff, connected to my new Beaglebone Black

Respectfully submitted,
Bret Craytor

RobertCNelson, 2020-04-22:
@craytb BeagleBoard.org decided to EOL of the Bone script server documentation and replace it with Cloud9, all documentation is now available in the left side plane.

All stored documentation is available on the left plane, here is the repo that contains that information.
https://github.com/beagleboard/cloud9-examples

Regards,

RobertCNelson, 2020-04-22:
@craytb as far as "picking" your update... https://beagleboard.org/latest-images shows:

[image]

What do you want us to say?

I'm not in charge of that website, @jadonk controls it..

craytb, 2020-04-24:
So, both of there are the IoT versions which don’t have the desktop, am I right? And for those of us trying to learn a new platform, the desktop seems to make it simpler.  So, are there versions with the desktop that aren’t recommended anymore?  I guess I am just looking for a more clear explanation. I am sorry I troubled your forum spot as I was just trying to get the right people.  I hope you have a new endeavor soon and don’t have problems with it changing right out of the box to something that was not expected, and not knowing where to look for answers.  Thank you again for your reply.  

Bret Craytor

RobertCNelson, 2020-04-24:
Correct the "Recommended" IoT version's do not have a desktop by default.

There is a LXQt snapshot generated once a month, it's listed here:

https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#Debian_Stretch_LXQt_Snapshot

That url is the same base url that shows up on the Login screen of every default image.

So go ahead and give it shot.  Just remember, we have a single 1Ghz core, limited ram, no 2d hardware accelerator, running of eMMC/microSD...

After about a miniute, you should reliaize.. We can't compete with the PI for a desktop, we just don't have the horse power or the man power..

But you know what, the am335x really good at:  PRUSS, PWM, CAN, SPI, USARTS, GPIO, Ethernet... A Desktop really doesn't fit, other then taking away RAM from the end application.

So since Desktop sucks, we have to give use a really good non-desktop tool... Hence Cloud9...

Regards,

craytb, 2020-04-25:
Ahh, that helps me a lot!  Thank you so much for your input. And I understand.  Those PRUs are awesome though!   

Bret Craytor