beagleboard/Latest-Images #37: bonescript is broken

mvduin, 2020-04-29:
[Bone101](http://beagleboard.org/Support/bone101/) and the [BoneScript documentation](http://beagleboard.org/Support/BoneScript/) try to load `http://HOST/bonescript.js` (where HOST is whatever you fill into the form, or 192.168.7.2 by default).  However, on the latest IoT image this results in a 404 error, thus the ability to run examples does not function.

The [getting started guide's networking section](http://beagleboard.org/getting-started#step2) likewise tries (and fails) to load `http://HOST/bonescript.js`.  It also tries to load `http://HOST:8000/bonescript.js`, which causes bonescript.service to be socket-activated on my beaglebone.  However, it seems to fail to adopt the socket provided by systemd and instead tried to open its own socket, which fails (EADDRINUSE).

I haven't tried to debug this issue, but I did notice that the "[systemd](https://www.npmjs.com/package/systemd)" module bonescript uses for socket activation does not support nodejs 0.11.0 or later (according to its package.json), presumably because of the hideous hacks it does (uses a nodejs-internal binding and globally patches `net.Server.prototype.listen`).  You may want to consider replacing it by my [node-sd-daemon](https://github.com/dutchanddutch/node-sd-daemon#readme) library, which supports socket activation and a lot more, works on all recent nodejs versions (afaik) and is actually maintained.  I haven't gotten around to pushing it to npm yet, but you can just use the git version directly (`npm install github:dutchanddutch/node-sd-daemon`).

RobertCNelson, 2020-04-30:
@jadonk what do we really want to do with Bonescript?  I know you've been transiting many demos to 'fs' over 'bonescript'... 

https://github.com/beagleboard/cloud9-examples/commit/b135461b903423c8d0279de68698a4a5398b8d03

https://github.com/beagleboard/cloud9-examples/commit/950e02c0d0be04c355e763cbbdb320349f3aad08

I know you mentioned one the big problems with bonescript, i that it is just so slow on startup..

Do we kill it, is there anything useful, that hasn't been converted to cloud-examples repo?

Regards,

jadonk, 2020-05-01:
I fixed the loading time, mostly. I just want to get more useful bug reports and community fixes.  Vaishnav fixed a lot of the on usage front. If the server security can be enabled, I’d be okay to leave and support it...if people actually say they want it.

mvduin, 2020-05-01:
I hope you're doing an origin-check to ensure only beagleboard.org and the beaglebone itself can connect a websocket?

jadonk, 2020-05-15:
> I hope you're doing an origin-check to ensure only beagleboard.org and the beaglebone itself can connect a websocket?

Not at all.

mvduin, 2020-05-15:
Ah, so any website can control any beaglebone on the visitor's network?  Great feature :-)

RobertCNelson, 2020-05-15:
@mvduin it's worse then that..  Give me an ip address of a on world accessible Beagle..

http://beagleboard.org/updater

Regards,

mvduin, 2020-05-15:
who on earth would have a beagle with a world accessible IP address?  If you do that without properly securing your beaglebone you kinda have it coming.  The fact that any website can access beaglebones on any visitor's local network is far, FAR worse.