beagleboard/Latest-Images #40: Increase DHCP lease times

JoseGoncalves, 2020-05-04:
**Is your feature request related to a problem? Please describe.**
When debugging my BeagleBone software and looking into `/var/log/syslog` and/or `/var/log/daemon.log` I always see a lot of messges like this:

```
May  4 09:56:32 beaglebone dnsmasq-dhcp[2022]: DHCPREQUEST(usb0) 192.168.7.1 fc:69:47:32:59:26 
May  4 09:56:32 beaglebone dnsmasq-dhcp[2022]: DHCPACK(usb0) 192.168.7.1 fc:69:47:32:59:26 HP-ProBook-450-G6
May  4 09:56:51 beaglebone dnsmasq-dhcp[2022]: DHCPREQUEST(usb1) 192.168.6.1 fc:69:47:32:59:28 
May  4 09:56:51 beaglebone dnsmasq-dhcp[2022]: DHCPACK(usb1) 192.168.6.1 fc:69:47:32:59:28 HP-ProBook-450-G6
May  4 09:57:15 beaglebone dnsmasq-dhcp[2022]: DHCPREQUEST(usb0) 192.168.7.1 fc:69:47:32:59:26 
May  4 09:57:15 beaglebone dnsmasq-dhcp[2022]: DHCPACK(usb0) 192.168.7.1 fc:69:47:32:59:26 HP-ProBook-450-G6
May  4 09:57:44 beaglebone dnsmasq-dhcp[2022]: DHCPREQUEST(usb1) 192.168.6.1 fc:69:47:32:59:28 
May  4 09:57:44 beaglebone dnsmasq-dhcp[2022]: DHCPACK(usb1) 192.168.6.1 fc:69:47:32:59:28 HP-ProBook-450-G6
```
which are caused by a DHCP lease time set to 2 minutes in the USB interfaces (in script `/usr/bin/bb_dnsmasq_config.sh` ). This is very distracting and does not provide anything usefull (unless you are debugging for issues in the DHCP server).

**Describe the solution you'd like**
To increase the DHCP lease time for the USB interfaces from 2 minutes to something between 20 to 30 minutes and the SoftAP0 interface from 10 minutes to something between 1 to 2 hours.

RobertCNelson, 2020-05-04:
okay: https://github.com/beagleboard/customizations/pull/5

RobertCNelson, 2020-05-17:
This was merged, closing.. ;)

closed 2020-05-17 by RobertCNelson