beagleboard/Latest-Images #41: Distro boot

jadonk, 2020-05-15:
**Is your feature request related to a problem? Please describe.**
If folks have aligned on a way to boot distros from u-boot, why can't we use it? If it doesn't handle our use cases, then it probably needs to be augmented to support them. Always needing to patch mainline to boot our images stinks.

**Describe the solution you'd like**
I want to be able to run mainline u-boot. We can still have some custom variations for ease-of-use specific to our community, but at least swapping out our u-boot with mainline u-boot should "just work".

**Describe alternatives you've considered**
Not sure what alternatives there are to consider. We might lose some features, like add-on board overlays, but I'd like to see what we can do to resolve that. This is probably the biggest thing missing upstream. We probably need to tackle it with some actual u-boot code and not just u-boot script.

**Additional context**
This is also about running other distro images and seeing capes supported in other distros.

I'm not sure who is best to tackle this problem. Perhaps this could be a challenge for Major League Hackers?

RobertCNelson, 2020-05-15:
u-boot distro boot uses "extlinux.conf" it's works pretty well for static *.dtb

https://github.com/u-boot/u-boot/blob/master/doc/README.distro#L128-L156

I'd expect this to already work today on mainline u-boot with our images.

/boot/extlinux/extlinux.conf
```
label Linux 4.14.108-ti-r134
    kernel /boot/vmlinuz-4.14.108-ti-r134
    append console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100
    fdtdir /boot/dtbs/4.14.108-ti-r134/
```

Regards,

RobertCNelson, 2020-05-15:
Close:

```
sudo apt install bison flex
```

```
git clone -b v2020.07-rc2 https://github.com/u-boot/u-boot --depth=1
cd u-boot/
```

```
make ARCH=arm CROSS_COMPILE= distclean
make ARCH=arm CROSS_COMPILE= am335x_evm_defconfig
make ARCH=arm CROSS_COMPILE=
```

```
sudo dd if=/home/debian/u-boot/MLO of=/dev/mmcblk0 count=1 seek=1 bs=128k
sudo dd if=/home/debian/u-boot/u-boot.img of=/dev/mmcblk0 count=2 seek=1 bs=384k
```

```
sudo mkdir /boot/extlinux/
sudo nano /boot/extlinux/extlinux.conf
```

```
label Linux 4.14.108-ti-r131
    kernel /boot/vmlinuz-4.14.108-ti-r134
    append console=ttyO0,115200n8 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100
    fdtdir /boot/dtbs/4.14.108-ti-r131/
```

```
U-Boot SPL 2020.07-rc2 (May 15 2020 - 15:46:31 +0000)
Trying to boot from MMC1

U-Boot 2020.07-rc2 (May 15 2020 - 15:46:31 +0000)

CPU  : AM335X-GP rev 2.1
Model: TI AM335x BeagleBone Black
DRAM:  512 MiB
WDT:   Started with servicing (60s timeout)
NAND:  0 MiB
MMC:   OMAP SD/MMC: 0, OMAP SD/MMC: 1
Loading Environment from FAT... Unable to use mmc 0:1... <ethaddr> not set. Validating first E-fuse MAC
Net:   eth0: ethernet@4a100000
Warning: usb_ether MAC addresses don't match:
Address in ROM is		de:ad:be:ef:00:01
Address in environment is	b0:d5:cc:fc:03:fe
, eth1: usb_ether
Hit any key to stop autoboot:  0 
switch to partitions #0, OK
mmc0 is current device
SD/MMC found on device 0
** Unrecognized filesystem type **
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found /boot/extlinux/extlinux.conf
Retrieving file: /boot/extlinux/extlinux.conf
265 bytes read in 4 ms (64.5 KiB/s)
1:	Linux 4.14.108-ti-r131
Retrieving file: /boot/vmlinuz-4.14.108-ti-r134
Skipping Linux 4.14.108-ti-r131 for failure retrieving kernel
SCRIPT FAILED: continuing...
switch to partitions #0, OK
mmc0 is current device
SD/MMC found on device 0
** Invalid partition 2 **
switch to partitions #0, OK
mmc1(part 0) is current device
** No partition table - mmc 1 **
switch to partitions #0, OK
mmc1(part 0) is current device
SD/MMC found on device 1
** No partition table - mmc 1 **
## Error: "bootcmd_nand0" not defined
starting USB...
Bus usb@47401800: Port not available.
link up on port 0, speed 100, full duplex
BOOTP broadcast 1
BOOTP broadcast 2
BOOTP broadcast 3
BOOTP broadcast 4
BOOTP broadcast 5
DHCP client bound to address 192.168.1.172 (4801 ms)
Using ethernet@4a100000 device
TFTP from server 192.168.1.1; our IP address is 192.168.1.172
```

RobertCNelson, 2020-05-15:
Quick hack to make it easier to convert..

https://github.com/RobertCNelson/boot-scripts/blob/master/tools/generate_extlinux.sh

```
sudo /opt/scripts/tools/generate_extlinux.sh
```

```
debian@beaglebone:~$ cat /boot/extlinux/extlinux.conf 
label Linux 4.14.108-ti-r131
    kernel /boot/vmlinuz-4.14.108-ti-r131
    append console=ttyO0,115200n8 root=/dev/mmcblk0p1 ro rootfstype=ext4 coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet
    fdtdir /boot/dtbs/4.14.108-ti-r131/
```

Everything get's copied from /boot/uEnv.txt except , for what u-boot set's by default: "console=ttyO0,115200n8 root=/dev/mmcblk0p1 ro rootfstype=ext4"

console=ttyO0,115200n8 = this should now be set by default form the *.dtb

"root=/dev/mmcblk0p1 ro rootfstype=ext4" = u-boot currently set's this by decoding what drive we boot from...

RobertCNelson, 2020-05-15:
test, removed console=ttyO0,115200n8

```
debian@beaglebone:~$ cat /boot/extlinux/extlinux.conf 
label Linux 4.14.108-ti-r131
    kernel /boot/vmlinuz-4.14.108-ti-r131
    append root=/dev/mmcblk0p1 ro rootfstype=ext4 coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet
    fdtdir /boot/dtbs/4.14.108-ti-r131/
```

Nope, that breaks..

```
U-Boot SPL 2020.07-rc2 (May 15 2020 - 15:46:31 +0000)
Trying to boot from MMC1

U-Boot 2020.07-rc2 (May 15 2020 - 15:46:31 +0000)

CPU  : AM335X-GP rev 2.1
Model: TI AM335x BeagleBone Black
DRAM:  512 MiB
WDT:   Started with servicing (60s timeout)
NAND:  0 MiB
MMC:   OMAP SD/MMC: 0, OMAP SD/MMC: 1
Loading Environment from FAT... Unable to use mmc 0:1... <ethaddr> not set. Validating first E-fuse MAC
Net:   eth0: ethernet@4a100000
Warning: usb_ether MAC addresses don't match:
Address in ROM is		de:ad:be:ef:00:01
Address in environment is	b0:d5:cc:fc:03:fe
, eth1: usb_ether
Hit any key to stop autoboot:  0 
switch to partitions #0, OK
mmc0 is current device
SD/MMC found on device 0
** Unrecognized filesystem type **
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found /boot/extlinux/extlinux.conf
Retrieving file: /boot/extlinux/extlinux.conf
239 bytes read in 4 ms (57.6 KiB/s)
1:	Linux 4.14.108-ti-r131
Retrieving file: /boot/vmlinuz-4.14.108-ti-r131
9654768 bytes read in 618 ms (14.9 MiB/s)
append: root=/dev/mmcblk0p1 ro rootfstype=ext4 coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet
Retrieving file: /boot/dtbs/4.14.108-ti-r131/am335x-boneblack.dtb
60180 bytes read in 10 ms (5.7 MiB/s)
## Flattened Device Tree blob at 88000000
   Booting using the fdt blob at 0x88000000
   Loading Device Tree to 8ffee000, end 8ffffb13 ... OK

Starting kernel ...

(nothing.. but led's blinking..)
```

RobertCNelson, 2020-05-15:
This also works with our forked u-boot:

```
U-Boot SPL 2019.04-00002-g07d5700e21 (Mar 06 2020 - 11:24:55 -0600)
Trying to boot from MMC1
Loading Environment from EXT4... ** File not found /boot/uboot.env **

** Unable to read "/boot/uboot.env" from mmc0:1 **

U-Boot 2019.04-00002-g07d5700e21 (Mar 06 2020 - 11:24:55 -0600), Build: jenkins-github_Bootloader-Builder-137

CPU  : AM335X-GP rev 2.1
I2C:   ready
DRAM:  512 MiB
No match for driver 'omap_hsmmc'
No match for driver 'omap_hsmmc'
Some drivers were not found
Reset Source: Global warm SW reset has occurred.
Reset Source: Power-on reset has occurred.
RTC 32KCLK Source: External.
MMC:   OMAP SD/MMC: 0, OMAP SD/MMC: 1
Loading Environment from EXT4... ** File not found /boot/uboot.env **

** Unable to read "/boot/uboot.env" from mmc0:1 **
Board: BeagleBone Black
<ethaddr> not set. Validating first E-fuse MAC
BeagleBone Black:
BeagleBone: cape eeprom: i2c_probe: 0x54:
BeagleBone: cape eeprom: i2c_probe: 0x55:
BeagleBone: cape eeprom: i2c_probe: 0x56:
BeagleBone: cape eeprom: i2c_probe: 0x57:
Net:   eth0: MII MODE
cpsw, usb_ether
Press SPACE to abort autoboot in 0 seconds
board_name=[A335BNLT] ...
board_rev=[00C0] ...
switch to partitions #0, OK
mmc0 is current device
SD/MMC found on device 0
switch to partitions #0, OK
mmc0 is current device
Scanning mmc 0:1...
Found /boot/extlinux/extlinux.conf
Retrieving file: /boot/extlinux/extlinux.conf
262 bytes read in 57 ms (3.9 KiB/s)
1:	Linux 4.14.108-ti-r131
Retrieving file: /boot/vmlinuz-4.14.108-ti-r131
9654768 bytes read in 655 ms (14.1 MiB/s)
append: console=ttyO0,115200n8 root=/dev/mmcblk0p1 ro rootfstype=ext4 coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet
Retrieving file: /boot/dtbs/4.14.108-ti-r131/am335x-boneblack.dtb
60180 bytes read in 196 ms (299.8 KiB/s)
## Flattened Device Tree blob at 88000000
   Booting using the fdt blob at 0x88000000
   Loading Device Tree to 8ffee000, end 8ffffb13 ... OK

Starting kernel ...

[    0.000772] timer_probe: no matching timers found
[    0.398702] wkup_m3_ipc 44e11324.wkup_m3_ipc: could not get rproc handle
[    0.638248] omap_voltage_late_init: Voltage driver support not added
```

RobertCNelson, 2020-05-15:
@jadonk i'm a little confused, which downstream distro is having problems, do we have a direct user question?  As it looks we've had /boot/extlinux/extlinux.conf support now for over a  good year. Haven't tested any version of u-boot before v2019.04..

Regards,

jadonk, 2020-05-15:
I don't know of any having problems, just keep forgetting that it is in there. I'm checking buildroot right now.

jadonk, 2020-05-15:
k, so it looks like bone_capemgr just needs to be added to mainline u-boot somehow---and we need to start generating extlinux.conf when installing kernels.

RobertCNelson, 2020-05-15:
In buildroot/u-boot, beaglebone's post-image.sh needs this removed:

```
# copy the uEnv.txt to the output/images directory
cp board/beaglebone/uEnv.txt $BINARIES_DIR/uEnv.txt
```

```
install -m 0644 -D $BOARD_DIR/extlinux.conf $TARGET_DIR/boot/extlinux/extlinux.conf
```

I'm guessing buildroot has an extlinux config?