beagleboard/Latest-Images #42: Switch both usb0/usb1 to cdc-ncm and tie usb1 to connman

jadonk, 2020-05-15:
**Is your feature request related to a problem? Please describe.**
People always want to know how to get their USB-only-connected Beagles to use the host's Internet connection. They need to issue 'ip addr flush dev usb1; dhclient usb1'. Of course, if we use cdc-ncm, we shouldn't need both usb0 and usb1 to ensure they get a port that works on their OS as all OSes should support it. That means, we'd have a free port to configure as a DHCP client by default and perhaps even have it managed by connman to facilitate the DHCP requests and routing (though I'm less sure on that one).

**Describe the solution you'd like**
* Switch both usb0 and usb1 to use cdc-ncm by default
* Work on the descriptors to make sure usb0 comes up across latest Mac, Linux and *Windoze*
* Switch usb1 to not be a hard-coded IP address or providing a DHCP server and instead make it look like an unconfigured adapter to the host such that if Internet Connection Sharing is enabled on the host, this port will grab an address and start routing stuff out of it.

**Describe alternatives you've considered**
Not sure which port should be swapped. Not sure of other alternatives.

**Additional context**
Got nothing.

RobertCNelson, 2020-05-15:
[image]

Who wants to do the Windoze side of this?

jadonk, 2020-05-16:
https://docs.microsoft.com/en-us/windows-hardware/drivers/usbcon/microsoft-os-2-0-descriptors-specification

jadonk, 2020-05-16:
https://ostconf.com/system/attachments/files/000/001/537/original/WYOGv3.pdf?1540914672

Poking around for a Compatible ID using NCM:
USB\CLASS_FF&SUBCLASS_03&PROT_16

https://docs.microsoft.com/en-us/windows-hardware/drivers/network/mb-union-function-descriptors

Also FYI https://github.com/microsoft/NCM-Driver-for-Windows