beagleboard/Latest-Images #44: Boot screen usability improvments now that the GUI is gone

jadonk, 2020-05-20:
**Is your feature request related to a problem? Please describe.**
Silly boot display branding/usability stuff:
* Support URL: Can we use http://bbb.io/debian?
* Boot icon: Can we use Boris rather than Tux?
* IP address for Cloud9: Can we display the magic URL?
* Dogtag: Can we be more specific on the image by providing a URL/SHA?

**Describe the solution you'd like**
In /etc/issue, can we switch from elinux as the first page to something easier to type and hosted by BeagleBoard.org?

IP for Cloud9: If we have wireless, can we print the SSID, password and https://192.168.8.1 magic URL? If we have ethernet, is there a chance we can have an IP address before /etc/issue is printed? Don't want to slow down boot so much, but really want to simplify things if people attach a monitor or LCD. Alternative is to super-impose the IP address once it arrives.

I'm less concerned about the dogtag, just want to make sure when people go to the support URL, they can figure out the image that was used as the starting point.

**Describe alternatives you've considered**
Not sure here.

**Additional context**
N/A. You've seen the boot screen.

RobertCNelson, 2020-05-20:
<cough> LXQt Buster  https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#Debian_Buster_LXQt_Snapshot ;) </ cough>

Support URL
Yeah i have no problem switching the Support URL..  it would be nice if some of the useful FAQ's where copied over: https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#FAQ

Boot Icon:
sure..

IP address:
We tried this, "login" comes up "way" before any ip address is set..  Ubuntu did this a few years ago, this was the "2 minute" login wait era, that we hacked out..

Dogtag: for 2020-04-06 i had extended this quite a bit...
```
debian@bbb-pwr01-ser09:~$ cat /etc/dogtag 
BeagleBoard.org Debian Buster IoT Image 2020-04-06
```

SHA will be problem, url i could fudge?

Regards,

RobertCNelson, 2020-05-21:
████████████████████████████████▀▀▀░¿╓╓▄▄░▀▀▀███████████████████████████████████
    ████████████████████████████▀;╖╢▒▒▒▒▒▒▒░█▄▒▒▒H╖░▀███████████████████████████████
    ███████████████████████▀▀▀, "╙╢▒▒▒▒▒▒▒▒▒▒▀█░▒▒▒▒╢╖▀█████████████████████████████
    █████████████████████▀ @▒▒▒▒▒▒╖ ║▒▒▒▒▒▒▒▒▒░█░▒▒▒▒▒╢ ▀███████████████████████████
    ████████████████████ ╓▒▒▒▒▒▒▒▒▒▒ ║▒▒▒▒▒▒▒▒▒░█▄░▒▒▒╜╜ ╙██████████████████████████
    ███████████████████ ╓▒▒▒▒▒▒▒▒▒▒▒╢ ▒▒╢▒▄`"╙▒▒░██░▒[└▀  "█████████████████████████
    ██████████████████ ╓▒▒▒▒▒▒▒▒▒▒▒▒▒┐╙▒  ╙   ]▒▒░████▄▄╖╖┐▀████████████████████████
    █████████████████ ╓▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒ ▒▒╖╖╖╖╢▒▒▒▒███████████▄▄▀▀███████████████████
    ████████████████  ▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒╖╙▒▒▒▒▒▒▒▒░████████████████▄▄▀████████████████
    ███████████████" ▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒ ║▒▒▒▒▒░▄██████████████████▀▀ ;▀▀████████████
    ██████████████ ,▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒╢ ╢▒▒▒▄█████████████████▀ ╓▄███▀└  ██████████
    █████████████ ╓▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒╖ ▒▒███████████████████  └└       ▐█████████
    ████████████U ▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒ ╙▒███████████████████▄          ██████████
    ████████████ ]▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒░▒╢ ▒█████████████████████▄▄╓╓▄▄▄▀ ██████████
    ████████████ ]▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒░░▒▒ ▒▐██████████████████████████▀  ██████████
    ████████████⌐ ╢▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒▒░░░░░▒╜ ▒░██████████████████████▀▀    ███████████
    █████████████  ║▒▒▒▒▒▒▒▒▒▒▒▒░░░░░░░▒╢ ╢▒▒░██████████████▀▀▀▀'  ,╓@╢▓ ▀██████████
    ██████████████▄  ╙▒▒▒▒▒▒░░░░░░░░▒▒╢╜╓╢▒▒▒▒▐█████└            ╙▒▒▒╢▒▒▌ ██████████
    █████████████████▄  `╙╜╜╢▒▒▒▒╢╜╜",╖╢▒▒▒▒▒▒███████▄▄▄,          ╩▒▒╢╫ ▄██████████
    █████████████ ▄▄└█████L ╔╖╖╖╖╢╢▒▒▒▒▒▒▒▒░▄██▀▀▀▀▀▀▀▀▀▀▀▀└▄▄▄▄██▄    ╓████████████
    ████████████▌ ███"████ ╓▒▒▒▒▒▄███████████████████░▒▒▒▒▒ ▐███████████████████████
    █████████████ ▐█░ ▐██C ▒▒▒▒░██████████████████████░▒▒▒▒▒ ███████████████████████
    █████████████▌ ╙▒▒ ▀█ ]▒▒▒▒▐███████████████████████╣▒▒▒▒ ███████████████████████
    ██████████████▄ ║▒▒,▀ ]▒▒▒▒▀███████████████████████ ▒▒▒▒[▐██████████████████████
    ███████████████▄ ╙▒▒╖ ╠▒▒▒[ ███████████████████████ ▒▒▒▒▒ ██████████████████████
    ████████████████▄ ╙▒▒▒▒▒▒▒[ ███████████████████████ ║▒▒▒▒ ██████████████████████
    ██████████████████  ╢▒▒▒▒▒L ██████ █████████ ██████ ║▒▒▒[▐██████████████████████
    ███████████████████▄ ╙▒▒▒▒h ██████ ████████▌ ██████ ]▒▒▒⌐▐██████████████████████
    █████████████████████  ▒▒▒  ██████ ████████▌ ██████ ]▒▒▒ ███████████████████████
    ██████████████████████ └▒▒ j██████ ▐███████U ██████ ]▒▒C▐███████████████████████
    ███████████████████▀╙,╗@▒▒ ▐██████ ▐███████═▐██████r ▒▒@╖└▀█████████████████████
    ██████████████████▀ ▒░▒▒▒▒ ▐██████rj███████ ▐██████▌ ▒▒▒▒▒╖ ████████████████████
    █████████████████▌ ]▒▒▒▒▒ ,███████▌ ▀▀▀▀▀▀▀ ████████ ╙▒▒▒▒▒▒▐███████████████████
    ██████████████████  ╢▒▒▒ ╒█████████ ▐████▌  █████████, ▒▒▒╢`▐███████████████████
    ███████████████████▄▄▄   █████████  ██████  '█████████  ▄▄▄█████████████████████
    ███████████████████████  └▀▀▀▀▀└ ,▄████████▄,  "▀▀▀▀▀└ ▄████████████████████████
    
---
^[ [^ascii ^art ^generator](http://asciiart.club) ^]

jadonk, 2020-05-30:
> < cough> LXQt Buster  https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#Debian_Buster_LXQt_Snapshot ;) </ cough>
>
>
>
> Support URL
>
> Yeah i have no problem switching the Support URL..  it would be [...]

I will copy FAQs to bbb.io/debian

> 
> 
> Boot Icon:
> 
> sure..
> 
> 
> 
> IP address:
> 
> We tried this, "login" comes up "way" before any ip address is set..  Ubuntu did this a few years ago, this was the "2 minute" login wait era, that we hacked out..
> 

What does \4 do in /etc/issue before there is an IP?

> 
> 
> Dogtag: for 2020-04-06 i had extended this quite a bit...
> 
> ```
> 
> debian@bbb-pwr01-ser09:~$ cat /etc/dogtag 
> 
> BeagleBoard.org Debian Buster IoT Image 2020-04-06
> 
> ```
> 
> 
> 
> SHA will be problem, url i could fudge?

The translation to the filename is non-trivial. 

> 
> 
> 
> Regards,
> 
> 
> 
> 
> 
> 
> 
>