beagleboard/Latest-Images #45: [pwm] v5.4.x: retest mainline udev/pwm rule...

RobertCNelson, 2020-05-21:
This may now be fixed.. https://github.com/beagleboard/linux/issues/124#issuecomment-631803740

RobertCNelson, 2020-05-22:
```
debian@bbb-pwr01-ser09:~$ uname -r
5.4.38-ti-r8
```

```
debian@bbb-pwr01-ser09:/dev/pwm$ ls -lha
total 0
drwxr-xr-x  2 root root  220 May 22 13:56 .
drwxr-xr-x 18 root root 3.4K May 22 13:55 ..
lrwxrwxrwx  1 root root   22 May 22 13:56 ecap0 -> /sys/class/pwm/pwm-0:0
lrwxrwxrwx  1 root root   22 May 22 13:56 ecap1 -> /sys/class/pwm/pwm-3:0
lrwxrwxrwx  1 root root   22 May 22 13:56 ecap2 -> /sys/class/pwm/pwm-6:0
lrwxrwxrwx  1 root root   22 May 22 13:56 ehrpwm0a -> /sys/class/pwm/pwm-1:0
lrwxrwxrwx  1 root root   22 May 22 13:56 ehrpwm0b -> /sys/class/pwm/pwm-1:1
lrwxrwxrwx  1 root root   22 May 22 13:56 ehrpwm1a -> /sys/class/pwm/pwm-4:0
lrwxrwxrwx  1 root root   22 May 22 13:56 ehrpwm1b -> /sys/class/pwm/pwm-4:1
lrwxrwxrwx  1 root root   22 May 22 13:56 ehrpwm2a -> /sys/class/pwm/pwm-7:0
lrwxrwxrwx  1 root root   22 May 22 13:56 ehrpwm2b -> /sys/class/pwm/pwm-7:1
```

````
debian@bbb-pwr01-ser09:/dev/pwm$ cd /sys/class/pwm/pwm-1:0
debian@bbb-pwr01-ser09:/sys/class/pwm/pwm-1:0$ ls -lha
total 0
drwxrwxr-x 3 root pwm    0 May 22 13:55 .
drwxrwxr-x 5 root pwm    0 May 22 13:55 ..
-r--r--r-- 1 root pwm 4.0K May 22 13:55 capture
lrwxrwxrwx 1 root pwm    0 May 22 13:55 device -> ../../pwmchip1
-rw-rw-r-- 1 root pwm 4.0K May 22 13:55 duty_cycle
-rw-rw-r-- 1 root pwm 4.0K May 22 13:55 enable
-rw-rw-r-- 1 root pwm 4.0K May 22 13:55 period
-rw-rw-r-- 1 root pwm 4.0K May 22 13:55 polarity
drwxrwxr-x 2 root pwm    0 May 22 13:55 power
lrwxrwxrwx 1 root pwm    0 May 22 13:55 subsystem -> ../../../../../../../../../../../class/pwm
-rw-rw-r-- 1 root pwm 4.0K May 22 13:55 uevent
```

pdp7, 2020-05-22:
@RobertCNelson here is what I see:

```
root@beaglebone:/sys/class/pwm# uname -a
Linux beaglebone 5.7.0-rc4-omap2plus-r0 #3 SMP Thu May 21 21:57:33 CEST 2020 armv7l GNU/Linux
```

```
root@beaglebone:/sys/class/pwm# ls -la pwmchip?/pwm?/duty_cycle 
-rw-rw-r-- 1 root gpio 4096 Nov  3  2016 pwmchip0/pwm0/duty_cycle
-rw-rw-r-- 1 root gpio 4096 Nov  3  2016 pwmchip0/pwm1/duty_cycle
-rw-rw-r-- 1 root gpio 4096 Nov  3  2016 pwmchip2/pwm0/duty_cycle
-rw-rw-r-- 1 root gpio 4096 Nov  3  2016 pwmchip2/pwm1/duty_cycle
```

group gpio is because of uevent?

RobertCNelson, 2020-05-22:
```
#!/bin/bash

if [ ! -d /dev/pwm/ ] ; then
	mkdir -p /dev/pwm/
fi

base_dir="/sys/devices/platform/ocp/48000000.interconnect/48000000.interconnect:segment@300000/"

check_module_dir (){
	if [ -d ${base_dir}/${module}/${pwm_node} ] ; then
	        label=$(ls ${base_dir}/${module}/${pwm_node}/ | grep pwm || true)
	        if [ ! "x${label}" = "x" ] ; then
        	        subnode=$(ls ${base_dir}/${module}/${pwm_node}/${label} | grep pwm | grep ${pwm_offset} || true)
	                if [ ! "x${subnode}" = "x" ] ; then
                	        echo "/bin/ln -sT /sys/class/pwm/${label}/${subnode} /dev/pwm/${pwm_label}"
        	                /bin/ln -sT /sys/class/pwm/${label}/${subnode} /dev/pwm/${pwm_label}
	                fi
        	fi
	        echo $label
	fi

}

module="48300000.target-module/48300000.epwmss"
pwm_node="48300100.ecap/pwm"
pwm_offset=":0"
pwm_label="ecap0"
check_module_dir

pwm_node="48300200.pwm/pwm"
pwm_offset=":0"
pwm_label="ehrpwm0a"
check_module_dir

pwm_node="48300200.pwm/pwm"
pwm_offset=":1"
pwm_label="ehrpwm0b"
check_module_dir

module="48302000.target-module/48302000.epwmss"
pwm_node="48302100.ecap/pwm"
pwm_offset=":0"
pwm_label="ecap1"
check_module_dir

pwm_node="48302200.pwm/pwm"
pwm_offset=":0"
pwm_label="ehrpwm1a"
check_module_dir

pwm_node="48302200.pwm/pwm"
pwm_offset=":1"
pwm_label="ehrpwm1b"
check_module_dir

module="48304000.target-module/48304000.epwmss"
pwm_node="48304100.ecap/pwm"
pwm_offset=":0"
pwm_label="ecap2"
check_module_dir

pwm_node="48304200.pwm/pwm"
pwm_offset=":0"
pwm_label="ehrpwm2a"
check_module_dir

pwm_node="48304200.pwm/pwm"
pwm_offset=":1"
pwm_label="ehrpwm2b"
check_module_dir
```

Rework of zmatt's udev rule, but in bash, so i can remove the pwm_channel uevent

pdp7, 2020-05-23:
@RobertCNelson is this meant to run on a system without the kernel patch?

RobertCNelson, 2020-05-23:
No, not yet.. that implants the same thing as zmatt's udev rule, I need to extend it to the mainline condition.

RobertCNelson, 2020-05-23:
I was heading back to work, so I did not want to lose my progress.

RobertCNelson, 2020-07-09:
@pdp7 this turned out to be a simple udev rule change.. add -> change..

https://github.com/beagleboard/customizations/commit/ca1c2485303c3fd5e506ef2abb41a2bc63957f02

closed 2020-07-09 by RobertCNelson