beagleboard/Latest-Images #46: Kernel used in Debian 10 images

JoseGoncalves, 2020-05-22:
Latest Debian 10.4 console images have kernel 5.4 installed by default.
Is this on purpose?
It does not seem right to me, has mainline Debian 10 disto uses kernel 4.19. 
If it's on purpose, [documentation](https://elinux.org/Beagleboard:Latest-images-testing#Debian_10_.28Buster.29_Console) should be updated accordingly.

RobertCNelson, 2020-05-22:
@JoseGoncalves yeah my wiki bash update script failed..

it has already messed up the version stamp..

https://elinux.org/index.php?title=Beagleboard%3ABeagleBoneBlack_Debian&type=revision&diff=516301&oldid=516251

it should have also had 5.4.x for the console..

i'm targeting v5.4.x for all image for the August release.

Regards,

JoseGoncalves, 2020-06-22:
Hi @RobertCNelson. Just to warn you that [Beagleboard:Latest-images-testing](https://elinux.org/Beagleboard:Latest-images-testing) last update was in April, 6. Stil issues with the wiki update script?

RobertCNelson, 2020-06-22:
hi @JoseGoncalves that page won't be updated till next week, when i push out alpha/rc1 for the August 2020 release..

closed 2020-08-23 by JoseGoncalves