beagleboard/Latest-Images #48: Beagleboard (original) not supported anymore?

nokceb, 2020-05-25:
Tested Latest-image (stable) on https://beagleboard.org/latest-images for beagleboard (original), which is listed as supporting Beagleboard,Beagleboard-xm

First, beagleboard older uboot does not load correctly, but by holding down the user button during boot SD card uboot can be loaded. Result is: Fails to boot past DTB.

OMAP3/DM3730 Debian 9.5 2018-10-07 4GB SD LXQT

```
U-Boot SPL 2018.09-00001-ge10814359b (Sep 10 2018 - 19:52:59 -0500)
Trying to boot from MMC1
spl_load_image_fat_os: error reading image args, err - -2

U-Boot 2018.09-00001-ge10814359b (Sep 10 2018 - 19:52:59 -0500), Build: jenkins-github_Bootloader-Builder-65

OMAP3530-GP ES3.1, CPU-OPP2, L3-165MHz, Max CPU Clock 600 MHz
Model: TI OMAP3 BeagleBoard
OMAP3 Beagle board + LPDDR/NAND
I2C:   ready
DRAM:  128 MiB
NAND:  512 MiB
MMC:   OMAP SD/MMC: 0
Loading Environment from NAND... ecc unrecoverable error
ecc unrecoverable error
ecc unrecoverable error
ecc unrecoverable error
ecc unrecoverable error
ecc unrecoverable error
ecc unrecoverable error
ecc unrecoverable error
ecc unrecoverable error
ecc unrecoverable error
ecc unrecoverable error
ecc unrecoverable error
ecc unrecoverable error
ecc unrecoverable error
ecc unrecoverable error
ecc unrecoverable error
ecc unrecoverable error
ecc unrecoverable error
ecc unrecoverable error
ecc unrecoverable error
ecc unrecoverable error
ecc unrecoverable error
ecc unrecoverable error
ecc unrecoverable error
ecc unrecoverable error
ecc unrecoverable error
ecc unrecoverable error
ecc unrecoverable error
ecc unrecoverable error
ecc unrecoverable error
ecc unrecoverable error
ecc unrecoverable error
ecc unrecoverable error
ecc unrecoverable error
ecc unrecoverable error
ecc unrecoverable error
ecc unrecoverable error
ecc unrecoverable error
ecc unrecoverable error
ecc unrecoverable error
ecc unrecoverable error
ecc unrecoverable error
ecc unrecoverable error
ecc unrecoverable error
ecc unrecoverable error
ecc unrecoverable error
ecc unrecoverable error
ecc unrecoverable error
ecc unrecoverable error
ecc unrecoverable error
[... 665 lines cut ...]
[   91.650360] omap-twl4030 sound: ASoC: failed to init link TWL4030 HiFi: -517
[   91.682861] omap-twl4030 sound: devm_snd_soc_register_card() failed: -517
[   98.784179] OF: graph: no port node found in /ocp@68000000/isp@480bc000/ports
[   99.171203] OF: graph: no port node found in /ocp@68000000/isp@480bc000/ports
[   99.536865] OF: graph: no port node found in /ocp@68000000/isp@480bc000/ports
[   99.896575] omap-twl4030 sound: ASoC: failed to init link TWL4030 HiFi: -517
[   99.954559] omap-twl4030 sound: devm_snd_soc_register_card() failed: -517
[  100.555328] omap-twl4030 sound: ASoC: failed to init link TWL4030 HiFi: -517
[  100.586853] omap-twl4030 sound: devm_snd_soc_register_card() failed: -517
[  112.805816] omap-twl4030 sound: ASoC: failed to init link TWL4030 HiFi: -517
[  112.858947] omap-twl4030 sound: devm_snd_soc_register_card() failed: -517
[  117.147308] omap-twl4030 sound: ASoC: failed to init link TWL4030 HiFi: -517
[  117.184814] omap-twl4030 sound: devm_snd_soc_register_card() failed: -517
[  117.239318] omap-twl4030 sound: ASoC: failed to init link TWL4030 HiFi: -517
[  117.291534] omap-twl4030 sound: devm_snd_soc_register_card() failed: -517
[  117.846771] omap-twl4030 sound: ASoC: failed to init link TWL4030 HiFi: -517
[  117.894775] omap-twl4030 sound: devm_snd_soc_register_card() failed: -517
[  118.320587] omap-twl4030 sound: ASoC: failed to init link TWL4030 HiFi: -517
[  118.354858] omap-twl4030 sound: devm_snd_soc_register_card() failed: -517
[  118.853088] omap-twl4030 sound: ASoC: failed to init link TWL4030 HiFi: -517
[  118.894622] omap-twl4030 sound: devm_snd_soc_register_card() failed: -517

```

Item was discussed briefly in IRC. Creating issue was recommended. It should be noted that the old angstrom image was functional without issue. (Angstrom Distribution (BeagleBoard and BeagleBoard-xM - 4GB SD) 2012-01-11).

nokceb, 2020-05-25:
I was able to get this image to boot:
https://elinux.org/Beagleboard:BeagleBoneBlack_Debian#Debian_Buster_Console_Snapshot_for_BeagleBoard-xM

Though there is a ~1 minute delay. Not bad. Some of the same errors regarding the (audio codec?) TWL4030 appear.

RobertCNelson, 2020-05-25:
@nokceb lack on onboard ethernet makes startup times laggy..

Sorry, I rarely test this board anymore..  We just use a very close mainline kernels (enable spidev and remove the broken cortex-a8 debug node..)

Regards,