beagleboard/Latest-Images #49: Help wanted for Adafruit_BBIO python library

pdp7, 2020-06-09:
There are several issues that people familiar with Python and BeagleBone could help with:

https://github.com/adafruit/adafruit-beaglebone-io-python/issues?q=is%3Aissue+is%3Aopen+label%3A%22help+wanted%22

Note: some issues will need familiar with C as the Python library uses C extensions.

pdp7, 2020-06-09:
@jadonk per your request

jadonk, 2020-06-10:
@pdp7 Thanks! Can you create and apply a "good first issue" label to some issues on that repo?

pdp7, 2020-06-10:
I have 'help-wanted' on them but I'm not sure what would be a good first issue.

Maybe this one to update the tutorials, so I added it there:
https://github.com/adafruit/adafruit-beaglebone-io-python/issues/325

These PWM issues might be good place to start actual development issues but probably do require some knowledge to be learn for someone brand new to Beagle:

- [BeagleBone Blue PWM](https://github.com/adafruit/adafruit-beaglebone-io-python/issues/313)
- [PWM enablement: P8_07, P8_08, P8_09 and P8_10](https://github.com/adafruit/adafruit-beaglebone-io-python/issues/229)

The BBAI support is probably the most important enhancemet but it would take some effort for a new person I think: [Add support for BeagleBone AI with Adafruit BBIO.GPIO](https://github.com/adafruit/adafruit-beaglebone-io-python/issues/317)

MarkAYoder, 2020-08-08:
@pdp7 I have a start at the PWM module.  [PWMmay.py](https://github.com/MarkAYoder/BeagleBoard-exercises/blob/master/python/PWMmay.py).. Simple test code is: [PWMtest.py](https://github.com/MarkAYoder/BeagleBoard-exercises/blob/master/python/PWMtest.py)

silver2row, 2020-12-10:
Hello,

I am back! I know I have not participated at all really. I try to purchase the hardware and look for new, exciting ways to familiarize myself w/ Linux, the equipment, and let me say, the nice images. I am not sure who is behind it all but thank you thus far.

...

Now, about this library and adding support. I think I can. I will try. So, name it and I will research the ideas in my time. I hope no one beats me to it. I am off to read about PWM.py or PWM.cpp or PWM.c libraries/modules. I know I get giving and taking and trying and passing/failing but...

Seth

silver2row, 2020-12-11:
Hello,

I have been researching this idea of a pwm.py file as a lib. I have come across many interesting facts in the kernel. It seems one must have a device tree and a board_setup_file. Do I need to supply this board_setup_file? I am just browsing in the kernel and reading as much as I can attain for now. I also see that @MarkAYoder has put together a module already to be tested. This task is already teaching me a lot about config-pin, dt, and how to implement PWM as a lib. One thing I have not researched is the info. about the pwmchip in the am335x TRM. So, that is next.

...

It seems that the kernel does not specify the am335x SoC in the https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/tree/drivers/pwm?h=v4.19.162 listings. Oh? Should I make this task available, the task of a pwmchip lib, for future kernels from beagleboard.org, e.g. 5.4.x and etc?

Seth

silver2row, 2021-05-31:
Hello @MarkAYoder ,

Seth here. I am a novice in the embedded world compared to you fellows/ladies. Anyway, I have been following the PWMmay.py file(s) and looking to see if I could replicate or try my own config. of the file in time.

I started a file in C to handle PWM but I was getting permission errors for exporting the PWM chip in question. I stopped and freshly started again. 

...

Long story, short! Are people still after the PWM lib. in python?

Seth