beagleboard/Latest-Images #51: Removing BoneScript from cloud9-examples

jadonk, 2020-06-10:
See https://github.com/beagleboard/cloud9-examples/issues/38

There are still some places where it makes sense to use BoneScript, namely for RPC, but those examples need to be:
* Updated for the new socket location and years of system changes
* Secured so that people cannot use the RPC port for execution of arbitrary code

pala-sh, 2020-06-13:
Not sure what exactly you're trying to achieve, can you tell what is your objective

jadonk, 2020-06-19:
The point is to try to teach people about Linux and the abstractions it provides, rather than to create new abstractions. There's really not much need and it is best to teach people how to figure the information they need to use Linux in any programming language.