beagleboard/Latest-Images #53: PocketBeagle/TechLab/pushbutton.js depends on epoll being installed

MarkAYoder, 2020-07-01:
**Describe the bug**
PocketBeagle/TechLab/pushbutton.js depends on **epoll** which isn't installed on the "BeagleBoard.org Debian Buster IoT Image 2020-04-06" image.

A simple 
`npm install epoll`
does it.

RobertCNelson, 2020-07-01:
Thanks @MarkAYoder sadly epoll isn't packaged, so this will be a little fun to script, but i'll work on it..

MarkAYoder, 2020-07-02:
@RobertCNelson we should just drop that javascript example.  The python version works.

jadonk, 2020-07-22:
@MarkAYoder it is still useful having a JavaScript example for those wishing to learn how to do things with JavaScript.

MarkAYoder, 2020-07-24:
@jadonk I think it's worth keeping the example, but not worth preinstalling epoll.  We could include a comment that says how to install it.