beagleboard/Latest-Images #54: PocketBeagle/TechLab/fadeLED.js not working

MarkAYoder, 2020-07-01:
**Describe the bug**
`cd PocketBeagle/TechLab/`
The `setup.sh` file isn't modifying `/boot/uEnv.txt`

When `uEnv.txt` is hand edited `dmesg` returns:
`[ +12.972325] leds_pwm techlab_rgb: unable to request PWM for techlab::blue: -16
[  +0.099681] leds_pwm: probe of techlab_rgb failed with error -16`

**Describe which image you are running (example: bone-debian-10.2-iot-armhf-2020-02-03-4gb.img.xz) **
**BeagleBoard.org Debian Buster IoT Image 2020-04-06**

**Describe your hardware**
PocketBeagle with TechLab cape.

Same results for 5.4, 4.19 and 4.14 kernels.

RobertCNelson, 2020-07-01:
This is related to: https://github.com/beagleboard/Latest-Images/issues/45  i need the script for v4.14.x/4.19.x/v5.4.x then nuke the udev rule..

RobertCNelson, 2020-07-09:
fix pushed out! ;)

closed 2020-07-09 by RobertCNelson