beagleboard/Latest-Images #56: TechLab mma8452 accelerometer on being installed.

MarkAYoder, 2020-07-10:
**Describe the bug**
No such file or directory: '/sys/bus/i2c/drivers/mma8452/2-001c/iio:device1/in_accel_x_raw'

**Describe which image you are running (example: bone-debian-10.2-iot-armhf-2020-02-03-4gb.img.xz) **
BeagleBoard.org Debian Buster IoT Image 2020-07-06

**Describe your hardware**
PocketBeagle with TechLab board

**Please run this built-in script**
```sudo /opt/scripts/tools/version.sh```
sudo: unable to resolve host pocket: No address associated with hostname
git:/opt/scripts/:[029041f6866049997bbfd2c7667b3c6e8c95201c]
eeprom:[A335PBGL00A21750EPB00481]
model:[TI_AM335x_PocketBeagle]
dogtag:[BeagleBoard.org Debian Buster IoT Image 2020-07-06]
bootloader:[microSD]:[/dev/mmcblk0]:[U-Boot 2019.04-00002-g31a8ae0206]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-pocketbeagle.dts]
UBOOT: Loaded Overlay:[BB-ADC-00A0]
kernel:[5.4.47-ti-r12]
nodejs:[v10.21.0]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20200709.0-0~buster+20200709]
pkg:[bb-wl18xx-firmware]:[1.20200702.0-0~buster+20200702]
pkg:[kmod]:[26-1]
pkg:[librobotcontrol]:[1.0.4-git20190227.1-0rcnee0~buster+20190327]
pkg:[firmware-ti-connectivity]:[20190717-2rcnee1~buster+20200305]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal input bluetooth netdev i2c gpio admin spi iio docker tisdk weston-launch xenomai cloud9ide pwm eqep remoteproc]
cmdline:[console=ttyO0,115200n8 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet]
dmesg | grep remote
[    2.015444] remoteproc remoteproc0: wkup_m3 is available
[    2.202051] remoteproc remoteproc0: powering up wkup_m3
[    2.202700] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[    2.204511] remoteproc remoteproc0: remote processor wkup_m3 is now up
dmesg | grep pru
dmesg | grep pinctrl-single
[    1.721117] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
[    1.730966] gpio-of-helper ocp:cape-universal: ready
lsusb
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END

**To Reproduce**
Steps to reproduce the behavior:
1. **cd /var/lib/cloud9/PocketBeagle/TechLab**
2. **./accelerometer.py** 
Traceback (most recent call last):
  File "./accelerometer.py", line 11, in <module>
    fdx = open(path + '/in_accel_x_raw', 'r')
FileNotFoundError: [Errno 2] No such file or directory: '/sys/bus/i2c/drivers/mma8452/2-001c/iio:device1/in_accel_x_raw'

RobertCNelson, 2020-07-10:
```
UBOOT: Booted Device-Tree:[am335x-pocketbeagle.dts]
UBOOT: Loaded Overlay:[BB-ADC-00A0]
```

Missing overlay?

MarkAYoder, 2020-07-10:
Looks like it, but I copied BBORG_TECHLAB-00A2.dtbo from a working version and put it in /lib/firmware and got the same results.  Is it not detecting the TechLAB?

git:/opt/scripts/:[029041f6866049997bbfd2c7667b3c6e8c95201c]
eeprom:[A335PBGL00A21750EPB00481]
model:[TI_AM335x_PocketBeagle]
dogtag:[BeagleBoard.org Debian Buster IoT Image 2020-07-06]
bootloader:[microSD]:[/dev/mmcblk0]:[U-Boot 2019.04-00002-g31a8ae0206]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-pocketbeagle.dts]
UBOOT: Loaded Overlay:[BB-ADC-00A0]
kernel:[5.4.47-ti-r12]
nodejs:[v10.21.0]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20200709.0-0~buster+20200709]
pkg:[bb-wl18xx-firmware]:[1.20200702.0-0~buster+20200702]
pkg:[kmod]:[26-1]
pkg:[librobotcontrol]:[1.0.4-git20190227.1-0rcnee0~buster+20190327]
pkg:[firmware-ti-connectivity]:[20190717-2rcnee1~buster+20200305]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal input bluetooth netdev i2c gpio admin spi iio docker tisdk weston-launch xenomai cloud9ide pwm eqep remoteproc]
cmdline:[console=ttyO0,115200n8 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet]
dmesg | grep remote
[    2.015245] remoteproc remoteproc0: wkup_m3 is available
[    2.201977] remoteproc remoteproc0: powering up wkup_m3
[    2.202632] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[    2.204430] remoteproc remoteproc0: remote processor wkup_m3 is now up
dmesg | grep pru
dmesg | grep pinctrl-single
[    1.719426] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
[    1.729401] gpio-of-helper ocp:cape-universal: ready
lsusb
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END

RobertCNelson, 2020-07-10:
eeprom is used yet, you need to specifiy the overlay for techlab..

jadonk, 2020-07-22:
Can we change this issue to add EEPROM support for PocketBeagle overlays?

MarkAYoder, 2020-08-05:
Looks like the same thing for **/sys/class/leds/techlab::seg**.

@RobertCNelson this was working before.  What's changed?

RobertCNelson, 2020-08-05:
@MarkAYoder what permissions does it have?

MarkAYoder, 2020-08-05:
@RobertCNelson  **/sys/class/leds/techlab::seg** doesn't even show up.

```sudo /opt/scripts/tools/version.sh 
sudo: unable to resolve host ece434: No address associated with hostname
[sudo] password for debian: 
git:/opt/scripts/:[7cdc270818b76d829d247cf05fe309c320a3929a]
eeprom:[A335PBGL00A21750EPB00481]
model:[TI_AM335x_PocketBeagle]
dogtag:[BeagleBoard.org Debian Buster IoT Image 2020-08-04]
bootloader:[microSD]:[/dev/mmcblk0]:[U-Boot 2019.04-00002-g23f263cc3f]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-pocketbeagle.dts]
UBOOT: Loaded Overlay:[AM335X-PRU-RPROC-4-19-TI-00A0]
UBOOT: Loaded Overlay:[BB-ADC-00A0]
kernel:[4.19.94-ti-r45]
nodejs:[v10.21.0]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-19-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20200722.0-0~buster+20200722]
pkg:[bb-wl18xx-firmware]:[1.20200702.0-0~buster+20200702]
pkg:[kmod]:[26-1]
pkg:[librobotcontrol]:[1.0.5-git20200715.0-0~buster+20200716]
pkg:[firmware-ti-connectivity]:[20190717-2rcnee1~buster+20200305]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal input bluetooth netdev i2c gpio admin spi iio docker tisdk weston-launch xenomai cloud9ide pwm eqep remoteproc]
cmdline:[console=ttyO0,115200n8 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet]
dmesg | grep remote
[   53.028962] remoteproc remoteproc0: wkup_m3 is available
[   53.213309] remoteproc remoteproc0: powering up wkup_m3
[   53.213341] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[   53.213622] remoteproc remoteproc0: remote processor wkup_m3 is now up
[   54.804469] remoteproc remoteproc1: 4a334000.pru is available
[   54.806806] remoteproc remoteproc2: 4a338000.pru is available
[  155.554940] remoteproc remoteproc1: powering up 4a334000.pru
[  155.564086] remoteproc remoteproc1: Booting fw image am335x-pru0-fw, size 34012
[  155.564143] remoteproc remoteproc1: remote processor 4a334000.pru is now up
[  419.051594] remoteproc remoteproc1: stopped remote processor 4a334000.pru
[  421.111787] remoteproc remoteproc1: powering up 4a334000.pru
[  421.112092] remoteproc remoteproc1: Booting fw image am335x-pru0-fw, size 34012
[  421.112137] remoteproc remoteproc1: remote processor 4a334000.pru is now up
[  570.085594] remoteproc remoteproc2: powering up 4a338000.pru
[  570.086141] remoteproc remoteproc2: Booting fw image am335x-pru1-fw, size 94200
[  570.101391] remoteproc remoteproc2: registered virtio0 (type 7)
[  570.101412] remoteproc remoteproc2: remote processor 4a338000.pru is now up
dmesg | grep pru
[   54.804469] remoteproc remoteproc1: 4a334000.pru is available
[   54.804638] pru-rproc 4a334000.pru: PRU rproc node pru@4a334000 probed successfully
[   54.806806] remoteproc remoteproc2: 4a338000.pru is available
[   54.806983] pru-rproc 4a338000.pru: PRU rproc node pru@4a338000 probed successfully
[  155.554940] remoteproc remoteproc1: powering up 4a334000.pru
[  155.564086] remoteproc remoteproc1: Booting fw image am335x-pru0-fw, size 34012
[  155.564143] remoteproc remoteproc1: remote processor 4a334000.pru is now up
[  419.051594] remoteproc remoteproc1: stopped remote processor 4a334000.pru
[  421.111787] remoteproc remoteproc1: powering up 4a334000.pru
[  421.112092] remoteproc remoteproc1: Booting fw image am335x-pru0-fw, size 34012
[  421.112137] remoteproc remoteproc1: remote processor 4a334000.pru is now up
[  570.085594] remoteproc remoteproc2: powering up 4a338000.pru
[  570.086141] remoteproc remoteproc2: Booting fw image am335x-pru1-fw, size 94200
[  570.086925] pruss 4a300000.pruss: configured system_events[63-0] = 00000000,000c0000
[  570.086940] pruss 4a300000.pruss: configured intr_channels = 0x0000000a host_intr = 0x0000000a
[  570.101412] remoteproc remoteproc2: remote processor 4a338000.pru is now up
[  570.219574] virtio_rpmsg_bus virtio0: creating channel rpmsg-pru addr 0x1e
[  570.310022] rpmsg_pru virtio0.rpmsg-pru.-1.30: new rpmsg_pru device: /dev/rpmsg_pru30
dmesg | grep pinctrl-single
[    0.894370] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
[    0.903370] gpio-of-helper ocp:cape-universal: ready
lsusb
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END
```