beagleboard/Latest-Images #58: BeagleBone Blue Pin 9_30 conflict

epena29, 2020-07-18:
pin conflict failing in starting pru1, result of this fix. below:> https://github.com/beagleboard/Latest-Images/issues/7

image:>>> bone-debian-10.3-iot-armhf-2020-04-06-4gb.img.xz
uname -a :>>>Linux beaglebone 4.19.94-ti-r45 #1buster SMP PREEMPT Thu Jun 18 19:18:41 UTC 2020 armv7l GNU/Linux

Hardware: >>>>BeagleBone Blue

```sudo /opt/scripts/tools/version.sh```:>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>

git:/opt/scripts/:[f495239e456d8ec99554a35ae04b33ddce5c26cc]
eeprom:[A335BNLTBLA21712EL004921]
model:[TI_AM335x_BeagleBone_Blue]
dogtag:[BeagleBoard.org Debian Buster IoT Image 2020-04-06]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2019.04-00002-g07d5700e21]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2019.04-00002-g07d5700e21]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-boneblue.dts]
UBOOT: Loaded Overlay:[AM335X-PRU-UIO-00A0]
UBOOT: Loaded Overlay:[BB-ADC-00A0]
kernel:[4.19.94-ti-r45]
nodejs:[v10.21.0]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-UIO-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20200709.0-0~buster+20200709]
pkg:[bb-wl18xx-firmware]:[1.20200702.0-0~buster+20200702]
pkg:[kmod]:[26-1]
pkg:[roboticscape]:[0.4.4-git20180608.0-0rcnee0~buster+20180609]:[GOT_REPLACED_BY_NEXT]
pkg:[librobotcontrol]:[1.0.5-git20200715.0-0~buster+20200716]
pkg:[firmware-ti-connectivity]:[20190717-2rcnee1~buster+20200305]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal bluetooth netdev i2c gpio pwm eqep remoteproc admin spi iio docker tisdk weston-launch xenomai cloud9ide]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet]
dmesg | grep remote
[   43.362822] remoteproc remoteproc0: wkup_m3 is available
[   43.960187] remoteproc remoteproc0: powering up wkup_m3
[   43.960222] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[   43.960516] remoteproc remoteproc0: remote processor wkup_m3 is now up
dmesg | grep pru
dmesg | grep pinctrl-single
[    0.912507] pinctrl-single 44e10800.pinmux: 142 pins, size 568
[    1.114317] pinctrl-single 44e10800.pinmux: pin PIN102 already requested by 48300180.eqep; cannot claim for ocp:P9_30_pinmux
[    1.125716] pinctrl-single 44e10800.pinmux: pin-102 (ocp:P9_30_pinmux) status -22
[    1.133303] pinctrl-single 44e10800.pinmux: could not request pin 102 (PIN102) from group pinmux_P9_30_spi_pin  on device pinctrl-single
dmesg | grep gpio-of-helper
lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END

**To Reproduce**
Steps to reproduce the behavior:
run rc_test_escs -c 1 -s .5
observe the output :>
ERROR in rc_pru_stop opening remoteproc driver: No such file or directory
PRU probably not enabled in device tree
ERROR in rc_servo_init, failed to start PRU1

**Expected behavior**
To be able to run the basic rc tests

jadonk, 2020-07-28:
Can you try to update librobotcontrol to 1.0.5?

bjnilsen, 2020-11-12:
#67