beagleboard/Latest-Images #60: update_kernel.sh broken on BeagleBone Green Gateway

mvduin, 2020-07-30:
On IRC it has been reported that a BBGG reported that the update_kernel.sh script fails to work:
```
debian@beaglebone:/opt/scripts/tools$ sudo ./update_kernel.sh
Machine: [SeeedStudio_BeagleBone_Green_Gateway]
/usr/sbin/update-initramfs: option requires an argument -- 'k'
 
Usage: update-initramfs {-c|-d|-u} [-k version] [-v] [-b directory]
 
Options:
 -k version     Specify kernel version or 'all'
 -c             Create a new initramfs
 -u             Update an existing initramfs
 -d             Remove an existing initramfs
 -b directory   Set alternate boot directory
 -v             Be verbose
 
See update-initramfs(8) for further details.
```

Version info:
```
git:/opt/scripts/:[b39ec679648a6be8f25f48bd1c9784c1fc5a0c46]
eeprom:[A335BNLTGG1A202005000165]
model:[SeeedStudio_BeagleBone_Green_Gateway]
dogtag:[BeagleBoard.org Debian Buster IoT Image 2020-04-06]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2019.04-00002-g23f263cc3f]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-boneblack-uboot-univ.dts]
UBOOT: Loaded Overlay:[AM335X-PRU-RPROC-4-19-TI-00A0]
UBOOT: Loaded Overlay:[BB-ADC-00A0]
UBOOT: Loaded Overlay:[BB-BBGG-WL1835-00A0]
UBOOT: Loaded Overlay:[BB-BONE-eMMC1-01-00A0]
UBOOT: Loaded Overlay:[M-BB-BBGG-00A0]
kernel:[4.19.94-ti-r42]
```

git-pulling in /opt/scripts did not help.

RobertCNelson, 2020-07-30:
Thanks @mvduin just pushed a fix for this board, i'm glad these are finally shipping!!

closed 2020-07-30 by RobertCNelson

silver2row, 2020-07-30:
Hello,

The boards are taking a month to arrive to the US (southern states) from China but they are neat boards. Thank you for your pushed fix.

Seth