beagleboard/Latest-Images #62: pwm's are group 'root' instead of group 'pwm'

MarkAYoder, 2020-08-04:
**Describe the bug**
When a pwm is exported it isn't put into group 'pwm' with group write permissions

**Describe which image you are running (example: bone-debian-10.2-iot-armhf-2020-02-03-4gb.img.xz) **
bone-debian-10.5-iot-armhf-2020-08-03-4gb.img.xz
uname -a
Linux black **4.19.94-ti-r45** #1buster SMP PREEMPT Thu Jun 18 19:18:41 UTC 2020 armv7l GNU/Linux

**Describe your hardware**
BB Black

**Describe any Capes Installed**
None

**Please run this built-in script**
```sudo /opt/scripts/tools/version.sh```
```
git:/opt/scripts/:[7cdc270818b76d829d247cf05fe309c320a3929a]
eeprom:[A335BNLT0A5A1513BBBK3586]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Buster IoT Image 2020-08-03]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2019.04-00002-g23f263cc3f]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-boneblack-uboot-univ.dts]
UBOOT: Loaded Overlay:[AM335X-PRU-RPROC-4-19-TI-00A0]
UBOOT: Loaded Overlay:[BB-ADC-00A0]
UBOOT: Loaded Overlay:[BB-BONE-eMMC1-01-00A0]
kernel:[4.19.94-ti-r45]
nodejs:[v10.21.0]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-19-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20200722.0-0~buster+20200722]
pkg:[bb-wl18xx-firmware]:[1.20200702.0-0~buster+20200702]
pkg:[kmod]:[26-1]
pkg:[librobotcontrol]:[1.0.5-git20200715.0-0~buster+20200716]
pkg:[firmware-ti-connectivity]:[20190717-2rcnee1~buster+20200305]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal input bluetooth netdev i2c gpio admin spi iio docker tisdk weston-launch xenomai cloud9ide pwm eqep remoteproc]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet]
dmesg | grep remote
[   61.001430] remoteproc remoteproc0: wkup_m3 is available
[   61.083777] remoteproc remoteproc0: powering up wkup_m3
[   61.083812] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[   61.084101] remoteproc remoteproc0: remote processor wkup_m3 is now up
[   64.747756] remoteproc remoteproc1: 4a334000.pru is available
[   64.764244] remoteproc remoteproc2: 4a338000.pru is available
[  155.538925] remoteproc remoteproc2: powering up 4a338000.pru
[  155.550378] remoteproc remoteproc2: Booting fw image am335x-pru1-fw, size 33528
[  155.550435] remoteproc remoteproc2: remote processor 4a338000.pru is now up
[  180.732465] remoteproc remoteproc1: powering up 4a334000.pru
[  180.732801] remoteproc remoteproc1: Booting fw image am335x-pru0-fw, size 34052
[  180.732849] remoteproc remoteproc1: remote processor 4a334000.pru is now up
[  190.798933] remoteproc remoteproc1: stopped remote processor 4a334000.pru
[  193.397179] remoteproc remoteproc1: powering up 4a334000.pru
[  193.397730] remoteproc remoteproc1: Booting fw image am335x-pru0-fw, size 94268
[  193.408493] remoteproc remoteproc1: registered virtio0 (type 7)
[  193.408574] remoteproc remoteproc1: remote processor 4a334000.pru is now up
[  270.408168] remoteproc remoteproc1: stopped remote processor 4a334000.pru
[  271.224073] remoteproc remoteproc1: powering up 4a334000.pru
[  271.224436] remoteproc remoteproc1: Booting fw image am335x-pru0-fw, size 32112
[  271.224469] remoteproc remoteproc1: remote processor 4a334000.pru is now up
[ 2333.662759] remoteproc remoteproc1: stopped remote processor 4a334000.pru
[ 2360.078044] remoteproc remoteproc1: powering up 4a334000.pru
[ 2360.078236] remoteproc remoteproc1: Booting fw image am335x-pru0-fw, size 3756
[ 2360.078289] remoteproc remoteproc1: remote processor 4a334000.pru is now up
[ 2413.759070] remoteproc remoteproc1: stopped remote processor 4a334000.pru
[ 2419.359334] remoteproc remoteproc1: powering up 4a334000.pru
[ 2419.359532] remoteproc remoteproc1: Booting fw image am335x-pru0-fw, size 3756
[... 60 lines cut ...]
dmesg | grep gpio-of-helper
[    0.969964] gpio-of-helper ocp:cape-universal: ready
lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END
```
**To Reproduce**
Steps to reproduce the behavior:
bone$ **cd /sys/class/pwm/pwmchip0**
bone$ **echo 0 > export** 
bone$ **cd pwm-0\:0/**
bone$ **ls -ls**
total 0
0 -r--r--r-- 1 root root 4096 Aug  4 14:37 capture
0 lrwxrwxrwx 1 root gpio    0 Aug  4 14:37 device -> ../../pwmchip0
0 -rw-r--r-- 1 root root 4096 Aug  4 14:37 duty_cycle
0 -rw-r--r-- 1 root root 4096 Aug  4 14:37 enable
0 -rw-r--r-- 1 root root 4096 Aug  4 14:37 period
0 -rw-r--r-- 1 root root 4096 Aug  4 14:37 polarity
0 drwxr-xr-x 2 root root    0 Aug  4 14:37 power
0 lrwxrwxrwx 1 root pwm     0 Aug  4 14:37 subsystem -> ../../../../../../../../class/pwm
0 -rw-r--r-- 1 root root 4096 Aug  4 14:37 uevent

**Expected behavior**
Should be in group 'pwm' with group  write permissions.

MarkAYoder, 2020-08-05:
This is true for the ```bone-debian-10.5-iot-armhf-2020-08-04-4gb.img.xz``` image too.

MarkAYoder, 2020-08-05:
Seems to be a udev problem.  The current file (`/etc/udev/rules.d/81-pwm-noroot.rules`) uses ```ACTION=="change"``` and it isn't working.  Using ```ACTION=="add"``` works.

But there is a note in **88-leds-noroot.rules** saying to use **change** rather than **add**.   So far ```add``` is working fine for me.

RobertCNelson, 2020-08-05:
Hi @MarkAYoder can a udev do both add and change??

Using change fixed another bug..

https://github.com/beagleboard/customizations/commit/ca1c2485303c3fd5e506ef2abb41a2bc63957f02#diff-f8d5a59f55b59983fe86d2c52c61d633

Regards,

RobertCNelson, 2020-08-05:
ah there is..

```
ACTION=="add|change"
```

closed 2020-08-05 by RobertCNelson

MarkAYoder, 2020-08-05:
@RobertCNelson  The add|change is working for me.  Thanks.

RobertCNelson, 2020-08-05:
awesome, i've pushed the package..  now to finally build the firmware. ;)