beagleboard/Latest-Images #64: BBBW eMMC flasher debian 10.4 console image

sreipro, 2020-08-18:
Hello everybody,

Impossible for me to flash my BBBW with "bone-eMMC-flasher-debian-10.4-console-armhf-2020-05-18-1gb.img.xz"
I tried "bone-debian-10.4-console-armhf-2020-05-18-1gb.img.xz", and exec manualy the script to get a log:

> ... Checking for Valid bbb-eeprom header
Traceback (last called is first):
     do_we_have_eeprom() in /opt/scripts/tools/eMMC/functions.sh:483
     do_we_have_am335x_eeprom() in /opt/scripts/tools/eMMC/functions.sh:491
     check_am335x_eeprom() in /opt/scripts/tools/eMMC/functions.sh:499
     check_eeprom() in /opt/scripts/tools/eMMC/functions.sh:535
     main() in /opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh:46
The command ls /sys/devices/ocp*/44e0b000.i2c/i2c-0/0-0050/eeprom 2> /dev/null exited with exit code 2.
100000+0 records in
100000+0 records out
51200000 bytes (51 MB, 49 MiB) copied, 1.44574 s, 35.4 MB/s
Traceback (last called is first):
     do_we_have_eeprom() in /opt/scripts/tools/eMMC/functions.sh:483
     do_we_have_am335x_eeprom() in /opt/scripts/tools/eMMC/functions.sh:491
     check_am335x_eeprom() in /opt/scripts/tools/eMMC/functions.sh:499
     check_eeprom() in /opt/scripts/tools/eMMC/functions.sh:535
     main() in /opt/scripts/tools/eMMC/init-eMMC-flasher-v3.sh:46
The command eeprom_location=$(ls /sys/devices/ocp*/44e0b000.i2c/i2c-0/0-0050/eeprom 2> /dev/null) exited with exit code 2.
================================================================================
Tearing Down script environment
==> Unmounting /tmp
umount: /tmp: not mounted.
==> Stopping Cylon LEDs ...
==> Stopping Cylon LEDs ...
==> Setting LEDs to

When I use "bone-eMMC-flasher-debian-9.12-console-armhf-2020-05-02-1gb.img.xz", I have no problem.

Thanks

RobertCNelson, 2020-08-18:
This was fixed on:

https://github.com/RobertCNelson/boot-scripts/commit/9b2c9d6a090fe468944b97408b3dcda2d2ec9fce

v5.4.x kernel's had the eeprom moved location..

Regards,

sreipro, 2020-08-18:
Fixed.

Thanks a lot.

closed 2020-08-18 by sreipro