beagleboard/Latest-Images #65: BBBW bluetooth issue

sreipro, 2020-08-27:
Hello,

On **bone-eMMC-flasher-debian-10.4-console-armhf-2020-05-18-1gb.img.xz**
Bluetooth not recognized by connman, Blue led is on, but something wrong on startup.

> $ journalctl | grep "blu"
Aug 27 16:19:47 beaglebone bb-wl18xx-bluetooth[2170]: Found a Texas Instruments' chip!
Aug 27 16:19:47 beaglebone bb-wl18xx-bluetooth[2170]: Firmware file : /lib/firmware/ti-connectivity/TIInit_11.8.32.bts
Aug 27 16:19:47 beaglebone bb-wl18xx-bluetooth[2170]: Loaded BTS script version 1
Aug 27 16:19:47 beaglebone bb-wl18xx-bluetooth[2170]: texas: changing baud rate to 3000000, flow control to 1
Aug 27 16:19:48 beaglebone bb-wl18xx-bluetooth[2170]: Can't set line discipline: Invalid argument
Aug 27 16:19:48 beaglebone bb-wl18xx-bluetooth[2170]: Can't initialize device: Invalid argument

With the stretch **bone-eMMC-flasher-debian-9.12-console-armhf-2020-05-02-1gb.img.xz**, all is ok.

> journalctl | grep "blu"
Aug 27 17:22:25 beaglebone bb-wl18xx-bluetooth[3048]: Found a Texas Instruments' chip!
Aug 27 17:22:25 beaglebone bb-wl18xx-bluetooth[3048]: Firmware file : /lib/firmware/ti-connectivity/TIInit_11.8.32.bts
Aug 27 17:22:25 beaglebone bb-wl18xx-bluetooth[3048]: Loaded BTS script version 1
Aug 27 17:22:25 beaglebone bb-wl18xx-bluetooth[3048]: texas: changing baud rate to 3000000, flow control to 1
Aug 27 17:22:27 beaglebone bb-wl18xx-bluetooth[3048]: Device setup complete
Aug 27 17:22:43 beaglebone bluetoothd[4148]: Bluetooth daemon 5.43
Aug 27 17:22:44 beaglebone bluetoothd[4148]: Starting SDP server
Aug 27 17:22:44 beaglebone bluetoothd[4148]: Bluetooth management interface 1.14 initialized
Aug 27 17:22:44 beaglebone bluetoothd[4148]: Failed to obtain handles for "Service Changed" characteristic
Aug 27 17:22:44 beaglebone bluetoothd[4148]: Sap driver initialization failed.
Aug 27 17:22:44 beaglebone bluetoothd[4148]: sap-server: Operation not permitted (1)
Aug 27 17:22:44 beaglebone bluetoothd[4148]: Failed to set privacy: Rejected (0x0b)

Thanks,

RobertCNelson, 2020-08-27:
HI @sreipro this should now be fully fixed ;)

https://elinux.org/Beagleboard:Latest-images-testing

Regards,

sreipro, 2020-08-27:
Excellent :)

All is working indeed with the 10.5:

> $ journalctl | grep "blu"
Feb 14 10:12:54 beaglebone bb-wl18xx-bluetooth[2043]: Found a Texas Instruments' chip!
Feb 14 10:12:54 beaglebone bb-wl18xx-bluetooth[2043]: Firmware file : /lib/firmware/ti-connectivity/TIInit_11.8.32.bts
Feb 14 10:12:54 beaglebone bb-wl18xx-bluetooth[2043]: Loaded BTS script version 1
Feb 14 10:12:54 beaglebone bb-wl18xx-bluetooth[2043]: texas: changing baud rate to 3000000, flow control to 1
Feb 14 10:12:56 beaglebone bluetoothd[2092]: Bluetooth daemon 5.50
Feb 14 10:12:56 beaglebone bluetoothd[2092]: Starting SDP server
Feb 14 10:12:56 beaglebone bluetoothd[2092]: Bluetooth management interface 1.14 initialized
Feb 14 10:12:56 beaglebone dbus-daemon[675]: [system] Activating via systemd: service name='org.freedesktop.hostname1' unit='dbus-org.freedesktop.hostname1.service' requested by ':1.6' (uid=0 pid=2092 comm="/usr/lib/bluetooth/bluetoothd ")
Feb 14 10:12:56 beaglebone bluetoothd[2092]: Sap driver initialization failed.
Feb 14 10:12:56 beaglebone bluetoothd[2092]: sap-server: Operation not permitted (1)
Feb 14 10:12:56 beaglebone bluetoothd[2092]: Failed to set privacy: Rejected (0x0b)
Feb 14 10:12:56 beaglebone bb-wl18xx-bluetooth[2043]: Device setup complete

Thanks a lot.

closed 2020-08-27 by sreipro