beagleboard/Latest-Images #67: Encoder 48300180.eqep return onlu -1 or 1 or 0

mdlugosz, 2020-09-19:
**Describe the bug**
The encoder EQEP0 return only three values -1, 0, 1. 

**Describe which image you are running (example: bone-debian-10.2-iot-armhf-2020-02-03-4gb.img.xz) **
bone-ubuntu-18.04.4-ros-iot-armhf-2020-05-18-6gb.img

**Describe your hardware**
BeagleBone Blue

**Describe any Capes Installed**

**Please run this built-in script**
```sudo /opt/scripts/tools/version.sh```
git:/opt/scripts/:[029041f6866049997bbfd2c7667b3c6e8c95201c]
eeprom:[A335BNLTBLA21712EL002910]
model:[TI_AM335x_BeagleBone_Blue]
dogtag:[BeagleBoard.org ROS Image 2020-05-18]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2019.04-00002-g31a8ae0206]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.09-00002-g0b54a51eee]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-boneblue.dts]
UBOOT: Loaded Overlay:[AM335X-PRU-RPROC-4-19-TI-00A0]
UBOOT: Loaded Overlay:[BB-ADC-00A0]
kernel:[4.19.94-ti-rt-r43]
nodejs:[v8.10.0]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-19-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20200814.0-0~bionic+20200814]
pkg:[bb-wl18xx-firmware]:[1.20200813.1-0~bionic+20200813]
pkg:[kmod]:[24-1ubuntu3.5rcnee0~bionic+20200810]
pkg:[librobotcontrol]:[1.0.5-git20200715.0-0~bionic+20200716]
groups:[beagle : beagle root adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal input bluetooth netdev i2c cloud9ide gpio pwm eqep remoteproc admin spi iio docker tisdk weston-launch xenomai]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet]
dmesg | grep remote
[   42.330772] remoteproc remoteproc0: 4a334000.pru is available
[   42.346759] remoteproc remoteproc1: 4a338000.pru is available
[  154.575230] remoteproc remoteproc2: wkup_m3 is available
[  154.592590] remoteproc remoteproc2: powering up wkup_m3
[  154.592613] remoteproc remoteproc2: Booting fw image am335x-pm-firmware.elf, size 217168
[  154.592894] remoteproc remoteproc2: remote processor wkup_m3 is now up
[  282.595682] remoteproc remoteproc0: powering up 4a334000.pru
[  282.604229] remoteproc remoteproc0: Booting fw image am335x-pru0-rc-encoder-fw, size 36040
[  282.604275] remoteproc remoteproc0: remote processor 4a334000.pru is now up
[  853.519686] remoteproc remoteproc0: stopped remote processor 4a334000.pru
dmesg | grep pru
[   42.330772] remoteproc remoteproc0: 4a334000.pru is available
[   42.330953] pru-rproc 4a334000.pru: PRU rproc node pru@4a334000 probed successfully
[   42.346759] remoteproc remoteproc1: 4a338000.pru is available
[   42.346958] pru-rproc 4a338000.pru: PRU rproc node pru@4a338000 probed successfully
[  282.595682] remoteproc remoteproc0: powering up 4a334000.pru
[  282.604229] remoteproc remoteproc0: Booting fw image am335x-pru0-rc-encoder-fw, size 36040
[  282.604275] remoteproc remoteproc0: remote processor 4a334000.pru is now up
[  853.519686] remoteproc remoteproc0: stopped remote processor 4a334000.pru
dmesg | grep pinctrl-single
[    1.065640] pinctrl-single 44e10800.pinmux: 142 pins, size 568
[    1.289110] pinctrl-single 44e10800.pinmux: pin PIN102 already requested by 48300180.eqep; cannot claim for ocp:P9_30_pinmux
[    1.305932] pinctrl-single 44e10800.pinmux: pin-102 (ocp:P9_30_pinmux) status -22
[    1.313470] pinctrl-single 44e10800.pinmux: could not request pin 102 (PIN102) from group pinmux_P9_30_spi_pin  on device pinctrl-single
dmesg | grep gpio-of-helper
lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END

**To Reproduce**
Steps to reproduce the behavior:
1. Connect quadrant encoder to eqep0
2. Run rc_test_encoders
or 
watch -n 0.1 cat /sys/devices/platform/ocp/48300000.epwmss/48300180.eqep/position
3. Rotate shaft of encoder whith is connected to EQEP0 and see error, returned value is -1 or 0 or 1
Other encoders are working properly.

**Expected behavior**
The encoder EQEP0 should work like other encoders and return valid values.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

bjnilsen, 2020-10-07:
Same issue with bone-debian-10.5-iot-armhf-2020-08-25-4gb.img.xz

Please let me know if there is anything I can provide to debug this issue further.

jadonk, 2020-10-15:
Really odd. Any chance you can put a scope on the 2 qep lines?

jadonk, 2020-10-15:
Oh, also, is there anything in the output of 'dmesg' about the qep?

mdlugosz, 2020-10-15:
### debian@beaglebone:~$ dmesg | grep qep
[    1.108918] eqep 48300180.eqep: ver. 1.0
[    1.109175] eqep 48300180.eqep: count_mode:0
[    1.109186] eqep 48300180.eqep: invert_qa:1
[    1.109195] eqep 48300180.eqep: invert_qb:1
[    1.109203] eqep 48300180.eqep: invert_qi:0
[    1.109211] eqep 48300180.eqep: invert_qs:0
[    1.109220] eqep 48300180.eqep: swap_inputs:0
[    1.109228] eqep 48300180.eqep: QDECCTL:0x0180
[    1.109237] eqep 48300180.eqep: QPOSINIT:0x00000000
[    1.109244] eqep 48300180.eqep: QPOSMAX:0xffffffff
[    1.109251] eqep 48300180.eqep: QPOSCNT:0x00000000
[    1.109259] eqep 48300180.eqep: omit_interrupt:0
[    1.109267] eqep 48300180.eqep: QEINT:0x0800
[    1.109274] eqep 48300180.eqep: QUPRD:0x05f5e100
[    1.109281] eqep 48300180.eqep: QEPCTL:0x009e write
[    1.109289] eqep 48300180.eqep: QEPCTL:0x009e read
[    1.109316] eqep 48300180.eqep: irq:60, clk_rate:100000000
[    1.109864] eqep 48302180.eqep: ver. 1.0
[    1.110005] eqep 48302180.eqep: count_mode:0
[    1.110016] eqep 48302180.eqep: invert_qa:1
[    1.110024] eqep 48302180.eqep: invert_qb:1
[    1.110033] eqep 48302180.eqep: invert_qi:0
[    1.110041] eqep 48302180.eqep: invert_qs:0
[    1.110048] eqep 48302180.eqep: swap_inputs:0
[    1.110056] eqep 48302180.eqep: QDECCTL:0x0180
[    1.110064] eqep 48302180.eqep: QPOSINIT:0x00000000
[    1.110071] eqep 48302180.eqep: QPOSMAX:0xffffffff
[    1.110078] eqep 48302180.eqep: QPOSCNT:0x00000000
[    1.110087] eqep 48302180.eqep: omit_interrupt:0
[    1.110094] eqep 48302180.eqep: QEINT:0x0800
[    1.110102] eqep 48302180.eqep: QUPRD:0x05f5e100
[    1.110109] eqep 48302180.eqep: QEPCTL:0x009e write
[    1.110116] eqep 48302180.eqep: QEPCTL:0x009e read
[    1.110212] eqep 48302180.eqep: irq:62, clk_rate:100000000
[    1.110466] eqep 48304180.eqep: ver. 1.0
[    1.110662] eqep 48304180.eqep: count_mode:0
[    1.110671] eqep 48304180.eqep: invert_qa:1
[    1.110679] eqep 48304180.eqep: invert_qb:1
[    1.110687] eqep 48304180.eqep: invert_qi:0
[    1.110695] eqep 48304180.eqep: invert_qs:0
[    1.110703] eqep 48304180.eqep: swap_inputs:0
[    1.110710] eqep 48304180.eqep: QDECCTL:0x0180
[    1.110718] eqep 48304180.eqep: QPOSINIT:0x00000000
[    1.110726] eqep 48304180.eqep: QPOSMAX:0xffffffff
[    1.110733] eqep 48304180.eqep: QPOSCNT:0x00000000
[    1.110741] eqep 48304180.eqep: omit_interrupt:0
[    1.110748] eqep 48304180.eqep: QEINT:0x0800
[    1.110755] eqep 48304180.eqep: QUPRD:0x05f5e100
[    1.110762] eqep 48304180.eqep: QEPCTL:0x009e write
[    1.110769] eqep 48304180.eqep: QEPCTL:0x009e read
[    1.110787] eqep 48304180.eqep: irq:64, clk_rate:100000000
[    1.114175] pinctrl-single 44e10800.pinmux: pin PIN102 already requested by 48300180.eqep; cannot claim for ocp:P9_30_pinmux

### or whole dmesg:
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.19.94-ti-r42 (voodoo@x3-am57xx-beagle-x15-2gb) (gcc version 8.3.0 (Debian 8.3.0-6)) #1buster SMP PREEMPT Tue Mar 31 19:38:29 UTC 2020
[    0.000000] CPU: ARMv7 Processor [413fc082] revision 2 (ARMv7), cr=10c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: TI AM335x BeagleBone Blue
[    0.000000] Memory policy: Data cache writeback
[    0.000000] cma: Reserved 48 MiB at 0x9c800000
[    0.000000] On node 0 totalpages: 130560
[    0.000000]   Normal zone: 1148 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 130560 pages, LIFO batch:31
[    0.000000] CPU: All CPU(s) started in SVC mode.
[    0.000000] AM335X ES2.1 (sgx neon)
[    0.000000] random: get_random_bytes called from start_kernel+0xac/0x4b4 with crng_init=0
[    0.000000] percpu: Embedded 17 pages/cpu s38988 r8192 d22452 u69632
[... 325 lines cut ...]
[   54.077121] wlcore: loaded
[   54.516177] Bluetooth: RFCOMM TTY layer initialized
[   54.516221] Bluetooth: RFCOMM socket layer initialized
[   54.516285] Bluetooth: RFCOMM ver 1.11
[   54.688299] wlcore: PHY firmware version: Rev 8.2.0.0.242
[   54.729301] wlcore: firmware booted (Rev 8.9.0.0.79)
[   54.774380] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   56.123689] wlan0: authenticate with f0:99:bf:0c:c4:c0
[   56.136830] wlan0: send auth to f0:99:bf:0c:c4:c0 (try 1/3)
[   56.268098] wlan0: send auth to f0:99:bf:0c:c4:c0 (try 2/3)
[   56.287697] wlan0: authenticated
[   56.296123] wlan0: associate with f0:99:bf:0c:c4:c0 (try 1/3)
[   56.300877] wlan0: RX AssocResp from f0:99:bf:0c:c4:c0 (capab=0x1411 status=0 aid=2)
[   56.314607] wlan0: associated
[   56.569781] cryptd: max_cpu_qlen set to 1000
[   57.592775] wlan0: Limiting TX power to 20 (20 - 0) dBm as advertised by f0:99:bf:0c:c4:c0
[   57.619078] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   57.649765] wlcore: Association completed.
[   57.824232] IPv6: ADDRCONF(NETDEV_UP): SoftAp0: link is not ready
[   59.846359] wlcore: down
[   59.963169] IPv6: ADDRCONF(NETDEV_CHANGE): SoftAp0: link becomes ready
[  361.046294] remoteproc remoteproc0: powering up 4a334000.pru
[  361.063572] remoteproc remoteproc0: Booting fw image am335x-pru0-rc-encoder-fw, size 36040
[  361.063616] remoteproc remoteproc0: remote processor 4a334000.pru is now up
[  426.526981] remoteproc remoteproc0: stopped remote processor 4a334000.pru

bjnilsen, 2020-10-16:
Attached is qep signals. Looks good.

This issue is related to SW, it worked just fine with the pre-installed image from factory. I believe it was 4.4 kernel (I forgot to check this before upgrading to the latest image).
[image]

bjnilsen, 2020-10-16:
root@beaglebone:/home/debian# dmesg | grep qep
[    1.088854] eqep 48300180.eqep: ver. 1.0
[    1.089102] eqep 48300180.eqep: count_mode:0
[    1.089113] eqep 48300180.eqep: invert_qa:1
[    1.089122] eqep 48300180.eqep: invert_qb:1
[    1.089131] eqep 48300180.eqep: invert_qi:0
[    1.089139] eqep 48300180.eqep: invert_qs:0
[    1.089148] eqep 48300180.eqep: swap_inputs:0
[    1.089157] eqep 48300180.eqep: QDECCTL:0x0180
[    1.089165] eqep 48300180.eqep: QPOSINIT:0x00000000
[    1.089173] eqep 48300180.eqep: QPOSMAX:0xffffffff
[    1.089180] eqep 48300180.eqep: QPOSCNT:0x00000000
[    1.089189] eqep 48300180.eqep: omit_interrupt:0
[    1.089196] eqep 48300180.eqep: QEINT:0x0800
[    1.089204] eqep 48300180.eqep: QUPRD:0x05f5e100
[    1.089211] eqep 48300180.eqep: QEPCTL:0x009e write
[    1.089219] eqep 48300180.eqep: QEPCTL:0x009e read
[    1.089247] eqep 48300180.eqep: irq:60, clk_rate:100000000
[    1.089820] eqep 48302180.eqep: ver. 1.0
[    1.089972] eqep 48302180.eqep: count_mode:0
[    1.089982] eqep 48302180.eqep: invert_qa:1
[    1.089991] eqep 48302180.eqep: invert_qb:1
[    1.089999] eqep 48302180.eqep: invert_qi:0
[    1.090007] eqep 48302180.eqep: invert_qs:0
[    1.090015] eqep 48302180.eqep: swap_inputs:0
[    1.090023] eqep 48302180.eqep: QDECCTL:0x0180
[    1.090031] eqep 48302180.eqep: QPOSINIT:0x00000000
[    1.090038] eqep 48302180.eqep: QPOSMAX:0xffffffff
[    1.090045] eqep 48302180.eqep: QPOSCNT:0x00000000
[    1.090053] eqep 48302180.eqep: omit_interrupt:0
[    1.090060] eqep 48302180.eqep: QEINT:0x0800
[    1.090067] eqep 48302180.eqep: QUPRD:0x05f5e100
[    1.090149] eqep 48302180.eqep: QEPCTL:0x009e write
[    1.090157] eqep 48302180.eqep: QEPCTL:0x009e read
[    1.090179] eqep 48302180.eqep: irq:62, clk_rate:100000000
[    1.090525] eqep 48304180.eqep: ver. 1.0
[    1.090653] eqep 48304180.eqep: count_mode:0
[    1.090663] eqep 48304180.eqep: invert_qa:1
[    1.090671] eqep 48304180.eqep: invert_qb:1
[    1.090679] eqep 48304180.eqep: invert_qi:0
[    1.090687] eqep 48304180.eqep: invert_qs:0
[    1.090696] eqep 48304180.eqep: swap_inputs:0
[    1.090704] eqep 48304180.eqep: QDECCTL:0x0180
[    1.090712] eqep 48304180.eqep: QPOSINIT:0x00000000
[    1.090720] eqep 48304180.eqep: QPOSMAX:0xffffffff
[    1.090727] eqep 48304180.eqep: QPOSCNT:0x00000000
[    1.090735] eqep 48304180.eqep: omit_interrupt:0
[    1.090743] eqep 48304180.eqep: QEINT:0x0800
[    1.090750] eqep 48304180.eqep: QUPRD:0x05f5e100
[    1.090757] eqep 48304180.eqep: QEPCTL:0x009e write
[    1.090764] eqep 48304180.eqep: QEPCTL:0x009e read
[    1.090786] eqep 48304180.eqep: irq:64, clk_rate:100000000
[    1.094121] pinctrl-single 44e10800.pinmux: pin PIN102 already requested by 48300180.eqep; cannot claim for ocp:P9_30_pinmux

mdlugosz, 2020-10-16:
Exactly, I confirm. Everything work perfectly when I use pre-instaled
system, but when I want use another debian or ubuntu running from sd-card
this EP1 not work. So this is software (or configure) problem not hardware.

bjnilsen, 2020-10-16:
The pinmux error looks suspicious:

[ 1.114175] pinctrl-single 44e10800.pinmux: pin PIN102 already requested by 48300180.eqep; cannot claim for ocp:P9_30_pinmux

mdlugosz, 2020-10-17:
Ok, thank you!
What and where we should correct to make it work?
As I wrote above, if system is booted from eMMC all four encoders work propertly.

bjnilsen, 2020-11-03:
Any idea what's wrong or how to fix it?

MarkAYoder, 2020-11-03:
I got the -1, 0, 1 outputs when config-pin was used to mux multiple pins to the same eQEP.

bjnilsen, 2020-11-12:
The problem is that device tree am335x-boneblue.dts uses the wrong pin:

`/* E1 */
	eqep0_pins: pinmux_eqep0_pins {
		pinctrl-single,pins = <
			AM33XX_PADCONF(AM335X_PIN_MCASP0_AXR0, PIN_INPUT, MUX_MODE1)		/* (B12) mcasp0_aclkr.eQEP0A_in */
			AM33XX_PADCONF(AM335X_PIN_MCASP0_FSR, PIN_INPUT, MUX_MODE1)		/* (C13) mcasp0_fsr.eQEP0B_in */
		>;
	};`

Wrong:
AM335X_PIN_MCASP0_AXR0

Correct:
AM335X_PIN_MCASP0_ACLKR

RobertCNelson, 2020-11-12:
Nice find @bjnilsen , can you issue a pull request to: https://github.com/beagleboard/BeagleBoard-DeviceTrees to fix it ;)

bjnilsen, 2020-11-12:
Pull request added: [https://github.com/beagleboard/BeagleBoard-DeviceTrees/pull/28](https://github.com/beagleboard/BeagleBoard-DeviceTrees/pull/28)

bjnilsen, 2020-11-12:
Pull request merged. Until the fix is released one can use the following steps to solve the problem:
(assuming 4.19.94-ti-r55 kernel on BeagleBone Blue)

1. git clone https://github.com/beagleboard/BeagleBoard-DeviceTrees.git
2. cd BeagleBoard-DeviceTrees
3. make
4. sudo cp src/arm/am335x-boneblue.dtb /boot/dtbs/4.19.94-ti-r55/am335x-boneblue.dtb
5. reboot

ldbrouwer, 2023-08-05:
All: I just had the same issue with reading eqep encoder 1. It reads either 1 and 0; or -1 and 0. I just purchased my Beaglebone blue recently and upgraded the software. Has the kernel been upgraded? Can I check to see if I have a  working kernel or if I need to implement the fix referenced above?