beagleboard/Latest-Images #71: Improve apt update speed

JoseGoncalves, 2020-10-30:
**Is your feature request related to a problem? Please describe.**
Updating Debian packages with `apt update` command is slow.

**Describe the solution you'd like**
Configure `apt` default behaviour to disable the usage of "diff" files, adding a file in `/etc/apt/apt.conf.d/` with the following content:
```
Acquire::PDiffs "false";
```

**Additional context**
* Information on `apt update` methodologies with relevant info for this issue on "6.2.1 - TIP Incremental updates":
  https://debian-handbook.info/browse/stable/sect.apt-get.html

* This apt configuration is also used in Raspberry Pi OS.

RobertCNelson, 2020-11-02:
Thanks @JoseGoncalves so we left the PDiff's enabled years ago to save download data... I think at this point, i agree let's disable it..

RobertCNelson, 2020-11-02:
We have no-langues also enabled..

When did CompressionTypes::Order come out?/

We also enable...
```
Acquire::GzipIndexes "true"; APT::Compressor::xz::Cost "40";
```

closed 2020-11-02 by RobertCNelson

JoseGoncalves, 2020-11-02:
Hi @RobertCNelson. `0` and `false` both disable an `apt` feature, but I think it's more explicit to use the later (as indicated in the [Debian Handbook](https://debian-handbook.info/browse/stable/sect.apt-get.html)), because it's more explicit on the nature of that config variable (a boolean).