beagleboard/Latest-Images #72: Strange uid and gid in files at /etc/apt/apt.conf.d/

JoseGoncalves, 2020-10-30:
**Describe the bug**
A minor issue. Some files in `/etc/apt/apt.conf.d` have `uid` 110 and `gid` 113.

**Describe which image you are running**
bone-debian-10.6-console-armhf-2020-10-28-1gb.img.xz

**Expected behavior**
These files should be owned by root.

**Additional context**
```
debian@beaglebone:~$ ls -ln /etc/apt/apt.conf.d/
total 24
-rw-r--r-- 1   0   0 1018 May 12 10:57 01autoremove
-r--r--r-- 1   0   0 2074 Oct 28 13:03 01autoremove-kernels
-rw-r--r-- 1 110 113   27 Oct 28 13:02 02-no-languages
-rw-r--r-- 1 110 113  245 Oct 28 13:02 02apt-get-clean
-rw-r--r-- 1 110 113   61 Oct 28 13:02 02compress-indexes
-rw-r--r-- 1   0   0  182 Feb 26  2019 70debconf
```

RobertCNelson, 2020-11-02:
Thanks ! @JoseGoncalves  anytime you see 110:113 that's a uid leak from Jenkins..

closed 2020-11-02 by RobertCNelson

JoseGoncalves, 2020-11-02:
Just noticed that `/etc/apt/sources.list` also has the same wrong uid:gid...

RobertCNelson, 2020-11-02:
Grabbed these too:

https://github.com/RobertCNelson/omap-image-builder/commit/f3e795fe2e3ac22c55ad6dd3438b36c9e780221a