beagleboard/Latest-Images #74: boot failure after updating to 2020-04-06 image on Beaglebone Green Wireless

dwliu, 2020-11-18:
Hi there,

I was unable to update my BBGW board to the latest image. The existing image onboard is 2016-05-27. I tried the 2020-04-06 images with emmc flasher. Burning looked working, the LEDs kept cycling as expected and after ~5 mins all LEDs were off. 

However, after removing SD card and powering the board, the booting was not successful. At the beginning those LED blinked a few times, then settled with fewer on. I couldn't ssh to it via either USB or WiFi. 

What I don't understand is even when booting from SD card, the image version displayed at the terminal still shows the old one. I don't have a USB2TTL cable to see what's going on with booting. Can anyone help?

Thanks,
Dawei

RobertCNelson, 2020-11-18:
@dwliu other then no usb interface, everything sounds correct..  Sounds like it booted but just failed.

What OS are you running on your Development PC, Windows (7,8,10,etc), Linux or Mac?

Do you use see a new "BeagleBone-WXYZ" ssid when you scan for wireless devices on any phone/computer?

WXYZ = random mac, so it won't match..

What was the file name of the 2020-04-06 release that you tried to use? Did use use https://www.balena.io/etcher/ to write the microSD?

Regards,

dwliu, 2020-11-20:
@RobertCNelson Sorry for the late response. Thank you so much for replying! I am on MacOS 10.11. However, I am not sure if it is related to my computer. I tried burning the BBGW with power supply only, and it booted failed the same way.

I had wireless working before I burned the new image. After burning, the green LED near WiFi module was not on, and the SSID was not detected either.

For the images, I tried both AM3358 Debian 10.3 2020-04-06 4GB SD IoT, and AM3358 Debian 10.3 2020-04-06 4GB eMMC IoT Flasher. They ended up the same. I didn't use BalenaEtcher to write the image though. I used Raspberry Pi Imager and manually assigned the image (thought it should work because verification succeeded).

Given what you said, I am leaning toward that either microSD or the burning has issue. I am going to try the burning tool you recommended.

Thanks again.

Dawei

dwliu, 2020-11-20:
BalenaEther imager didn't help either. Got the same issue. I have only one 4GB microSD card. Tried with 16GB card but the imager reported error. Unless you have any other suggestion, I guess I'd wait until I have the USB2TTL cable to check the boot process.