beagleboard/Latest-Images #75: No wifi on am57xx-debian-10.5-iot-armhf-2020-08-25-4gb.img.xz

dmconover, 2020-11-20:
**Problem:**

The am57xx-debian-10.5-iot-armhf-2020-08-25-4gb.img.xz image hosted on https://elinux.org/Beagleboard:Latest-images-testing#Debian_10_.28Buster.29_IoT no longer provides a wifi connection or network 

**Images:**
I have observed this on:
  - am57xx-debian-10.5-iot-armhf-2020-08-25-4gb.img.xz
  - https://rcn-ee.com/rootfs/bb.org/testing/2020-11-16/buster-iot/am57xx-debian-10.6-iot-armhf-2020-11-16-4gb.bmap
  - https://rcn-ee.com/rootfs/bb.org/testing/2020-09-07/buster-iot/am57xx-debian-10.5-iot-armhf-2020-09-07-4gb.img.xz

**Hardware**
  - BBAI, no custom capes installed

**Result of Script:**
git:/opt/scripts/:[7cdc270818b76d829d247cf05fe309c320a3929a]
model:[BeagleBoard.org_BeagleBone_AI]
dogtag:[BeagleBoard.org Debian Buster IoT Image 2020-08-25]
UBOOT: Booted Device-Tree:[am5729-beagleboneai.dts]
kernel:[4.19.94-ti-r50]
nodejs:[v10.21.0]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20200814.0-0~buster+20200814]
pkg:[bb-wl18xx-firmware]:[1.20200813.1-0~buster+20200813]
pkg:[kmod]:[26-1]
pkg:[librobotcontrol]:[1.0.5-git20200715.0-0~buster+20200716]
pkg:[firmware-ti-connectivity]:[20190717-2rcnee1~buster+20200305]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal input bluetooth netdev i2c gpio admin spi iio docker tisdk weston-launch xenomai cloud9ide pwm eqep remoteproc]
cmdline:[console=ttyS0,115200n8 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 rng_core.default_quality=100 quiet]
dmesg | grep remote
[    4.633835] remoteproc remoteproc0: 4b234000.pru is available
[    4.634630] remoteproc remoteproc1: 4b238000.pru is available
[    4.644737] remoteproc remoteproc2: 4b2b4000.pru is available
[    4.647344] remoteproc remoteproc3: 4b2b8000.pru is available
[   17.529317] remoteproc remoteproc4: 58820000.ipu is available
[   17.532595] remoteproc remoteproc4: Direct firmware load for dra7-ipu1-fw.xem4 failed with error -2
[   17.532633] remoteproc remoteproc4: powering up 58820000.ipu
[   17.532728] remoteproc remoteproc4: Direct firmware load for dra7-ipu1-fw.xem4 failed with error -2
[   17.532758] remoteproc remoteproc4: request_firmware failed: -2
[   17.540263] remoteproc remoteproc5: 55020000.ipu is available
[   17.563272] remoteproc remoteproc6: 40800000.dsp is available
[   17.571449] remoteproc remoteproc7: 41000000.dsp is available
[   17.580463] remoteproc remoteproc6: Direct firmware load for dra7-dsp1-fw.xe66 failed with error -2
[   17.580505] remoteproc remoteproc6: powering up 40800000.dsp
[   17.580607] remoteproc remoteproc6: Direct firmware load for dra7-dsp1-fw.xe66 failed with error -2
[   17.580634] remoteproc remoteproc6: request_firmware failed: -2
[   17.587809] remoteproc remoteproc7: Direct firmware load for dra7-dsp2-fw.xe66 failed with error -2
[   17.587850] remoteproc remoteproc7: powering up 41000000.dsp
[   17.587951] remoteproc remoteproc7: Direct firmware load for dra7-dsp2-fw.xe66 failed with error -2
[   17.587979] remoteproc remoteproc7: request_firmware failed: -2
[   17.618471] remoteproc remoteproc5: powering up 55020000.ipu
[   17.618517] remoteproc remoteproc5: Booting fw image dra7-ipu2-fw.xem4, size 3751356
[   17.983344] remoteproc remoteproc5: registered virtio0 (type 7)
[   17.983374] remoteproc remoteproc5: remote processor 55020000.ipu is now up
[   27.988105] remoteproc remoteproc5: crash detected in 55020000.ipu: type watchdog
[   27.995849] remoteproc remoteproc5: crash detected in 55020000.ipu: type watchdog
[   28.003589] remoteproc remoteproc5: handling crash #1 in 55020000.ipu
[   28.010327] remoteproc remoteproc5: recovering 55020000.ipu
[   28.086869] remoteproc remoteproc5: stopped remote processor 55020000.ipu
[   28.086901] remoteproc remoteproc5: powering up 55020000.ipu
[   28.115851] remoteproc remoteproc5: Booting fw image dra7-ipu2-fw.xem4, size 3751356
[   28.435933] remoteproc remoteproc5: registered virtio0 (type 7)
[   28.435962] remoteproc remoteproc5: remote processor 55020000.ipu is now up
[   38.440758] remoteproc remoteproc5: crash detected in 55020000.ipu: type watchdog
[   38.448461] remoteproc remoteproc5: crash detected in 55020000.ipu: type watchdog
[   38.456194] remoteproc remoteproc5: handling crash #2 in 55020000.ipu
[   38.469033] remoteproc remoteproc5: recovering 55020000.ipu
[   38.535631] remoteproc remoteproc5: stopped remote processor 55020000.ipu
[   38.535663] remoteproc remoteproc5: powering up 55020000.ipu
[   38.551879] remoteproc remoteproc5: Booting fw image dra7-ipu2-fw.xem4, size 3751356
[   39.087907] remoteproc remoteproc5: registered virtio0 (type 7)
[   39.087957] remoteproc remoteproc5: remote processor 55020000.ipu is now up
[   49.092609] remoteproc remoteproc5: crash detected in 55020000.ipu: type watchdog
[   49.100338] remoteproc remoteproc5: crash detected in 55020000.ipu: type watchdog
[   49.108136] remoteproc remoteproc5: handling crash #3 in 55020000.ipu
[   49.117068] remoteproc remoteproc5: recovering 55020000.ipu
[   49.277061] remoteproc remoteproc5: stopped remote processor 55020000.ipu
[   49.277092] remoteproc remoteproc5: powering up 55020000.ipu
[   49.329400] remoteproc remoteproc5: Booting fw image dra7-ipu2-fw.xem4, size 3751356
[   49.881000] remoteproc remoteproc5: registered virtio0 (type 7)
[   49.881031] remoteproc remoteproc5: remote processor 55020000.ipu is now up
dmesg | grep pru
[    4.633835] remoteproc remoteproc0: 4b234000.pru is available
[    4.633909] pru-rproc 4b234000.pru: PRU rproc node pru@4b234000 probed successfully
[    4.634630] remoteproc remoteproc1: 4b238000.pru is available
[    4.634688] pru-rproc 4b238000.pru: PRU rproc node pru@4b238000 probed successfully
[    4.644737] remoteproc remoteproc2: 4b2b4000.pru is available
[    4.644820] pru-rproc 4b2b4000.pru: PRU rproc node pru@4b2b4000 probed successfully
[    4.647344] remoteproc remoteproc3: 4b2b8000.pru is available
[    4.647407] pru-rproc 4b2b8000.pru: PRU rproc node pru@4b2b8000 probed successfully
[   51.452529] pruss_uio_shmem 4b200000.pruss_shmem: Allocating gdev
[   51.452558] pruss_uio_shmem 4b200000.pruss_shmem: Allocating info
[   51.452580] pruss_uio_shmem 4b200000.pruss_shmem: Requesting resource
[   51.452646] pruss_uio_shmem 4b200000.pruss_shmem: Mapping resource
[   51.452694] pruss_uio_shmem 4b200000.pruss_shmem: Registering with uio driver
[   51.500290] pruss_uio_shmem 4b200000.pruss_shmem: Saving platform data
[   51.500669] pruss_uio_shmem 4b280000.pruss_shmem: Allocating gdev
[   51.500693] pruss_uio_shmem 4b280000.pruss_shmem: Allocating info
[   51.500717] pruss_uio_shmem 4b280000.pruss_shmem: Requesting resource
[   51.500777] pruss_uio_shmem 4b280000.pruss_shmem: Mapping resource
[   51.500821] pruss_uio_shmem 4b280000.pruss_shmem: Registering with uio driver
[   51.513210] pruss_uio_shmem 4b280000.pruss_shmem: Saving platform data
dmesg | grep pinctrl-single
[    0.991029] pinctrl-single 4a003400.pinmux: 282 pins, size 1128
dmesg | grep gpio-of-helper
[    1.003372] gpio-of-helper 44000000.ocp:cape-universal: ready
lsusb
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END

**To Reproduce**
  - Boot it up

**Expected behavior**
  - a Wifi network appearing as BeagleBone-****** should become visible on my pc

**Additional context**
I have flashed the recomended image:
  -  AM5729 Debian 10.3 2020-04-06 8GB SD IoT TIDL
off of https://beagleboard.org/latest-images
And the network shows up.

RobertCNelson, 2020-11-20:
Humm...

```
debian@beaglebone:~$ journalctl | grep tether
Nov 16 11:26:55 beaglebone systemd[1]: Starting BBAI brcmfmac tether Service...
Nov 16 11:27:21 beaglebone bb-bbai-tether[1296]: bbai:tether waiting for /sys/class/net/wlan0
Nov 20 18:42:57 beaglebone bb-bbai-tether[1296]: bbai:wlan0:Power Management:off
Nov 20 18:42:57 beaglebone bb-bbai-tether[1296]: bbai:tether [iw phy phy0 interface add SoftAp0 type __ap]
Nov 20 18:42:57 beaglebone bb-bbai-tether[1296]: bbai:tether [phy0 created]
Nov 20 18:42:57 beaglebone bb-bbai-tether[1296]: bbai:tether [ip link set dev SoftAp0 down]
Nov 20 18:42:57 beaglebone bb-bbai-tether[1296]: bbai:tether [ip link set dev SoftAp0 address 00:00:00:00:00:00]
Nov 20 18:42:57 beaglebone bb-bbai-tether[1296]: bbai:tether [ip link set dev SoftAp0 up]
Nov 20 18:42:57 beaglebone bb-bbai-tether[1296]: bbai:tether [ip addr add 192.168.8.1/24 broadcast 192.168.8.255 dev SoftAp0]
Nov 20 18:42:58 beaglebone bb-bbai-tether[1296]: bbai:tether [iptables -w -t nat -A POSTROUTING -o wlan0 -j MASQUERADE]
Nov 20 18:42:58 beaglebone bb-bbai-tether[1296]: bbai:tether [iptables -w -A FORWARD -i wlan0 -o SoftAp0 -m state --state RELATED,ESTABLISHED -j ACCEPT]
Nov 20 18:42:58 beaglebone bb-bbai-tether[1296]: bbai:tether [iptables -w -A FORWARD -i SoftAp0 -o wlan0 -j ACCEPT]
Nov 20 18:42:58 beaglebone bb-bbai-tether[1296]: Configuration file: /tmp/hostapd-bbai.conf
Nov 20 18:42:58 beaglebone bb-bbai-tether[1296]: SoftAp0: Could not connect to kernel driver
Nov 20 18:42:58 beaglebone bb-bbai-tether[1296]: Using interface SoftAp0 with hwaddr 00:00:00:00:00:00 and ssid "BeagleBone-0000"
Nov 20 18:42:58 beaglebone bb-bbai-tether[1296]: SoftAp0: interface state UNINITIALIZED->ENABLED
Nov 20 18:42:58 beaglebone bb-bbai-tether[1296]: SoftAp0: AP-ENABLED
Nov 20 18:42:58 beaglebone systemd[1]: Started BBAI brcmfmac tether Service.
```

all 0's, something is not loading..

```
[   30.744321] brcmfmac: brcmf_fw_map_chip_to_name: using brcm/brcmfmac43455-sdio.bin for chip 0x004345(17221) rev 0x000006
[   31.102397] brcmfmac: brcmf_c_preinit_dcmds: Firmware version = wl0: Mar  1 2015 07:29:38 version 7.45.18 (r538002) FWID 01-6a2c8ad4
```

closed 2020-11-20 by RobertCNelson

RobertCNelson, 2020-11-20:
and brain fart:

https://github.com/RobertCNelson/bb-bbai-firmware/commit/f7537f390e191648b3cc9b74e4e25b72666f7f75

something changed where we are getting all 0's for the mac address, so we grab it from the cached version: /etc/cpsw_0_mac

```
debian@beaglebone:~$ hexdump -v -e '1/1 "%02X" ":"'  /proc/device-tree/ocp/ethernet@48484000/slave@48480200/mac-address 
00:00:00:00:00:00
```

```
debian@beaglebone:~$ cat /etc/cpsw_*
1C:BA:8C:A2:ED:68
1C:BA:8C:A2:ED:6A
1C:BA:8C:A2:ED:6B
1C:BA:8C:A2:ED:6C
1C:BA:8C:A2:ED:6D
1C:BA:8C:A2:ED:6E
```

Regards,

RobertCNelson, 2020-11-20:
```
debian@beaglebone:~$ journalctl | grep tether
Nov 20 19:21:48 beaglebone systemd[1]: Starting BBAI brcmfmac tether Service...
Nov 20 19:21:48 beaglebone bb-bbai-tether[1280]: bbai:tether [ethernet@48484000/slave@48480200/mac-address returned 00 mac]
Nov 20 19:21:53 beaglebone bb-bbai-tether[1280]: bbai:tether waiting for /sys/class/net/wlan0
Nov 20 19:21:58 beaglebone bb-bbai-tether[1280]: bbai:tether waiting for /sys/class/net/wlan0
Nov 20 19:22:03 beaglebone bb-bbai-tether[1280]: bbai:tether waiting for /sys/class/net/wlan0
Nov 20 19:22:08 beaglebone bb-bbai-tether[1280]: bbai:wlan0:Power Management:off
Nov 20 19:22:08 beaglebone bb-bbai-tether[1280]: bbai:tether [iw phy phy0 interface add SoftAp0 type __ap]
Nov 20 19:22:08 beaglebone bb-bbai-tether[1280]: bbai:tether [phy0 created]
Nov 20 19:22:08 beaglebone bb-bbai-tether[1280]: bbai:tether [ip link set dev SoftAp0 down]
Nov 20 19:22:08 beaglebone bb-bbai-tether[1280]: bbai:tether [ip link set dev SoftAp0 address 1C:BA:8C:A2:ED:68]
Nov 20 19:22:08 beaglebone bb-bbai-tether[1280]: bbai:tether [ip link set dev SoftAp0 up]
Nov 20 19:22:09 beaglebone bb-bbai-tether[1280]: bbai:tether [ip addr add 192.168.8.1/24 broadcast 192.168.8.255 dev SoftAp0]
Nov 20 19:22:09 beaglebone bb-bbai-tether[1280]: bbai:tether [iptables -w -t nat -A POSTROUTING -o wlan0 -j MASQUERADE]
Nov 20 19:22:09 beaglebone bb-bbai-tether[1280]: bbai:tether [iptables -w -A FORWARD -i wlan0 -o SoftAp0 -m state --state RELATED,ESTABLISHED -j ACCEPT]
Nov 20 19:22:09 beaglebone bb-bbai-tether[1280]: bbai:tether [iptables -w -A FORWARD -i SoftAp0 -o wlan0 -j ACCEPT]
Nov 20 19:22:10 beaglebone bb-bbai-tether[1280]: Configuration file: /tmp/hostapd-bbai.conf
Nov 20 19:22:10 beaglebone bb-bbai-tether[1280]: SoftAp0: Could not connect to kernel driver
Nov 20 19:22:10 beaglebone bb-bbai-tether[1280]: Using interface SoftAp0 with hwaddr 1c:ba:8c:a2:ed:68 and ssid "BeagleBone-ED68"
Nov 20 19:22:10 beaglebone bb-bbai-tether[1280]: SoftAp0: interface state UNINITIALIZED->ENABLED
Nov 20 19:22:10 beaglebone bb-bbai-tether[1280]: SoftAp0: AP-ENABLED
Nov 20 19:22:10 beaglebone systemd[1]: Started BBAI brcmfmac tether Service.
```

and i see BeagleBone-ED68 on my phone..

```
Nov 20 19:23:01 beaglebone hostapd[3306]: SoftAp0: STA be:4c:be:ce:ff:18 IEEE 802.11: associated
Nov 20 19:23:01 beaglebone hostapd[3306]: SoftAp0: STA be:4c:be:ce:ff:18 RADIUS: starting accounting session E2EABAC7C10BD937
Nov 20 19:23:01 beaglebone hostapd[3306]: SoftAp0: STA be:4c:be:ce:ff:18 WPA: pairwise key handshake completed (RSN)
Nov 20 19:23:05 beaglebone dnsmasq-dhcp[3295]: DHCPDISCOVER(SoftAp0) be:4c:be:ce:ff:18
Nov 20 19:23:05 beaglebone dnsmasq-dhcp[3295]: DHCPOFFER(SoftAp0) 192.168.8.146 be:4c:be:ce:ff:18
Nov 20 19:23:05 beaglebone dnsmasq-dhcp[3295]: DHCPDISCOVER(SoftAp0) be:4c:be:ce:ff:18
Nov 20 19:23:05 beaglebone dnsmasq-dhcp[3295]: DHCPOFFER(SoftAp0) 192.168.8.146 be:4c:be:ce:ff:18
Nov 20 19:23:05 beaglebone dnsmasq-dhcp[3295]: DHCPDISCOVER(SoftAp0) be:4c:be:ce:ff:18
Nov 20 19:23:05 beaglebone dnsmasq-dhcp[3295]: DHCPOFFER(SoftAp0) 192.168.8.146 be:4c:be:ce:ff:18
Nov 20 19:23:05 beaglebone dnsmasq-dhcp[3295]: DHCPREQUEST(SoftAp0) 192.168.8.146 be:4c:be:ce:ff:18
Nov 20 19:23:05 beaglebone dnsmasq-dhcp[3295]: DHCPACK(SoftAp0) 192.168.8.146 be:4c:be:ce:ff:18
```

and it connected..

So just run:

```
sudo apt update
sudo apt upgrade
sudo reboot
```

Regards,

dmconover, 2020-11-20:
Works for me! Thanks so much!