beagleboard/Latest-Images #80: Include overlayroot in IOT OS images (again :-)

stepleton, 2021-01-23:
overlayroot allows you to remount / (or any other filesystem) as read-only, after which any modifications to / are only committed to an overlaid RAM risk. This facility is great for appliances and IOT devices, since it gives you even more certainty that power cuts (e.g. from the user pulling the cord) are unlikely to damage the filesystem. You also won't delay start-up at the next power-on so the system can do journal recovery after a dirty shutdown. Finally, it also means that you can recover from anything that messes up the filesystem by simply rebooting, which enhances reliability in the field.

overlayroot was present in later 9.x images and seems no longer to be included in 10.x images. It is enabled by default in [a project I'm maintaining](https://github.com/stepleton/cameo/tree/master/aphid#8-optional-enable-power-off-robustness-for-the-root-filesystem).

The overlayroot package can be found on [repos.rcn-ee.com](http://repos.rcn-ee.com/debian/pool/main/c/cloud-initramfs-tools/), but it is difficult to install on a PocketBeagle since the PocketBeagle lacks network access. It doesn't seem very big (but I don't know how big its dependencies are). Would it be possible to include it in future IOT OS images?

stepleton, 2021-01-23:
overlaid RAM disk, that is.

stepleton, 2021-01-23:
Another minor correction: overlayroot is present in the 10.0 image dated 2019-07-07; it's absent in the later 10.3 image.

RobertCNelson, 2021-01-24:
Hi @stepleton , sure i'll bring it back no problem..  We need to find a way to quiet the "cryptsetup-initramfs" warnings, as i got a lot of confusion/questions from users about it.

Regards,

closed 2021-01-24 by RobertCNelson

stepleton, 2021-01-24:
Oh, thanks very much! Please feel free to contact me at <my github username>@gmail if I can help in any way.