beagleboard/Latest-Images #81: Add user node-red to group users

jadonk, 2021-02-24:
For people using commands that use /dev/uio* which is assigned to group 'users', it would be handy to add the node-red users.

jadonk, 2021-02-24:
Reported by Ritesh Bhatt of Georgia Tech.

RobertCNelson, 2021-02-25:
@jadonk where is the /dev/uio rule set? i had mostly converted everything to use a generic "gpio" group, instead of random ones..

unless my eye's aren't working.. i don't see it..

https://github.com/beagleboard/customizations/tree/master/etc/udev/rules.d

Regards,

RobertCNelson, 2021-02-25:
Brain fart, i see it..

https://github.com/RobertCNelson/omap-image-builder/blob/master/scripts/chroot.sh#L982-L989

i'll nuke those and convert it to 'bb-customizations' package..

Regards,

RobertCNelson, 2021-02-25:
Fixed with: https://github.com/beagleboard/customizations/commit/65255a59650375b9b83b3a3d7cc7837cad124d76

closed 2021-02-25 by RobertCNelson