beagleboard/Latest-Images #83: Up to Date Documentation?

McNugget6750, 2021-03-24:
Not a software bug in general - but you could call it a documentation bug in general.

I am trying to get the BBB into service, I have never worked with it before, only during the last month or so.
After many days/weeks of research I have Python, I2C, and some very basic GPIO working. Why does it take me so long? Because the BBB Debian was changed so much since its release that no tutorial writer could keep up.
Every single example about every single device in the BB, doesn't work. There are seemingly no current tutorials. The Adafruit libraries are all deprecated. TI abandoned pasm. Code does not compile for PRUs. Simply blinking an LED takes days of research.

Okey, enough venting. I realize, maybe I'm simply looking in the wrong spot and don't use google well enough for BBB. Where can I find good starting points for the latest BBB image? I'd like to get into PRU coding, I'm experimenting with I2C, GPIO, SPI, and other devices.

Kind regards,
Timo

McNugget6750, 2021-03-24:
Example, straight from the PRU Cookbook: https://beagleboard.org/static/prucookbook/#_blinking_an_led

debian@beaglebone:/var/lib/cloud9/pru_tests/PRUCookbook/docs/02start/code$ source setup.sh
TARGET=hello.pru0
debian@beaglebone:/var/lib/cloud9/pru_tests/PRUCookbook/docs/02start/code$ make
/var/lib/cloud9/common/Makefile:27: MODEL=TI_AM335x_BeagleBone_Black,TARGET=hello.pru0,COMMON=/var/lib/cloud9/common
/var/lib/cloud9/common/Makefile:146: GEN_DIR=/tmp/cloud9-examples,CHIP=am335x,PROC=pru,PRUN=0,PRU_DIR=/dev/remoteproc/pruss-core0,EXE=.out
-    Stopping PRU 0
/bin/sh: 1: cannot create /dev/remoteproc/pruss-core0/state: Directory nonexistent
Cannot stop 0
-       copying firmware file /tmp/cloud9-examples/hello.pru0.out to /lib/firmware/am335x-pru0-fw
write_init_pins.sh
writing "none" to "/sys/class/leds/beaglebone:green:usr3/trigger"
-    Starting PRU 0
/bin/sh: 1: cannot create /dev/remoteproc/pruss-core0/state: Directory nonexistent
make: *** [/var/lib/cloud9/common/Makefile:166: start] Error 2
debian@beaglebone:/var/lib/cloud9/pru_tests/PRUCookbook/docs/02start/code$

RobertCNelson, 2021-03-24:
@McNugget6750 , that should be fixed via:

```
sudo apt update 
sudo apt upgrade
```

if not, please run:

```
sudo /opt/scripts/tools/version.sh
```

This is the udev rule that set's that up..

https://github.com/beagleboard/customizations/blob/master/etc/udev/rules.d/86-remoteproc-noroot.rules

Regards,

McNugget6750, 2021-03-24:
Thanks for the quick reply, Robert!

I'll try
`sudo /opt/scripts/tools/version.sh`

I know it's not solved by updating. I do that almost every day before I start any work on the BBB.

Cheers
Timo

McNugget6750, 2021-03-24:
Unfortunately, it doesn't help. I also tried rebooting after executing the script. Still the same error during compile.

debian@beaglebone:/var/lib/cloud9$ cd /var/lib/cloud9/pru_tests/PRUCookbook/docs/02start/code
debian@beaglebone:/var/lib/cloud9/pru_tests/PRUCookbook/docs/02start/code$ source setup.sh
TARGET=hello.pru0
debian@beaglebone:/var/lib/cloud9/pru_tests/PRUCookbook/docs/02start/code$ make
/var/lib/cloud9/common/Makefile:27: MODEL=TI_AM335x_BeagleBone_Black,TARGET=hello.pru0,COMMON=/var/lib/cloud9/common
/var/lib/cloud9/common/Makefile:146: GEN_DIR=/tmp/cloud9-examples,CHIP=am335x,PROC=pru,PRUN=0,PRU_DIR=/dev/remoteproc/pruss-core0,EXE=.out
-    Stopping PRU 0
**/bin/sh: 1: cannot create /dev/remoteproc/pruss-core0/state: Directory nonexistent**
Cannot stop 0
CC      hello.pru0.c
"/var/lib/cloud9/common/prugpio.h", line 105: warning #1181-D: #warning directive: "Found am335x"
LD      /tmp/cloud9-examples/hello.pru0.o
-       copying firmware file /tmp/cloud9-examples/hello.pru0.out to /lib/firmware/am335x-pru0-fw
write_init_pins.sh
writing "none" to "/sys/class/leds/beaglebone:green:usr3/trigger"
-    Starting PRU 0
**/bin/sh: 1: cannot create /dev/remoteproc/pruss-core0/state: Directory nonexistent**
make: *** [/var/lib/cloud9/common/Makefile:166: start] Error 2
rm /tmp/cloud9-examples/hello.pru0.o
debian@beaglebone:/var/lib/cloud9/pru_tests/PRUCookbook/docs/02start/code$

McNugget6750, 2021-03-24:
debian@beaglebone:/var/lib/cloud9/pru_tests/PRUCookbook/docs/02start/code$ sudo /opt/scripts/tools/version.sh
[sudo] password for debian:
git:/opt/scripts/:[e4e4854ef8ff9ada5c85553376043ee7679167ca]
eeprom:[A335BNLT000C1815BBBK0865]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Buster IoT Image 2020-04-06]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot SPL 2019.04-00002-g07d5700e21 (Mar 06 2020 - 11:24:55 -0600)]:[location: dd MBR]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2019.04-00002-g07d5700e21]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot SPL 2018.03-00002-gac9cce7c6a (Apr 05 2018 - 13:07:46 -0500)]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.03-00002-gac9cce7c6a]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-boneblack.dts]
UBOOT: Loaded Overlay:[beaglelogic-00A0]
kernel:[4.19.94-ti-r42]
nodejs:[v10.15.2]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[disable_uboot_overlay_emmc=1]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[disable_uboot_overlay_audio=1]
uboot_overlay_options:[disable_uboot_overlay_wireless=1]
uboot_overlay_options:[enable_uboot_cape_universal=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/beaglelogic-00A0.dtbo]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20210225.0-0~buster+20210225]
pkg:[bb-customizations]:[1.20210225.0-0~buster+20210225]
pkg:[bb-usb-gadgets]:[1.20200504.0-0~buster+20200504]
pkg:[bb-wl18xx-firmware]:[1.20200813.1-0~buster+20200813]
pkg:[kmod]:[26-1]
pkg:[librobotcontrol]:[1.0.5-git20200715.0-0~buster+20200716]
pkg:[firmware-ti-connectivity]:[20190717-2rcnee1~buster+20200305]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal bluetooth netdev i2c gpio pwm eqep remoteproc admin spi iio docker tisdk weston-launch xenomai cloud9ide beaglelogic]
cmdline:[console=ttyO0,115200n8 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet]
dmesg | grep remote
[   52.225093] remoteproc remoteproc0: wkup_m3 is available
[   52.386934] remoteproc remoteproc0: powering up wkup_m3
[   52.386957] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217168
[   52.387229] remoteproc remoteproc0: remote processor wkup_m3 is now up
dmesg | grep pru
dmesg | grep pinctrl-single
[    0.848817] pinctrl-single 44e10800.pinmux: 142 pins, size 568
dmesg | grep gpio-of-helper
[    0.850558] gpio-of-helper ocp:cape-universal: ready
lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END

McNugget6750, 2021-03-24:
I don't know for sure but it looks like beaglelogic is attached to a PRU. I never succeeded installing BeagleLogic, the installer just won't succeed and as a result I wasn't able to capture any data. But here it does look like uboot loads the beaglelogic pru core? Do I have to get rid of that first? How?

RobertCNelson, 2021-03-24:
```
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot SPL 2019.04-00002-g07d5700e21 (Mar 06 2020 - 11:24:55 -0600)]:[location: dd MBR]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2019.04-00002-g07d5700e21]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot SPL 2018.03-00002-gac9cce7c6a (Apr 05 2018 - 13:07:46 -0500)]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2018.03-00002-gac9cce7c6a]:[location: dd MBR]
```
Let's fix that ^...
```
sudo apt update
sudo apt install bb-u-boot-am335x-evm
cd /opt/u-boot/bb-u-boot-am335x-evm/
sudo ./install.sh
sudo reboot
```

```
kernel:[4.19.94-ti-r42]
```
Fix that ^
```
sudo /opt/scripts/tools/update_kernel.sh
sudo reboot
```

```
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/beaglelogic-00A0.dtbo]
```

Fix that ^
```
in /boot/uEnv.xt:
```
Change:
```
uboot_overlay_pru=/lib/firmware/beaglelogic-00A0.dtbo
```
to:
```
uboot_overlay_pru=AM335X-PRU-RPROC-4-19-TI-00A0.dtbo
```
and reboot...

Then you'll be back to "default"...

McNugget6750, 2021-03-27:
Great support, thank you!
I'm a bit surprised because apart from my cape installation for BeagleLogic, this is a fresh installation from the beaglebone website. So all the kernel updates should have been done already. Maybe there is something to be updated in the documentation.

Summary: Your suggestions all worked with no errors and the example compiles now.

jadonk, 2021-05-06:
@McNugget6750 OK to close this issue?

treedy, 2022-04-03:
Resurrecting this bug/FR. I would say, no, it isn't ok to close. Documentation is still severely out of date or incorrect. Even the instructions in this bug have errors. For example, the errors at the end of `update_kernel.sh`. Is this project active or should the community move on to other alternatives?

```
$ sudo /opt/scripts/tools/update_kernel.sh
info: Updating libgnutls30
Hit:1 http://deb.debian.org/debian buster InRelease
Hit:2 http://deb.debian.org/debian buster-updates InRelease
Hit:3 http://deb.debian.org/debian-security buster/updates InRelease
Hit:4 http://repos.rcn-ee.com/debian buster InRelease
Reading package lists... Done
Building dependency tree       
Reading state information... Done
1 package can be upgraded. Run 'apt list --upgradable' to see it.
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libgnutls30 is already the newest version (3.6.7-4+deb10u7).
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.
info: checking archive
2022-04-03 11:26:04 URL:https://rcn-ee.com/repos/latest/buster-armhf/LATEST-ti [196/196] -> "LATEST-ti" [1]
-----------------------------
Kernel Options:
ABI:1 LTS414 4.14.108-ti-r144
ABI:1 LTS419 4.19.94-ti-r72
ABI:1 LTS54 5.4.106-ti-r40
ABI:1 LTS510 5.10.106-ti-r41
-----------------------------
Kernel version options:
-----------------------------
LTS414: --lts-4_14
LTS419: --lts-4_19
LTS54: --lts-5_4
LTS510: --lts-5_10
-----------------------------
info: you are running: [4.19.94-ti-r64], latest is: [4.19.94-ti-r72] updating...
Hit:1 http://deb.debian.org/debian buster InRelease
Hit:2 http://deb.debian.org/debian buster-updates InRelease
Hit:3 http://repos.rcn-ee.com/debian buster InRelease
Hit:4 http://deb.debian.org/debian-security buster/updates InRelease
Reading package lists... Done
Building dependency tree       
Reading state information... Done
1 package can be upgraded. Run 'apt list --upgradable' to see it.
Checking dpkg...
Checking apt-cache...
debug: reinstalling: [linux-image-4.19.94-ti-r72]
Reading package lists... Done
Building dependency tree       
Reading state information... Done
0 upgraded, 0 newly installed, 1 reinstalled, 0 to remove and 1 not upgraded.
Need to get 28.6 MB of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 http://repos.rcn-ee.com/debian buster/main armhf linux-image-4.19.94-ti-r72 armhf 1buster [28.6 MB]
Fetched 28.6 MB in 6s (4,965 kB/s)                     
(Reading database ... 89433 files and directories currently installed.)
Preparing to unpack .../linux-image-4.19.94-ti-r72_1buster_armhf.deb ...
Unpacking linux-image-4.19.94-ti-r72 (1buster) over (1buster) ...
Setting up linux-image-4.19.94-ti-r72 (1buster) ...
update-initramfs: Generating /boot/initrd.img-4.19.94-ti-r72
zz-uenv_txt: Updating /boot/uEnv.txt [uname_r=4.19.94-ti-r72]
Reading package lists... Done
Building dependency tree       
Reading state information... Done
0 upgraded, 0 newly installed, 1 reinstalled, 0 to remove and 1 not upgraded.
Need to get 6,064 B of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 http://repos.rcn-ee.com/debian buster/main armhf libpruio-modules-4.19.94-ti-r72 armhf 1buster [6,064 B]
Fetched 6,064 B in 0s (22.1 kB/s)                    
(Reading database ... 89433 files and directories currently installed.)
Preparing to unpack .../libpruio-modules-4.19.94-ti-r72_1buster_armhf.deb ...
Unpacking libpruio-modules-4.19.94-ti-r72 (1buster) over (1buster) ...
Setting up libpruio-modules-4.19.94-ti-r72 (1buster) ...
Reading package lists... Done
Building dependency tree       
Reading state information... Done
0 upgraded, 0 newly installed, 1 reinstalled, 0 to remove and 1 not upgraded.
Need to get 100 kB of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 http://repos.rcn-ee.com/debian buster/main armhf ti-sgx-ti335x-modules-4.19.94-ti-r72 armhf 1buster [100 kB]
Fetched 100 kB in 0s (240 kB/s)                          
(Reading database ... 89433 files and directories currently installed.)
Preparing to unpack .../ti-sgx-ti335x-modules-4.19.94-ti-r72_1buster_armhf.deb ...
Unpacking ti-sgx-ti335x-modules-4.19.94-ti-r72 (1buster) over (1buster) ...
Setting up ti-sgx-ti335x-modules-4.19.94-ti-r72 (1buster) ...
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package seeed-modules-4.19.94-ti-r72
E: Couldn't find any package by glob 'seeed-modules-4.19.94-ti-r72'
E: Couldn't find any package by regex 'seeed-modules-4.19.94-ti-r72'
update-initramfs: Generating /boot/initrd.img-4.19.94-ti-r72
```

RobertCNelson, 2022-04-03:
https://forum.beagleboard.org/t/debian-10-x-11-x-kernel-updates/30928