beagleboard/Latest-Images #84: Siddharth's Proposal for BeagleBoard.org

SiddharthSharabu-1729, 2021-05-24:
# [Github Externship](https://github-externships.github.io/externship/) Proposal

* Student ID: `#84086761`
* Mentors: `hendersa, Abhishek_, sakethr98`
* Repository: https://github.com/SiddharthSharabu-1729/beagleboard/Latest-Images

## Status

This is a Proposal from Siddhartha Achari Sharabu

## About you

* IRC: `SiddharthSharabu_1729`
* School: `SVR Engineering College`
* Country: `Nandyal, Kurnool Dist, Andhra Pradesh, India`
* Primary language: `Telugu, English`
* Typical work hours: `6AM to 9AM & 5PM to 9PM`
* Previous externship and open source experience: `I dont have any Experience on opensource projects`

## About your project

Project name: The Deck Penetration Testing and Forensics Platform
Description

Firsly I would like to test and Experience the The Deck Penetration Testing and Forensics Platform on the Beagle Board. After I would like make changes in kernel to support the onboard WIFI which supports monitoring mode. In Raspberry Pi the Rasbian operating system supports on board wifi monitoring mode. It would like to work on it. The Technologies and languages used are Linux Kernel, Python and C. 

### Timeline

Provide a development timeline with a milestone week and any pre-work. (A realistic timeline is critical to our selection process.)
| Date   | Milestone |
| ------ | --------- |
| First Week | Start     |
| 2nd & 3rd Week | Test the device |
| 4th - 5th Week | Talk with Mentors |
| 6th - 11th Week | Working on the Project |
| 12th - 13th Week | Testing and Finalising the Updates |

### Experience and approach

I am an Electrical and Electronics Engineering Graduate, I am also a Beginner in Penetration Testing & Bug Bounty Hunting. I had a good experience and Knowledge on the Linux Kernel and Operating System. I made some intresting projects using Raspberry Pi like Installing various Operating System and testing that operating systems, A Portable Whether Monitoring Station using Raspberry Pi, Hacked my own PI to test my skills by installing Damn Vulnarable PI (Sticky Fingers Kali-Pi). Also made Some Projects using Arduino like CNC using Servo and Stepper Motors etc.

I had practied and Hacked Various online machines in HackTheBox.com. And Participated in capturing the flags in Hackerone.com. And know I am very Excited to work with your company. Not only this project I am comfortable and I can work on any project on your company beacuse I am aware and had a good experience on all of your projects. I think I will work with you. Looking Forward for Your Positive Reply. I will attach a certificate on learning Ethical Hacking from Zaid Security .
Thank You

### What will you do if you get stuck on your project and your mentor isn’t around?

Now-a-days Internet had full of resources that we an find and learn anything & Everything to learn. I will work on my own and I will take only suggestions and doudts only. So even if the mentors are not available I can do my own work without any problem.

### Benefit

If we could sucessfully completed this project we can get great demand on our product. Beacause the Raspberry Pi is the latest and Trending single board computer and can use this for Penetration Testing, If we complete this we can get all and advanced feautures that Raspberry Pi had.

## Conclusion

If you are not interested in this project, I can work on any other latest projetcs like ArduPilot, RTEMS, DonkeyCar, etc. projects Looking Forward For your positive reply. Thankyou

ppolstra, 2021-06-02:
I would be willing to mentor you on this project.

SiddharthSharabu-1729, 2021-06-03:
> 

> I would be willing to mentor you on this project.

Sure sir looking forward to work with you.