beagleboard/Latest-Images #89: Abhishek Shetty's GitHub Externship proposal

Sheldor-abhi, 2021-05-31:
# [Github Externship](https://github-externships.github.io/externship/) Proposal

* Student ID: `Sheldor-abhi`
* Application No: `16-05_Abh437_tfa_291`
* Mentors: `jkridner, sakethr98, Abhishek_`
* Repository: https://github.com/Sheldor-abhi/Latest-Images

## Status

This is Abhishek Shetty's proposal for the GitHub externship. 

## About you

* School: `Manipal Institute of Technology`
* Country: `Pune, Maharastra, India`
* Primary languages: `English, Hindi, Tulu`
* Typical work hours (We have mentors in various time zones): `12pm-2pm and 6pm-9pm IST`
* Previous externship and open source experience: `I don't have any experience with open source projects`

## About your project

Project name: Home security system using bluetooth

Description:
I would like to make a security system which checks the person's name at the entrance of a house/room using his/her bluetooth ID of their phones and if their ID is registered they would be allowed to enter the house/room and the owner would be notified via text/mail.
If the person is not registered, their photo/video would be sent to the owner, and they may choose to give them a one time access/ register them or deny their entrance.
Also if someone without access enters, the owner is notified with a live feed.

### Timeline

Provide a development timeline with a milestone week and any pre-work. (A realistic timeline is critical to our selection process.)
| Date   | Milestone |
| ------ | --------- |
| Week 1 | Understand the RISC-V architecture|
| Week 2 | Understanding the linux kernel implementation in this project   |
| Week 3 and 4 | Talking with mentors |
| Week 5-9 | Start implementation of the project|
| Week 10-12 | Finishing up and getting feedback from the mentors |

The work for week 5-9 are as follows:

Bluetooth device trigerred action
Setting up communication between the board and user's phone
Working on a locking system which opens and closes on command
Working on setting up a live video feed from the board to the user's phone

### Experience and approach

I am currently pursuing Electronics and Instrumentation.
I have experience using the arduino. I am experienced in Python and C.
I have written code for pi-pi communication using python on my PC.
The only thing I need help with is with developing linux kernels, as this board is a RISC-V based board unlike other ARM based boards. I need the guidance of the mentors, and will be learning about it online on my own as well.

### Benefit

If completed succesfully, this might provide a base for a more complex and effective security system at an affordable price. 

## Conclusion

I would really like to make this project, and with your organisation's help, my project could be more than I envisioned.

jadonk, 2021-06-16:
How can this be more of a reusable software library?

Sheldor-abhi, 2021-06-18:
What I am thinking is to implement a REST API framework for my project. 
So instead of sending the data(from the camera and bluetooth verification results) just to the user's phone it can be accessed by any client who requests it. So that they have an option to build on it, and I will implement the client side with an android app for notifying the user and adding new people with access.