beagleboard/Latest-Images #91: Latest testing images updates

mjeveritt, 2021-07-08:
**Describe the bug**
Latest testing images are nearly a year old, and an older version of Debian than released.
uSD 'standalone' images are packed with the flasher script as default, and not self-booting.

**Describe which image you are running (example: bone-debian-10.2-iot-armhf-2020-02-03-4gb.img.xz) **
```bone-debian-10.5-console-armhf-2020-08-25-1gb.img.xz```
per https://elinux.org/Beagleboard:Latest-images-testing#Debian_10_.28Buster.29_Console

**Describe your hardware**
BBB-C (4Gb)

**Describe any Capes Installed**
N/A

**Please run this built-in script**
```sudo /opt/scripts/tools/version.sh```

**To Reproduce**
1. Download image
2. Write to microSD card
3. Boot BBB to card

**Expected behavior**
Current Debian image/kernels boot from SD card

mjeveritt, 2021-07-08:
It looks like the builder has stalled/etc as the images on https://forum.digikey.com/t/debian-getting-started-with-the-beaglebone-black/12967 are certainly unaffected.
Pinging @RobertCNelson as I suspect this is his domain?! :wave:

RobertCNelson, 2021-07-08:
> uSD 'standalone' images are packed with the flasher script as default, and not self-booting.
> bone-debian-10.5-console-armhf-2020-08-25-1gb.img.xz

Testing...

mjeveritt, 2021-07-08:
> @mjeveritt please state what is wrong with which specific image.. If it's actually still broken with 2020-08-25, it's probably also broken with https://rcn-ee.net/rootfs/bb.org/testing/2021-07-07/
> 
> Regards,

I'll double-check, as I think I misinterpreted some of the LED sequences in the new images! :roll_eyes: 
It does look like there are some issues, but I will test all of the 'latest-images' and 'latest-testing-images' and see what I'm doing wrong, or where which image is failing.

Thanks for the prompt ACK nonetheless ! :smiley: