beagleboard/Latest-Images #92: Can't pinmux P9_30 (SPI_HEADER_PIN_3) on BB_Blue

StjerneIdioten, 2021-07-31:
**Describe the bug**
I am fairly new to working with the beaglebone blue, so I don't know much about the inner workings. But I updated it with the newest flasher image from [the site](https://beagleboard.org/latest-images) and tried my hand at coding some C for it using the Robotics Control Library. Now for my particular project I need a bit of extra GPIO and don't need to use SPI, so I tried pinmuxing some of the SPI pins into GPIO pins. The first I've tried is pin *P9_30* or *SPI_HEADER_PIN_3* as the define is called in RCL. But I keep getting an error with the following wording:
```
ERORR opening pinmux driver: No such file or directory
can't open: /sys/devices/platform/ocp/ocp:P9_30_pinmux/state
```
Which at first lead me to believe it was a problem with permissions, but sudo didn't help either. Investigating a bit I can see that `/sys/devices/platform/ocp/ocp:P9_30_pinmux` does exist, but there is no *state* file to read as there are with the others. I've spent a lot of time reading around and trying to understand how BBB works now and I finally came by someone mentioning to try and run `sudo /opt/scripts/tools/version.sh` and check if there is anything suspicious in there and there actually is but I am just gonna post the entire output for brevity: 
```
git:/opt/scripts/:[b39ec679648a6be8f25f48bd1c9784c1fc5a0c46]
eeprom:[A335BNLTBLA21736EL001138]
model:[TI_AM335x_BeagleBone_Blue]
dogtag:[BeagleBoard.org Debian Buster IoT Image 2020-04-06]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2019.04-00002-g07d5700e21]:[location: dd MBR]
UBOOT: Booted Device-Tree:[am335x-boneblue.dts]
UBOOT: Loaded Overlay:[AM335X-PRU-RPROC-4-19-TI-00A0]
UBOOT: Loaded Overlay:[BB-ADC-00A0]
kernel:[4.19.94-ti-r42]
nodejs:[v10.24.0]
/boot/uEnv.txt Settings:
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[disable_uboot_overlay_emmc=1]
uboot_overlay_options:[disable_uboot_overlay_video=1]
uboot_overlay_options:[disable_uboot_overlay_audio=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-19-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20210416.0-0~buster+20210416]
pkg:[bb-wl18xx-firmware]:[1.20210520.0-0~buster+20210520]
pkg:[kmod]:[26-1]
pkg:[librobotcontrol]:[1.0.5-git20200715.0-0~buster+20200716]
pkg:[firmware-ti-connectivity]:[20190717-2rcnee1~buster+20200305]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal bluetooth netdev i2c gpio pwm eqep remoteproc admin spi iio docker tisdk weston-launch xenomai cloud9ide]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk1p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 lpj=1990656 rng_core.default_quality=100 quiet]
dmesg | grep remote
[    9.930331] remoteproc remoteproc0: 4a334000.pru is available
[    9.938051] remoteproc remoteproc1: 4a338000.pru is available
[   54.492738] remoteproc remoteproc2: wkup_m3 is available
[   54.500169] remoteproc remoteproc2: powering up wkup_m3
[   54.500195] remoteproc remoteproc2: Booting fw image am335x-pm-firmware.elf, size 217168
[   54.500458] remoteproc remoteproc2: remote processor wkup_m3 is now up
dmesg | grep pru
[    9.930331] remoteproc remoteproc0: 4a334000.pru is available
[    9.930549] pru-rproc 4a334000.pru: PRU rproc node pru@4a334000 probed successfully
[    9.938051] remoteproc remoteproc1: 4a338000.pru is available
[    9.938231] pru-rproc 4a338000.pru: PRU rproc node pru@4a338000 probed successfully
dmesg | grep pinctrl-single
[    0.909493] pinctrl-single 44e10800.pinmux: 142 pins, size 568
[    1.114151] pinctrl-single 44e10800.pinmux: pin PIN102 already requested by 48300180.eqep; cannot claim for ocp:P9_30_pinmux
[    1.125556] pinctrl-single 44e10800.pinmux: pin-102 (ocp:P9_30_pinmux) status -22
[    1.133144] pinctrl-single 44e10800.pinmux: could not request pin 102 (PIN102) from group pinmux_P9_30_spi_pin  on device pinctrl-single
dmesg | grep gpio-of-helper
lsusb
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
As can be seen it does mention some problems with *P9_30* but as I said I am new and don't know enough about these things yet. 

**Describe which image you are running (example: bone-debian-10.2-iot-armhf-2020-02-03-4gb.img.xz) **
bone-eMMC-flasher-debian-10.3-iot-armhf-2020-04-06-4gb.img.xz

StjerneIdioten, 2021-07-31:
Actually I came across #67 just now, which isn't a problem with the same thing. But actually might just be the cause for this error, since I can see that the pin definition that has been mixed up is actually with the pin causing me problems.

unl0ck, 2022-02-09:
I have still this issue, is any fix?