beagleboard/Latest-Images #94: No Hdmi output after update && upgrade

atomino-max, 2021-09-23:
I am using bone-debian-9.9-lxqt-armhf-2019-08-03-4gb.img.xz on sd card and booting from sd.
Everything it's working fine , also if sometimes the system is frozen for 20/30 sec., but I have output on HDMI with resolution 1024x768 @60hz.
I have kernel version 4.14.108-ti-r143

The same problem happens if I use  bone-debian-9.12-imgtec-armhf-2020-04-06-4gb.img.xz

My board is element 14 BeagleBone Black Industrial, not any capes installed
---------------------------------------------------------------------------------
root@beaglebone:/boot#  /opt/scripts/tools/version.sh
root@beaglebone:~#  /opt/scripts/tools/version.sh
git:/opt/scripts/:[109f74fb87e6034ae1a8971a244064a8d5e090a5]
eeprom:[A335BNLTEIA03718BBBK063F]
model:[TI_AM335x_BeagleBone_Black]
dogtag:[BeagleBoard.org Debian Image 2019-08-03]
bootloader:[microSD-(push-button)]:[/dev/mmcblk0]:[U-Boot 2019.04-00002-gbb4af0f50f]:[location: dd MBR]
bootloader:[eMMC-(default)]:[/dev/mmcblk1]:[U-Boot 2019.04-00002-g07d5700e21]:[location: dd MBR]
kernel:[4.14.108-ti-r143]
nodejs:[v6.17.0]
uboot_overlay_options:[enable_uboot_overlays=1]
uboot_overlay_options:[uboot_overlay_pru=/lib/firmware/AM335X-PRU-RPROC-4-14-TI-00A0.dtbo]
uboot_overlay_options:[enable_uboot_cape_universal=1]
pkg check: to individually upgrade run: [sudo apt install --only-upgrade <pkg>]
pkg:[bb-cape-overlays]:[4.14.20210821.0-0~stretch+20210821]
pkg:[bb-wl18xx-firmware]:[1.20190227.1-0rcnee0~stretch+20190227]
pkg:[kmod]:[23-2rcnee1~stretch+20171005]
pkg:[librobotcontrol]:[1.0.5-git20200715.0-0~stretch+20200716]
pkg:[firmware-ti-connectivity]:[20190717-2rcnee1~stretch+20200305]
groups:[debian : debian adm kmem dialout cdrom floppy audio dip video plugdev users systemd-journal i2c bluetooth netdev gpio pwm eqep remoteproc admin spi tisdk weston-launch xenomai cloud9ide]
cmdline:[console=ttyO0,115200n8 bone_capemgr.uboot_capemgr_enabled=1 root=/dev/mmcblk0p1 ro rootfstype=ext4 rootwait coherent_pool=1M net.ifnames=0 rng_core.default_quality=100 quiet]
dmesg | grep remote
[    1.116787] remoteproc remoteproc0: wkup_m3 is available
[    1.239297] remoteproc remoteproc0: powering up wkup_m3
[    1.239419] remoteproc remoteproc0: Booting fw image am335x-pm-firmware.elf, size 217148
[    1.241916] remoteproc remoteproc0: remote processor wkup_m3 is now up
[   44.330886] remoteproc remoteproc1: 4a334000.pru is available
[   44.342865] remoteproc remoteproc2: 4a338000.pru is available
dmesg | grep pru
[   44.300745] pruss 4a300000.pruss: creating PRU cores and other child platform devices
[   44.330886] remoteproc remoteproc1: 4a334000.pru is available
[   44.331006] pru-rproc 4a334000.pru: PRU rproc node /ocp/pruss_soc_bus@4a326004/pruss@0/pru@34000 probed successfully
[   44.342865] remoteproc remoteproc2: 4a338000.pru is available
[   44.342983] pru-rproc 4a338000.pru: PRU rproc node /ocp/pruss_soc_bus@4a326004/pruss@0/pru@38000 probed successfully
dmesg | grep pinctrl-single
[    0.796837] pinctrl-single 44e10800.pinmux: 142 pins at pa f9e10800 size 568
[    1.162588] pinctrl-single 44e10800.pinmux: pin PIN108 already requested by ocp:A15_pinmux; cannot claim for 0-0070
[    1.173258] pinctrl-single 44e10800.pinmux: pin-108 (0-0070) status -22
[    1.179946] pinctrl-single 44e10800.pinmux: could not request pin 108 (PIN108) from group nxp_hdmi_bonelt_pins  on device pinctrl-single
dmesg | grep gpio-of-helper
[    0.809770] gpio-of-helper ocp:cape-universal: ready
lsusb
Bus 001 Device 004: ID 0461:4e6e Primax Electronics, Ltd
Bus 001 Device 005: ID 03f0:094a HP, Inc Optical Mouse [672662-001]
Bus 001 Device 002: ID 2109:2815 VIA Labs, Inc.
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
END
--------------------------------------------------------------------------------

root@beaglebone:~# cd /boot
root@beaglebone:/boot# cat uEnv.txt
#Docs: http://elinux.org/Beagleboard:U-boot_partitioning_layout_2.0

uname_r=4.14.108-ti-r143
#uuid=
#dtb=

###U-Boot Overlays###
###Documentation: http://elinux.org/Beagleboard:BeagleBoneBlack_Debian#U-Boot_Overlays
###Master Enable
enable_uboot_overlays=1
###
[... 125 lines cut ...]
[   137.413] 	Module class: X.Org Video Driver
[   137.413] 	ABI class: X.Org Video Driver, version 23.0
[   137.413] (II) FBDEV: driver for framebuffer: fbdev
[   137.414] (WW) Falling back to old probe method for fbdev
[   137.414] (II) Loading sub module "fbdevhw"
[   137.415] (II) LoadModule: "fbdevhw"
[   137.416] (II) Loading /usr/lib/xorg/modules/libfbdevhw.so
[   137.417] (II) Module fbdevhw: vendor="X.Org Foundation"
[   137.417] 	compiled for 1.19.2, module version = 0.0.2
[   137.417] 	ABI class: X.Org Video Driver, version 23.0
[   137.418] (EE) open /dev/fb0: No such file or directory
[   137.418] (EE) No devices detected.
[   137.418] (EE)
Fatal server error:
[   137.418] (EE) no screens found(EE)
[   137.418] (EE)
Please consult the The X.Org Foundation support
	 at http://wiki.x.org
 for help.
[   137.418] (EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
[   137.419] (EE)
[   137.420] (EE) Server terminated with error (1). Closing log file.

I hope somebody can help me.
Atomino

RobertCNelson, 2021-09-23:
@atomino-max should be fixed now:

```
sudo apt-get update
sudo apt-get upgrade
sudo reboot
```

Please test and confirm..

atomino-max, 2021-09-24:
I Robert, thanks a lot for your quick response.
Now I have Hdmi output correctly.
I take advantage of your availability to list 2 problems using vncviewer remotely:
1- after opening System Tools->QTerminal the keyboard seems not set correctly, i am using US keyboard on Mac pushing : qwerty  the output is : c.gvn  ( 5 bad chars instead of 6 )
again : up    show : xm

2-If I open Accessories->PCManFm File Manager the system open a window on local screen (Hdmi output ) not on the remote connection

Kind regards
Max ( Atomino )