beagleboard/Latest-Images #95: missing boot partition

bexizuo, 2021-12-15:
i downloaded images: 
bone-debian-10.3-console-armhf-2020-04-06-1gb.img.xz
bone-debian-10.3-iot-armhf-2020-04-06-4gb.img.xz
... both images consists only 1 single partition with name "rootfs" , 
is it correct to have only 1 partition ? shouldn't it be 2 ? boot and rootfs ?

RobertCNelson, 2021-12-15:
@bexizuo nope. ;)  am335x also supports raw mode, MLO/u-boot.img are located in the 'hole' at the start of the partition.

Regards,

bexizuo, 2021-12-15:
my board "beagleboard" ( first version ) , revC4 does not start , unfortunately i don't have null modem cable

RobertCNelson, 2021-12-15:
Bone is not BeagleBoard..

RobertCNelson, 2021-12-15:
This "image" might work on the BeagleBoard Rev C4... it's been years since anyone asked or years since i've tested that old board:

https://elinux.org/BeagleBoardDebian#BeagleBoard_xM

Good Luck!

bexizuo, 2021-12-15:
oh ... i see ... i'll try ... sorry for this issue ...

closed 2021-12-15 by bexizuo

RobertCNelson, 2021-12-15:
Here's a slightly newer build that might work:

https://rcn-ee.net/rootfs/bb.org/testing/2020-06-01/buster-console-xm/

Regards,