beagleboard/Latest-Images #96: Cloud-init support

mjudeikis, 2022-01-03:
It would be nice to support [cloud-init](https://cloudinit.readthedocs.io/en/latest/) in the images. 
Ubuntu does this and it very easy to bootstrap OS for mass production or headless install.

If cloud-init would be installed in the base image we can add scripts into image before we ship to the cards.
In example:
Wifi configuration could look like :
https://github.com/synpse-hq/image-tools/blob/main/assets/cloud-init/50-wifi.cfg.template 

where you mount image and add the files in the image:
https://github.com/synpse-hq/image-tools/blob/main/hack/bootstrap/bootstrap_ubuntu.sh

This allows users to configure/reconfigure 

cc: @RobertCNelson RobertCNelson

RobertCNelson, 2022-01-03:
Sweet thanks @mjudeikis

RobertCNelson, 2022-01-04:
@mjudeikis please give this a test... 

https://rcn-ee.net/rootfs/debian-armhf/2022-01-03/

Regards,

mjudeikis, 2022-01-08:
@RobertCNelson it worked like a charm. 
I were able to deliver files and configure system on boot using cloud-init by just mounting image before start and adding cloud-init config.

There is one limitation I face is configuring wifi:
```
Wifi configuration is only available to distros with netplan rendering support. 
```
Any chance you can add `netplan.io` (already default on ubuntu IoT) into the same image and republish to try?

I already integrated it to synpse.net CLI:
```
synpse build --image am57xx-debian-11.2-iot-armhf-2022-01-03-4gb.img --wifi-name xxxx --wifi-password xxxxxx --controller-uri http://localhost:8080/api  --access-key xxxxxxxxx --namespace xxxxxxx --project xxxxxx
```
This basically returns you a pre-configured image which automatically connects to users wifi network (if netplan is there) and you can access the device immediately :)

RobertCNelson, 2022-01-08:
We can definitely figure it out.. we just moved from connman -> systemd-networkd... it looks like netplan.io can run on top of systemd-networkd so we will do a little testing. ;)

mjudeikis, 2022-01-08:
> We can definitely figure it out.. we just moved from connman -> systemd-networkd... it looks like netplan.io can run on top of systemd-networkd so we will do a little testing. ;)

Yes please :)

mjudeikis, 2022-01-10:
I tried this on base debian 11 and found that for some reason even with netplan this didnt worked. Not sure if my way of building images or something else..